espressif/arduino-esp32 #3903: SD_MMC library issue with BOARD_HAS_1BIT_SDMMC

Stanimir-Petev, 2020-04-13:
### Hardware:
Board:							ESP32-Gateway
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.10
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I tried to use the example sketch from SD_MMC library: SDMMC_Test in 1Bit mode. SD card example itself works without problems. But the other 3 pins that are not used in this mode behave like they have been initialized as alternative function so they can't be used as GPIO (I have attached LED on them and tried to blink it). After some research I have found a way to solve the problem but it involved editing the library itself.
In the file SD_MMC.cpp lines 65-67 by default are:
```cpp
    if(mode1bit) {
        host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD mode
    }
```
That changes the mode from 4 to 1 bit, but based on what I saw it doesn't change slot_config structure and thus the pins to be initialized. So I added one line within the body of the if operator and is now:
```cpp
    if(mode1bit) {
        host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD mode
        slot_config.width = 1;
    }
```
So now it works but I am not sure if this is the best approach to the problem since I have to change library file. Is there another way to be done? If no - is there any chance to see fix for that in one of the future official releases of the Espressif package?

### Sketch:
Basically the sketch is the default example SDMMC_Test + added blinking LED on pin 4 and the setup - attached LED on pin 4.
[SDMMC_Test_toggle_4.txt](https://github.com/espressif/arduino-esp32/files/4469677/SDMMC_Test_toggle_4.txt)

DanKoloff, 2020-04-13:
I can confirm this problem exists for both SD card libraries from the examples ("SD_MMC" and "SD"). Defining 1-bit SD card mode still bugs the pins that are used by default for 4-bit SD card mode.

lbernstone, 2020-04-13:
If you fork the repository, edit the file (right in your browser), and submit a PR, then this can be fixed.

Stanimir-Petev, 2020-04-14:
I tried that now. I committed that change into my fork repository. But since I already had a pending pull request (from ~3 weeks ago) it got affected too. Before the commit (on the SD library) the previous pull request had 21/21 checks passed but now is 0/0 and has a yellow dot on it.
What am I supposed to do?
Delete the old PR and create a new one?

lbernstone, 2020-04-14:
No, I didn't know you had a pull pending.  If you have multiple PRs you need to make a new branch for your new change.  Google knows a lot about working with git and github.

cyberman54, 2020-06-06:
This issue can be closed.
Fixed by https://github.com/espressif/arduino-esp32/pull/4020
@Stanimir-Petev