espressif/arduino-esp32 #3906: Using WiFi.softAPConfig(), the client connects to the AP and reports an exception

LusterWong, 2020-04-14:
### Hardware:
Board:							ESP32
Core Installation version:			1.0.3
IDE name:						Arduino IDE/Eclipse
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 7

### Description:
The program just enabled softAP and set the IP address, gateway, mask, everything started normally, the serial port output "IDLE."

When the device(PC or mobile) is connected, the serial port outputs:
dhcps: send_offer >> udp_sendto result 0
And it prompts Guru Meditation Error: Core 0 panic'ed (InstrFetchProhibited). Exception was unhandled.

It is not clear how to solve it, I will only use simple arduino without in-depth understanding of the core of idf, what is the problem? How to solve?

Thanks.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <Arduino.h>
#include <WiFi.h>

const char *SSID = "FreeWifi";
const char *Password = "12345678";

void setup()
{
    Serial.begin(115200);
    delay(5000);

    IPAddress localIP(192,168,1,1);
    IPAddress gateway(192,168,1,0);
    IPAddress subnet(255,255,255,0);

    WiFi.softAPConfig(localIP, gateway, subnet);
    WiFi.softAP(SSID, Password);
    Serial.println(WiFi.softAPIP());

}

void loop()
{
    delay(2000);
    Serial.println("IDLE.");
}
```

### Debug Messages:
```
192.168.1.1
IDLE.
IDLE.
IDLE.
dhcps: send_offer>>udp_sendto result 0
IDLE.
IDLE.
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80111af9  A1      : 0x3ffb3d30  
A2      : 0x3ffcbcf8  A3      : 0x3ffcbf2c  A4      : 0x3ffcbb38  A5      : 0x3ffcbba0  
A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8011199c  A9      : 0x3ffb3cf0  
A10     : 0x3ffcbd08  A11     : 0x3ffcbf2c  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x40111af6:0x3ffb3d70 0x4011c181:0x3ffb3d90 0x40121189:0x3ffb3dd0 0x401262ae:0x3ffb3df0 0x401121fb:0x3ffb3e10 0x4008aff9:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

```

### Exception Decoder
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x40111af6: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x4011c181: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
0x40121189: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x401262ae: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x401121fb: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x4008aff9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

LusterWong, 2020-04-14:
I have tried not to use WiFi.softAPConfig (localIP, gateway, subnet), everything will be normal, the device can connect to softAP and have normal IP allocation

lbernstone, 2020-04-15:
Are you sure your gateway is at 192.168.1.0?  This is not usually possible on consumer grade equipment.

robacarp, 2020-05-15:
I'm able to confirm this crash with this snippet (192.168.1.1). I notice that the crash doesn't happen the first time a device is connected to an ssid, but the second. Changing the ssid gets a one shot successful connect -- probably the dhcp lease is attempting to be renewed rather than requested.

```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <DNSServer.h>

IPAddress ip(192, 168, 1, 1);

void setup() {
  delay(3000);

  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);

  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(ip, ip, IPAddress(255, 255, 255, 0));
  WiFi.softAP("test");
}

void loop() { }
```

LusterWong, 2020-05-23:
> Are you sure your gateway is at 192.168.1.0? This is not usually possible on consumer grade equipment.

Tried replacing with：
IPAddress localIP(192,168,1,23);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

No abnormality occurs, but the serial port prompts:
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited) at pc=00000000. Setting bp and returning..

Then stuck, no ‘IDLE’ output

LusterWong, 2020-05-23:
> I'm able to confirm this crash with this snippet (192.168.1.1). I notice that the crash doesn't happen the first time a device is connected to an ssid, but the second. Changing the ssid gets a one shot successful connect -- probably the dhcp lease is attempting to be renewed rather than requested.
> 
> ```c++
> #include <WiFi.h>
> #include <WebServer.h>
> #include <DNSServer.h>
> 
> IPAddress ip(192, 168, 1, 1);
> 
> void setup() {
>   delay(3000);
> 
>   WiFi.disconnect();
>   WiFi.mode(WIFI_OFF);
> 
>   WiFi.mode(WIFI_AP);
>   WiFi.softAPConfig(ip, ip, IPAddress(255, 255, 255, 0));
>   WiFi.softAP("test");
> }
> 
> void loop() { }
> ```

This is a very normal usage, I hope the official can help parse

me-no-dev, 2020-05-23:
this is an issue in the underlying IDF libraries. All I can offer is to try the 4.0 branch or esp32s2 branch, which tracks IDF master.

LusterWong, 2020-05-23:
> this is an issue in the underlying IDF libraries. All I can offer is to try the 4.0 branch or esp32s2 branch, which tracks IDF master.

Thanks.
I do n’t understand IDF, so I use arduion framework.
I remember that I tried this last year and it didn't have this problem. At that time, IDF was still v3.xx version. Later, it was suspended because of my work, and I recently reopened the ESP32 test.

immakermatty, 2020-06-07:
Weird... From my experience, the crash doesn't happen when the DCORE_DEBUG_LEVEL is set to level 5. But does when it is set to level 0 or 1.

LusterWong, 2020-06-11:
> DCORE_DEBUG_LEVEL

How to change DCORE_DEBUG_LEVEL with eclipse ?

mshoe007, 2020-07-05:
This is the same as 

https://github.com/espressif/arduino-esp32/issues/2025

ESP8266 has similar https://github.com/homieiot/homie-esp8266/issues/588
https://github.com/homieiot/homie-esp8266/issues/637

The 'fix' for me is a two second delay between WiFi.mode(WIFI_AP) and WiFi.softAPConfig() as described in 2025

LusterWong, 2020-07-21:
> ESP8266 has similar homieiot/homie-esp8266#588
> homieiot/homie-esp8266#637

It's a little strange to do that. How did you find out?

Thank you very much. I'll try it

petrbrouzda, 2020-08-03:
> I'm able to confirm this crash with this snippet (192.168.1.1). I notice that the crash doesn't happen the first time a device is connected to an ssid, but the second. Changing the ssid gets a one shot successful connect -- probably the dhcp lease is attempting to be renewed rather than requested.

@robacarp , it seems like you are right!
Reconnecting to previously used SoftAP name -> ESP restart.
So I've added random suffix to SoftAP's name and the problem gone.

closed 2020-10-17 by stale[bot]

ScaredyCat, 2020-10-17:
Why has this been closed. It's still an issue. Please reopen

dswilkinson73, 2020-11-06:
This is still an issue, I read a comment that putting the "Core Debug Level" to 5 solves the problem. It worked for me but should be still be fixed. I prefer run at level 0.

mlord, 2021-12-17:
The solution is a 100ms (or greater) delay after WiFi.softAP()  before WiFi.softAPConfig() is invoked.  The reason is that the WiFi stack isn't ready for WiFi.softAPConfig() until SYSTEM_EVENT_AP_START has been delivered.  One could write more complex code to wait for SYSTEM_EVENT_AP_START, or instead just go with the 100msec delay:  delay(100);

3tawi, 2022-04-29:
#include <WiFi.h>
#include <WebServer.h>

const char* Apssid = "Esp32";
const char* Appassword = "3tawi-GP";
  IPAddress ip(192, 168, 1, 141);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 255, 0);

void setup() {
    Serial.begin(115200);
    WiFi.softAPConfig(ip, gateway, subnet);
    WiFi.softAP(Apssid, Appassword);
    delay(3000);
    String textip = " IP Address: http://";
           textip += WiFi.softAPIP().toString().c_str();
    Serial.println("");
    Serial.print(textip);
}

void loop() { }

3tawi, 2022-04-29:
IP Address: http://192.168.1.141

ScaredyCat, 2022-05-18:
In setup:

 ```
    WiFi.onEvent(WiFiEvent);
```

Then this gets called on an event. Make whatever calls you need  from the relevant event

```
void WiFiEvent(WiFiEvent_t event){

    switch(event) {

        case SYSTEM_EVENT_AP_START:                           
            break;

         case SYSTEM_EVENT_STA_START:
            break;
            
        case SYSTEM_EVENT_STA_CONNECTED:
            break;
            
        case SYSTEM_EVENT_AP_STA_GOT_IP6:
            break;
            
        case SYSTEM_EVENT_STA_GOT_IP:
            break;
            
        case SYSTEM_EVENT_STA_DISCONNECTED:
            break;
            
        default:
            break;
   }
}
```

Himanshu21358, 2023-03-10:
Sir I am new developer now we are create a ESP8266 as server and serial port receive data and send data successfully. but I have a issue now I am change my Wi-Fi SOFTAP credentials(like SSID, PASWORD) by receive in serial monitor, so please suggest any code ?