espressif/arduino-esp32 #3909: OTA updates from server error 8

Mr-HaleYa, 2020-04-15:
I'm doing an OTA update from an HTTP server using the example.  https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

I have it working now but I had to modify the Update.cpp file because I was getting an error: 8 even tho my magic bytes are starting correctly. 

I deleted https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L218-L221

Now it works perfectly. I'm assuming this is a bug... I found an old issues post on here #325 that describes this problem but it has been closed a long time and it doesn't seem that the issue was fully fixed... 

Is there any way to fix this other than modifying the library files?

Mr-HaleYa, 2020-06-16:
So it looks like no one cares that there's a bug that makes it so that some Arduino devices can't use the Arduino ota update but whatever I guess lol it's a pretty easy fix for the user if they just want to delete the magic byte check (doesn't fix the problem but a quick way around it)

viktak, 2020-08-06:
Same issue here. I removed the lines @Mr-HaleYa suggested and the update process starts and finishes fine.

Mr-HaleYa, 2020-10-06:
> Same issue here. I removed the lines @Mr-HaleYa suggested and the update process starts and finishes fine.

glad it worked for you. just wished someone would actually reply, even if to say this can be fixed (more like they won't since code can always be fixed) then have a useless stale bot close issues they don't want to reply to

viktak, 2020-12-06:
Keeping this issue alive....

viktak, 2021-02-08:
Still not resolved...

star297, 2021-03-18:
You could try something like what I use, just set the FIRMWARE_URL and call updateFIRMWARE() in your program.
Returns true if successful, restart and its done.
Works on HTTP and HTTPS url's. 
I use it to download ota updates that I put on my from Firebase Storage, works every time.

```
#include <Firebase_ESP_Client.h>
#include <Update.h>
#include "HTTPClient.h"

HTTPClient http;

char      FIRMWARE_URL[256] = "firmware update URL";

int updateFIRMWARE()
{ 
  http.begin(FIRMWARE_URL);
  int httpCode = http.GET();
  if (httpCode <= 0) {
    //Error HTTP failed
    return 0;
  }
  int contentLen = http.getSize();
  bool canBegin = Update.begin(contentLen);
  if (!canBegin) {
    // Error, not  enough space for the new binary.
    return 0;
  }
  WiFiClient* client = http.getStreamPtr();
  size_t written = Update.writeStream(*client);
  if (written != contentLen) {
    // Error, wrote partial binary.
    return 0;
  }
  if (!Update.end()) {
    // Error from Update.end()
    return 0;
  }
  if (Update.isFinished()) {
   // Update successfully completed. Reboot to apply update.
   return 1;
  } else {
    // Error from Update.isFinished()
    return 0;
  }
}
```

mazorvanunu, 2021-04-13:
anyone has any idea? i was trying to use both SPIFFS and LITTLEFS alas both of them are COMPLETLY FUNKY!
spiffs has various problems which are quite frightening (writes fail arbitrarily many times, cannot seek at certains time - with no definitive reason,  file size cannot be over 1 megabyte, calling remove seems to corrupt the entire systems are times...)
littlefs has other various problems as well... (files cannot be over 700kb~, writes fail arbitrarily, calling format seems to corrupt the system at times...)

however the worst of this (poorly written) components is the fact that the esp team for some UNKNOWN (and that i simply CANNOT understand) reason try to belittle those "features", trying to get any kind of good documentation comes with nothing, the examples are simply terrible as they usually over complicate things and does not work as properly or as expected at-all.

many libraries and code schemes does not act as expected and one must always try to do it the "esp" way to make it "work".

TBH, i wish i wouldve came across any comment (such as this) warning about many of the issues the esp32 firmware has... much time was lost, and quite frankly any desire to work with this "chip" 

I am honestly sorry to be this harsh, but the least you, the esp dev team, could have done is warn us about all this bizarre, e-relevant behavior of trying to show the ESP is a done-deal product - this was not only the professional thing to do but also the humane and honest one.

I hope this comment helps you understand the struggle (for nothing!) you've put many developers in.
(they dont even respond in their own forum.) 

Most of the (most) broken features surprisingly revolve around REALLY basic and already proven methodology which adds considerably to the frustration (flash memory, OVERWHELMY complex file system with totally broken or even pseudo working features - which is much worse, many ill-formed library implementation and my personal favorite - no good documentation for basic, trivial ways of operation during development) honestly - i feel i've been sold a lie.

ziadalkalza, 2021-05-16:
@star297 Do you have a place where I can learn how to perform ota updates from the firebase storage? I tried looking at the library you included, but I'm not that experienced so I need some help with it.

star297, 2021-05-17:
Create a Firebase account here:
https://firebase.google.com/products/realtime-database

You can look at our updater for some instructions:
This is a more advanced set up if you have many devices in many remote locations.
But there are no firmware files on the links at the moment.
https://esp-updater.netlify.app/docs/firebase

ziadalkalza, 2021-05-17:
@star297 Thank you for the quick reply. Yeah I need it for a product that I'm making, so it's potentially (fingers crossed) going to be hundreds of devices worldwide. Therefore, I need the device to support future updates, so not only will I be able to update the mobile app, but also the esp32 device.

sansillusion, 2021-05-17:
If you are developing a app mabe put a update feature in the app that could
"post" the update from the app.

1 - check for update on server using app.
2 - download update in app.
3 - post update from app.

>

ziadalkalza, 2021-05-17:
Why would I go through all that when I can just do an OTA update from the esp32 straight away. I'll basically tell the esp32 to update by sending a message via BLE (which is what the app uses) when a button is pressed on the mobile app. So that the esp32 can do its job and just fetch the bin file.

sansillusion, 2021-05-17:
You do you, I was just suggesting that since you would not need to store a
certificate in the esp32 for secure server connection.

star297, 2021-05-18:
We found using Firebase Storage the best solution for multiple remote ESP32 device ota updates in our case.
The firmware can be compiled from any global location placed on Firebase Storage and uploaded to a device to any global location providing there's wifi connection.
Devices will check to see if there's firmware available when active, download and update automatically.

We have used ESP8266 with 4Mbyte Flash, but we needed more analog functions.
Plenty of space for SSL even with a very large program.

ziadalkalza, 2021-05-19:
@star297 Yes this was my initial approach, but I can't seem to find a way to:
1. access the bin files from the Firebase storage securely.
2. download the bin file into the OTA partition, so I can then restart the ESP32.
How do you reckon I can go about doing that? I know there is a firebase arduino library but I'm still fairly new to this.

ziadalkalza, 2021-05-19:
Ok I think for a secure _anonymous_ connection to the Firebase storage, all we need is to include the host link and the _secrets_ authentication token in the device firmware. Conversely, if you want to use proper user sign in, you'll need to include the email and password (or log in credential if from other streams like Google or Facebook). 

Is my understanding correct? Or is there something I'm missing regarding this aspect of the code?

closed 2021-08-04 by stale[bot]

Introvertuous, 2023-06-19:
is anyone even maintaining this?

reinerbraun99, 2024-05-06:
> I'm doing an OTA update from an HTTP server using the example. https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino
>
> I have it [...]

Hello, I have the same issue, and I'm trying to delete the lines you attached here, but it seems the source code of Updater.cpp has changed. Could you please give me the exact lines I must delete in Updater.cpp?

Best regards,
Rainier Adrian