espressif/arduino-esp32 #3916: using ESP32 - master mode bluetooth connect to HC-06 - slave mode

j0r9e74, 2020-04-19:
It is possible to use ESP32 - master mode bluetooth connect to HC-06 (with arduino nano) - slave mode?
The usecase is to send/receive some control strings
I tried the different modified BluetoothSerial sources from github and got only two types of results:
    - always saying "Connected succesfully!", no matter what name or address I used; tried also to communicate in this state but without success
    - "Failed to connect" forever
    It is not a matter of HC-06 bt connection as slave because I can use a phone to connect at it and send/receive strings.

mprowe, 2020-04-20:
Hi,
I am not competent to advise on **how** to solve your problem. But it does sound like the same issue I had some time back? Unfortunately, at my age, my memory is not too good! 

However, with the help of the real experts I think I had to modify the BluetoothSerial Library. As I say, I cant remember where or why now. However, if it should be any help, I have attached my project which I did get working to my satisfaction (an OBD reader). I have also attach the BluetoothSerial library. I think it is formally known as v1.0.4. 

But as I said, I can't remember how I updated it. I can only assume I must have done it by hand, as I can't see it in the Arduino Library Manager?
Good luck, and sorry I can't be more help...
Regards, ~Martin
[BluetoothSerial.zip](https://github.com/espressif/arduino-esp32/files/4501631/BluetoothSerial.zip)
[odb2_ESP32_BT.zip](https://github.com/espressif/arduino-esp32/files/4501632/odb2_ESP32_BT.zip)

j0r9e74, 2020-04-20:
Hello,
Already tried 1.0.4 BluetoothSerial from the git and the results is described above.
I will also what have you send and let you know what I've got.
Thank you very much!

j0r9e74, 2020-04-20:
Hello,
The same behavior. I have got:
11:25:03.568 -> Failed to connect. Make sure remote device is available and in range, then restart app.

Maybe I do something wrong?
Should I pair the devices first? How? Any other step that must be done before running the code?
The code is bellow:
`#include` "BluetoothSerial.h"

BluetoothSerial SerialBT;

uint8_t address1[6]  = {0x98, 0xD3, 0x32, 0x20, 0x87, 0xA0}; // HC-06
uint8_t address2[6]  = {0x00, 0x34, 0xDA, 0xDE, 0x9D, 0xE0}; // G5
uint8_t address3[6]  = {0x64, 0x7B, 0xCE, 0x0D, 0xF2, 0x2F}; // Galaxy S10 5G
uint8_t address4[6]  = {0x00, 0x58, 0x56, 0x4A, 0x87, 0xCD}; // 6143 -> boxa mini
String name = "HC-06";
char *pin = "1234"; //<- standard pin would be provided by default
bool connected;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test", true); 
  //SerialBT.setPin(pin);
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  
  // connect(address) is fast (upto 10 secs max), connect(name) is slow (upto 30 secs max) as it needs
  // to resolve name to address first, but it allows to connect to different devices with the same name.
  // Set CoreDebugLevel to Info to view devices bluetooth address and device names
  //connected = SerialBT.connect(name);
  connected = SerialBT.connect(address4);
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
  }
  
  // disconnect() may take upto 10 secs max
  if (SerialBT.disconnect()) {
    Serial.println("Disconnected Succesfully!");
  }
  // this would reconnect to the name(will use address, if resolved) or address used with connect(name/address).
  SerialBT.connect();  
}

j0r9e74, 2020-04-24:
@mprowe  can you give me more details please?
Should I pair the devices first? How? Any other step that must be done before running the code?
I can pair/connect to HC-06 using any phone BT but for ESP32 I receive: 
"Failed to connect. Make sure remote device is available and in range, then restart app"
all the time.

mprowe, 2020-04-25:
@j0r9e74 _"Should I pair the devices first?"_

If you had taken the time to look at my code, you would see that the first thing I did was to remove any previous bonding.
As I have said previously, my one and only project with the ESP32 Bluetooth  interface was a long time ago and I have forgotten anything that I may have learned - apart from the fact that it was harder work than I would be prepared to go through again!
Regards,

brownby, 2020-05-07:
I was having a similar problem to you (not being able to connect to HC-05 modules, and always saying connection was successful even when it wasn't). Basically, there are two problems:

1. There is a bug in the BluetoothSerial library in 1.0.4 where the `waitForConnect()` function will always return true regardless of if a connection is actually successful or not. 

2. The patch to the BluetoothSerial library that added the ability to use the ESP32 as a master device was developed on version 1.0.2, but included in version 1.0.4. Something changed elsewhere between 1.0.2 and 1.0.3 that broke the ability to connect as a master to HC-05 modules with the BluetoothSerial library (and possibly other devices, I only have HC-05s handy so that's all I've tested). 

The fix (for me) was to go down to version 1.0.2, then manually change the BluetoothSerial.cpp file with [this one](https://github.com/espressif/arduino-esp32/blob/f0e2e2a62fe98ec9657f3935c41c88a0fd0e7acd/libraries/BluetoothSerial/src/BluetoothSerial.cpp) (which fixes the successful connection bug above). I suspect this was @mprowe's fix as well, as he was a participant in the discussion on the [pull request](https://github.com/espressif/arduino-esp32/pull/3219) where I finally found the solution, and the BluetoothSerial.cpp he shared has the fix for the `waitForConnect()` bug.

dexop, 2020-05-17:
I have similar problem.
I am trying to connect ESP32 with slave device who has alphanumeric pin (TECNO), instead of numbers (1234).
is it mission impossible or is there some hint?

vanniaz, 2020-06-11:
@brownby 

> 1. There is a bug in the BluetoothSerial library in 1.0.4 where the `waitForConnect()` function will always return true regardless of if a connection is actually successful or not.

I saw your comment after having found the same bug and opened a new issue for this: #4081 
Had I seen it before I would not have opened the issue, hopefully it will serve as a reminder to fix this bug (by the way, the same bug is in the "disconnect" and "isReady" functions)

zjay1995, 2020-06-11:
Hello @dexop and @brownby . I was trying to connect ESP32 to HC05 as you guys doing. Using the SerialMBT example, followed your steps, but I got this error:
'class BluetoothSerial' has no member named 'connect'
Have you encounter this? Thanks a lot!

brownby, 2020-06-12:
Haven't encountered that, no. `connect` is definitely defined in the BluetoothSerial.cpp file I linked. 

I guess check that you properly uninstalled 1.0.4 and downgraded to 1.0.2? I only say that because the [SerialToSerialBTM](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino) example wasn't included until 1.0.4.

tofurky, 2020-07-10:
@zjay1995 i found you also need to replace the .h file with https://raw.githubusercontent.com/espressif/arduino-esp32/f0e2e2a62fe98ec9657f3935c41c88a0fd0e7acd/libraries/BluetoothSerial/src/BluetoothSerial.h

regardless, i still can't get it to connect, even after downgrading to 1.0.2. 1.0.4 crashes with `Stack smashing protect failure!` as has been mentioned in #3219.

fwiw, i'm trying to connect to a UM25C usb multimeter, which apparently has a DX-BT18 module that is similar to the HC-06. using a "DEVKIT1" board.

closed 2020-09-24 by stale[bot]

buehlems, 2020-12-31:
brownby's fix made it work for me. Tx.

InrouteInnovacion, 2021-08-13:
Hi, how can I downgrade from version 1.0.4 to version 1.0.2?

jpoles1, 2022-01-31:
I've been struggling to get this to work, none of the fixes above seem to be doing the trick? Anyone have any updated recs to get this working in 2022?

Bernardo56458, 2022-02-14:
Hi, I'm new to all this world and may be I'm wrong, but after struggling a lot and read a lot of docs and forums, I've finally found an issue. Don't know if this'll be helpful to all of you, but works for me.

I'm working with a Heltec Wifi Kit 32 and trying to connect to a HC-05 in slave mode classical Bluetooth ( several HC-05, in fact ). The module bonds, pairs and link only after a UART Baud rate change. Default HC-05 baudrate is 4800, so I change this to 115200 ( with AT+UART = 115200 in AT mode ) and the example code for Master works like a charm. It was so simple that I'm both happy and angry.

Was tested also with a Nodemcu Esp32s Wifi + Bluetooth 4.2 Iot Wroom, same results, the HC-05 connects.

15:24:43.630 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
15:24:43.630 -> configsip: 0, SPIWP:0xee
15:24:43.630 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:24:43.630 -> mode:DIO, clock div:1
15:24:43.630 -> load:0x3fff0018,len:4
15:24:43.630 -> load:0x3fff001c,len:1100
15:24:43.630 -> load:0x40078000,len:9232
15:24:43.630 -> load:0x40080400,len:6400
15:24:43.630 -> entry 0x400806a8
15:24:44.469 -> ESP32 bluetooth address: 7c:9e:bd:5b:24:32
15:24:44.469 -> Bonded device count: 1
15:24:44.469 -> Found bonded device # 0 -> 00:21:06:be:4e:68
15:24:44.469 -> Removed bonded device # 0
15:24:44.469 -> The device started in master mode, make sure remote BT device is on!
15:24:50.371 -> Connected Succesfully!
15:24:50.371 -> Disconnected Succesfully!