espressif/arduino-esp32 #3928: WIFI Disconnects after MQTT Connects

eoun, 2020-04-22:
Hello

I am trying to connect my ESP32 to my existing HASSIO MQTT Network.

Everything works well until I upgradet my Router from the "magenta connect box" (So bassically my original Router/Modem from the ISP) to a Netgear RAX40.
All my ESP8266 are still working fine.

My Problem:
When I try to connect to the MQTT Server the WIFI disconnects. 
When I only connect the WIFI all is fine.

I had the problem were no Hostname were send to the Router, but I fixed it with `WiFi.config(INADDR_NONE,` INADDR_NONE, `INADDR_NONE);`
as shown in: 
[https://github.com/espressif/arduino-esp32/issues/2537](url)

I have written a test code:

`
#include <WiFi.h>
#include <PubSubClient.h>

const char* ssid = "";
const char* password = ""; 
const char* mqttServer = "192.168.0.24";
const int mqttPort = 1883;

WiFiClient espClient;
PubSubClient client(espClient);

void setup()
{
  Serial.begin(115200);
  WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
  WiFi.setHostname("ESP32");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) 
  {
    Serial.println(".");
    delay(500);
  }

  Serial.println("Connected to the WiFi network");
  Serial.println("IP Adress: ");
  Serial.println(WiFi.localIP());

  client.setServer(mqttServer, mqttPort);
  client.setCallback(callback);
  client.connect("1234"); //<- Problem
}

void callback(char* topic, byte* message, unsigned int length) 
{
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
}

void loop()
{
  if (WiFi.status() != WL_CONNECTED ) 
  { 
    delay(50);
    Serial.println("LostConnection");
  }

  if (!client.connected()) 
  { 
    delay(50);
    Serial.println("LostConnectionMQTT");
  }

}

`

The code behaves folowing:

The wifi connects (normaly)
The MQTT connects (normaly)
  (IF i connect to MQTT Toppics they are comming in)
  (if they are able to go out, I can not say)
after a fews second the WIFI disconnects
repeat

All my ESP8266 have a simular code, and are still working fine.

Any Idee? Do I have something wrong in the Router

niclaslind, 2020-05-19:
This is maybe not a solution for the problem it self, but I think you should use the AsyncMqtt library instead of PubSubClient for the MQTT part.  
This library is adapted for the ESP8266/ESP32. 

https://github.com/marvinroger/async-mqtt-client

bertmelis, 2020-05-19:
Or use the built-in MQTT client?

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/mqtt/mqtt_client.h

ChrSchultz, 2020-07-25:
you forgot in the loop() client.loop();
and have a look in my ESP32-BMP180 sketch:
[ESP32-BMP180.zip](https://github.com/espressif/arduino-esp32/files/4976240/ESP32-BMP180.zip)
this sketch works fine on my ESP, and the term if WIFI fails eg 5 times the ESP reboots via ESP.reboot() function. so all freezes are prevented.

closed 2020-10-09 by stale[bot]