espressif/arduino-esp32 #3931: Can't remove/reinsert SD card in SPI mode, esp32 just hangs

josedpedroso, 2020-04-23:
### Hardware:
Board:							ESP32-cam
Core Installation version:			esp32 boards 1.0.4
IDE name:						Arduino 1.8.12
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 7 x64

### Description:
One part of my current project involves making a web-based file explorer for the SD card in an esp32-cam board. So, I decided to start with the SDWebServer example. It works fine, as long as I don't remove the card while the sketch is running. However, I'd like for that to be possible. Looking at [sd_diskio.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp), I can see that there are timeouts, but for some reason they're not working.

### Sketch:
```cpp

/*
  SDWebServer - Example WebServer with SD Card backend for esp8266

  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the WebServer library for Arduino environment.

  This library is free software; you can redistribute it and... etc etc

*/
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <SPI.h>
#include <SD.h>

#define SDCARD_SCK 14
#define SDCARD_MISO 2
#define SDCARD_MOSI 15
#define SDCARD_SS 13

#define DBG_OUTPUT_PORT Serial

const char* ssid = "********";
const char* password = "********";
const char* host = "esp32sd";

WebServer server(80);

static bool hasSD = false;
File uploadFile;

void returnOK() {
  server.send(200, "text/plain", "");
}

void returnFail(String msg) {
  server.send(500, "text/plain", msg + "\r\n");
}

bool loadFromSdCard(String path) {
  String dataType = "text/plain";
  if (path.endsWith("/")) {
    path += "index.htm";
  }

  if (path.endsWith(".src")) {
    path = path.substring(0, path.lastIndexOf("."));
  } else if (path.endsWith(".htm")) {
    dataType = "text/html";
  } else if (path.endsWith(".css")) {
    dataType = "text/css";
  } else if (path.endsWith(".js")) {
    dataType = "application/javascript";
  } else if (path.endsWith(".png")) {
[... 298 lines cut ...]
			}
			
			return sdWait(_pdrv, 1000);
		}
};

#define new_SwappableSD SwappableSD(FSImplPtr(new VFSImpl()))

SwappableSD sdcard = new_SwappableSD;

bool init_sd() {
	if (sdcard.cardResponding()) {
		return true;
	}
	else {
		return sdcard.begin(SDCARD_SS) && (sdcard.cardType() != CARD_NONE);
	}
}
```

Then, on each request, I called init_sd(), and if it returned false, I just had the request fail. However, this doesn't work because the linker can't find sdWait(), and even if it did find it, I really don't know if this would fix the issue.

### 2nd workaround:

I just do SD.begin() and check SD.cardType() before serving any request, and afterwards, I call SD.end(). This works, but it's kinda ugly, and will probably still fail if I remove the card while it's reading or writing something.

dsyleixa, 2020-04-23:
also different other ESP SD lib functions are still working faulty, e.g. see https://github.com/espressif/arduino-esp32/issues/3603

rel1ct, 2020-05-06:
"Can't remove/reinsert SD card in SPI mode, esp32 just hangs"
Provide a link to the documentation that describes the possibility of these actions.

josedpedroso, 2020-05-06:
Looking at [sd_diskio.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp), there are many calls to sdWait(), which means that the code can deal with timeouts, which should then allow the higher level SD library to fail gracefully, ie, calls would return false.

The log shows 3 "no token received" messages just before hanging. It's possible that it's hanging in an endless loop somewhere after that, instead of just returning a failure.

Since this looks more like a bug, rather than an intentional limitation, instead of going on a possible snipe hunt myself, I thought I'd check with the people who wrote, or at least are familiar with this code. That is, not you.

And it's possible, I provided a workaround, but it's inefficient and it's just covering what looks like a bug.

rel1ct, 2020-05-06:
It would not occur to one sane person to remove a memory card from a working microcontroller.
If you really need it, then study the libraries and add the necessary functions (or fix the ones that do not work correctly).
No one will do this for you.

josedpedroso, 2020-05-06:
I don't really see why it's up to you to decide what should and should not get fixed, or at the very least discussed with the devs here. Go do something useful instead of trolling random issues on github.

rel1ct, 2020-05-07:
OK good luck.

dsyleixa, 2020-05-07:
@josedpedroso 
IMO it's just you  who is trolling - this issue tracker is for reporting library issues in order to get them fixed, most probably by the devs and contributors who have been developing these libs.

atanisoft, 2020-05-07:
@josedpedroso I suspect this usecase has not been implemented in the library itself for much the same reason as @rel1ct alludes to, why should the library support removal of the device it is actively using.. there are a lot of overheads to contend with in supporting removal that are not easily solved, one being closing/invalidating open file handles (not all are/can be tracked by the SD class!)

I would suggest if you need support for hot swapping of SD cards is to treat it similar to how Windows does.. provide a user option to gracefully shutdown usages and then inform the user to swap.

lbernstone, 2020-05-07:
If you have a HARDWARE card detect pin, then it is possible to set an interrupt on the pin.  If you paid $5 for a cheap SD module, you get what you paid for.

josedpedroso, 2020-05-07:
> there are a lot of overheads to contend with in supporting removal that are not easily solved, one being closing/invalidating open file handles (not all are/can be tracked by the SD class!)

That makes sense.

> provide a user option to gracefully shutdown usages and then inform the user to swap.

I considered doing something like that, though for now, my workaround does the trick, and doesn't require user interaction. But it's possible I'll have to do just that.

> If you have a HARDWARE card detect pin, then it is possible to set an interrupt on the pin. If you paid $5 for a cheap SD module, you get what you paid for.

The ESP32-cam doesn't have it, it's actually one of the things I checked while troubleshooting this.

closed 2020-05-07 by josedpedroso