espressif/arduino-esp32 #3933: BasicOTA

cxtal, 2020-04-23:
Hello,

I cannot seem to get BasicOTA to work under ESP32 however BasicOTA works fine with an ESP8266 using the exact same environment (network, cabling, etc.). I have two ESP32 boards available, both being a [DOIT Esp32 DevKit v1](https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html) that I have swapped amongst each other to check for a dud.

Connecting to the serial port and checking the output it seems that there is no crash at all. The only messages I get are "Receive Failed" (from line 55 in the BasicOTA sketch) from the ESP32 and the Arduino IDE reports "Upload failed".

When the OTA update is initiated via the Arduino IDE, some progress is made till the error is reported and the update fails. The behaviour is inconsistent: sometimes the update fails at 4%, sometimes it fails at 37% (fictive numbers) or the likes.

I have the following settings for the Arduino IDE:

ESP32 Dev Module
Upload Speed: 921600
CPU Frequency: 240Mhz (WiFi/BT)
Flash Frequency: 80Mhz
Flash Mode: QIO
Flash Size: 4MB (32MB)
Partition Scheme: Default 4MB with spiffs (1.2MB APP/1.5 SPIFFS)
PSRAM: Disabled

So far I have tried the following to no avail based on various Internet results:

- flash frequency was set to 40Mhz instead of 80Mhz,
- I have tried all the partition schemes except the "No OTA" ones,
- I have modified BasicOTA and tried pinning ArduinoOTA.begin() to either cores with xTaskCreatePinnedToCore,
- holding down the "BOOT" button whilst performing an OTA update...,
- adding delays in loop(),
- swapped USB cables...,
- changed between USB computer ports and an USB charger,
- changed Wifi networks

Nothing seems to work.

Is there a known issue? The open issues search results show some OTA talk but it does not seem to apply nor yield any results.

Thanks!

lbernstone, 2020-04-23:
Are you certain you can upload the firmware using serial?  The most likely cause for this failure is trying to upload a firmware that will not fit in the OTA parititon.  The other common problem is a simple connection issue.  Make sure your connection RSSI is good with the WiFiScan example.

cxtal, 2020-04-23:
Yep, it uploads fine (others seem to upload fine as well and work too - other than BasicOTA I mean). Here is a short video clip demonstrating the issue.

[2020-04-23-18-49-12.zip](https://github.com/espressif/arduino-esp32/files/4523749/2020-04-23-18-49-12.zip)

cxtal, 2020-04-25:
Just a small heads-up. I have downgraded the ESP32 board to the 1.0.3 release and now OTA does seem to work.

It is a little strange because the Arduino IDE does not report that the update was successful however the serial port does clearly show that the update was successful. It does indeed update the sketch and I have switched from BasicOTA to my own project.

closed 2020-07-08 by stale[bot]