espressif/arduino-esp32 #3935: Crash when trying to connect to wifi

MrPivato, 2020-04-23:
### Hardware:
Board:							ESP WROOM 32
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu 18.04

### Description:
I am trying to connect to WiFi via my ESP board. 
Every time the device tries the connection, it crashes and reboot.
I've cleaned my Flash before uploading the sketch.

I've done some research and have some questions:
Is there a possibility of my board being defective? Is there a workaround if so?
Maybe my USB port does not provide enough power?
Is my configuration wrong?

If someone could point what I am doing wrong, I would be thankful

### Sketch:
```cpp
#include <WiFi.h>

const char* ssid     = "Vortyce";
const char* password = "*******";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}
```

### Debug Messages:
```
Connecting to Vortyce
..................................
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d4f1c: 3d196892 3dd03df0 81f03df0
Core 0 register dump:
PC      : 0x400d4f21  PS      : 0x00060633  A0      : 0x800d53c1  A1      : 0x3ffbbfa0  
A2      : 0x3ffbd2bc  A3      : 0x3ffbbfcf  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060620  A7      : 0x00000000  A8      : 0x3ff5f000  A9      : 0x00000000  
A10     : 0x3ffbc0e8  A11     : 0x00060623  A12     : 0x00060623  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d4f21:0x3ffbbfa0 0x400d53be:0x3ffbbfc0 0x400d53cf:0x3ffbbff0 0x400d6379:0x3ffbc010 0x4008a361:0x3ffbc030 0x40088b7d:0x3ffbc050

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

PC: 0x400d4f21: reset_hw_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 116
EXCVADDR: 0x00000000

Decoding stack results
0x400d4f21: reset_hw_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 116
0x400d53be: esp_task_wdt_reset at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 319
0x400d53cf: idle_hook_cb at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 80
0x400d6379: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 51
0x4008a361: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

noorhaq, 2020-04-29:
Did you add the streamid and privatekey in the above code or not? And if yes then I think your issue lies with probably not entering your wifi SSID and password correctly.

MrPivato, 2020-04-29:
I added the streamid and privatekey. I also tried connecting to at least 5 wireless networks, double checking the SSID and the Password each time, no success either.
The problem goes like so:
1. The device tries to connect;
2. It takes a long time (2-3 minutes);
3. Crashes and reset.

But it can scan for networks with no problems, just doesn't connect to them.
Just was wondering if I got a bad board also, maybe could happen, I don't know, no big deal either.

lbernstone, 2020-04-29:
Power issues will typically show up immediately during the WiFi.begin.  You won't get a single dot.  Ensure the problem is not in your loop by deleting the whole function.  Try [wiping your device](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) and reflashing.

MrPivato, 2020-04-30:
I did wipe it clean, but the problem persists. I will try improving the power supply, currently I am using the USB port of my computer, shouldn't it be enough?

> Power issues will typically show up immediately during the WiFi.begin. You won't get a single dot. Ensure the problem is not in your loop by deleting the contents of the function. Try [wiping your device](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) and reflashing.

awyork, 2020-05-02:
> I did wipe it clean, but the problem persists. I will try improving the power supply, currently I am using the USB port of my computer, shouldn't it be enough?

I've seen my ESP32 write "brownout" to the serial port.

MrPivato, 2020-05-04:
Yeah I don't a oscilloscope, only a multi-meter, which didn't helped much in this case, I've read too that I can connect a 10 uF electrolytic capacitor between the EN pin and GND to workaround this issue. Any information if this is true? Thanks.

awyork, 2020-05-04:
I used this is a baseline for my board.  Check out the top right section.   Also notice the top left section.  Each as a 22uF cap on 3V3.  There is a  22nF cap right next to the VCC pin on the ESP32, and another 22uF cap coming out of the voltage regulator. 

The board I did actually has a pair of 100 uF caps and a 100 nF cap.  It's probably overkill.  The board is stable though.  Try to stick to ceramics if you can.  Particularly for the 100nF cap. 

https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf

PS, get you hands on a Rigol DS1054Z, or something similar.  This will save you a lot of headaches when debugging serial comms.  Whatever you get, make sure it can see decode the serial for you and has at least 4 channels.  (Well you only really need 2 channels and a trigger, but you will only see one side of the conversation. i.e. MISO or MOSI but not both)

MrPivato, 2020-05-04:
> I used this is a baseline for my board. Check out the top right section. Also notice the top left section. Each as a 22uF cap on 3V3. There is a 22nF cap right next to the VCC pin on the ESP32, and [...]

Yeah I see now, makes sense!
Thank you very much for the valuable information, and I'll be looking forward to getting a oscilloscope, it sure is a good investment for cutting some time in debugging, again, thanks for everything.

closed 2020-07-17 by stale[bot]