espressif/arduino-esp32 #3936: D-Duino-32 board variant doesn't compile for I2C

srolf, 2020-04-24:
I run into problems while setting up a new Arduino project including a D-Duino-32:  This former working board variant doesn't compile if I2C is used by the Adafruit SSD1306 lib. Has something changed in Wiring or SDK internals? I'm not sure.

`C:\Users\<user>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\variants\d-duino-32/pins_arduino.h:61:33: error: initializer element is not constant`

By changing SCL/SDA assignment in
`C:\Users\<user>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\variants\d-duino-32\pins_arduino.h`
the variant compiles again and works.

`static const uint8_t OLED_SCL = SCL;`
to
`static const uint8_t OLED_SCL = 4;`
and
`static const uint8_t OLED_SDA = SDA;`
to
`static const uint8_t OLED_SDA = 5;`

atanisoft, 2020-04-24:
SCL and SDA are defined in the same pins_arduino.h, it sounds like the Adafruit library or something else is overriding these incorrectly as a #define or similar. Without code to reproduce this it will be very difficult to say why it is failing in your env.

srolf, 2020-04-25:
I'm not sure because how the compiler log looks like. It points to
`error: initializer element is not constant`
`static const uint8_t OLED_SCL = SCL;`
in "...\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\wiring_pulse.c"

To reproduce the build issue, I used from "Example" -> "Adafruit SSD1306" -> "ssd1306_128x64_i2c.ino" and changed
`#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)`
to
`#define OLED_RESET     3 // Reset pin # (or -1 if sharing Arduino reset pin)`
as needed to get the display of the D-duino propery up and running.

For myself the error message related "wiring_pulse.c" looks like the root of the problem. But I'm just a bloody beginner.

atanisoft, 2020-04-25:
This seems like a bug in the library in that it is overriding the defaults set in pins_arduino.h and wiring_pulse.c seems to be unrelated.

closed 2020-07-09 by stale[bot]