espressif/arduino-esp32 #3937: abort() was called at PC 0x40170757 on core 1 ESP32 - CAM

paulgiordano123, 2020-04-24:
I am trying to do a few things in a forever loop as well as keeping track of a timer variable. My loop can be seen below:

    void loop() {
        //Code to send video frames through the websocket to the webserver 
        camera_fb_t *fb = esp_camera_fb_get();
        if(!fb){
           esp_camera_fb_return(FB);
           return;
        }

        if(fb->format != PIXFORMAT_JPEG){
          return;
        }

        client.sendBinary((const char*) fb->buf, fb->len);
        esp_camera_fb_return(FB);

        //Code to check the output of the PIR sensor as well as the timer variable 
        if (digitalRead(4) == 1 && timer > 45){
                  //sendEmail();
                  //Establish Firebase Connection
                  Firebase.begin("https://testapp-", "Uanv47YytD138HyK99F");
                  FirebaseJson json1;
                  timeClient.update();
                  formattedDate = timeClient.getFormattedDate();
                  Serial.println(formattedDate);
                  json1.set("timeDetected",formattedDate);
                  Firebase.pushJSON(firebaseData, "00001", json1);
                  Serial.println("Motion Detected");
                  timer = 0;                     
        }
        Serial.println(timer);
        timer++;
    }

The problem is that although these functions run fine independently when I put them together in this for loop a system crash is caused always after printing out "Motion Detected". Everything else in the if statement works and is completed but for some reason, after finishing the code inside it the system restarts giving me the following error below:

    abort() was called at PC 0x40173a1f on core 1

    Backtrace: 0x4008ef2c:0x3ffb1b90 0x4008f15d:0x3ffb1bb0 0x40173a1f:0x3ffb1bd0 
    0x40173a66:0x3ffb1bf0 0x40173b13:0x3ffb1c10 0x40173b96:0x3ffb1c30 
    0x40173ff5:0x3ffb1c50 0x4017412e:0x3ffb1c70 0x401744e9:0x3ffb1cb0 
    0x40174505:0x3ffb1cd0 0x40174515:0x3ffb1cf0 0x400d6ae5:0x3ffb1d10 
    0x400d6b86:0x3ffb1da0 0x400d5e72:0x3ffb1dd0 0x400d268e:0x3ffb1df0 
    0x400e86c1:0x3ffb1fb0 0x4008b675:0x3ffb1fd0

    Rebooting...
    ets Jun  8 2016 00:22:57

    rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:1
    load:0x3fff0018,len:4
    load:0x3fff001c,len:1216
    ho 0 tail 12 room 4
    load:0x40078000,len:9720
    ho 0 tail 12 room 4
    load:0x40080400,len:6352
    entry 0x400806b8

Is there anything obvious that is making the chip crash every time the if statement is invoked? Any help would be greatly appreciated.

awyork, 2020-04-26:
There should be some more diag info just above your dump.  Turn on debug mode and see if you can get more details in the trace.  Look for something like this.

```
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGe
nericReceive)- assert failed!
                                                                                         
abort() was called at PC 0x40088be7 on core 1
      
```

mickey9801, 2020-05-02:
I've same error but no more diag info "abort() was called at PC 0x40085775 on core 1" is the first line.

lbernstone, 2020-05-02:
https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md

awyork, 2020-05-02:
I've moved all of my Wifi (http, etc) work over to core 0.  This has solved these issues for me.

v-c, 2020-06-18:
> I've same error but no more diag info "abort() was called at PC 0x40085775 on core 1" is the first line.

If you are open to using PlatformIO, see my comments to the issue linked here on how to debug what is causing the abort() on core 1.

cgmckeever, 2020-08-06:
@awyork may I ask how you moved Wifi to core0?

closed 2020-10-20 by stale[bot]