espressif/arduino-esp32 #3939: I2S Problem, Noise on Microphone Output

Younes-SadatNejad, 2020-04-25:
I am working with ESP32 and MH-ET LIVE Omnidirectional Microphone Module (INMP441 ). They communicate through I2S Interface. Just two weeks ago, I could easily connect ESP and get sound data by defining I2S communication.  I have no idea what happened, but now, anything I do, I get a massive noise on top of my sound. All I hear is a buzzing sound, and if I make a very very loud sound, I can hear my sound under all of those noises.
I changed my ESP, my wires, my microphone, and even the PC, but still the same problem.

### Hardware:
Board:							DOIT ESP32 DEV KIT
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					Tried both 1,000,000 and 115200

### My sketch: 
#include <driver/i2s.h>  
#include <soc/i2s_reg.h>
const i2s_port_t I2S_PORT = I2S_NUM_0;
   
void setup() {
 
  Serial.begin(1000000);

  
  esp_err_t err;

  // The I2S config as per the example
  const i2s_config_t i2s_config = {
      .mode = i2s_mode_t(I2S_MODE_MASTER | I2S_MODE_RX), // Receive, not transfer
      .sample_rate = 8000,                         // 8KHz
      .bits_per_sample = I2S_BITS_PER_SAMPLE_32BIT, // could only get it to work with 32bits
      .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT, // although the SEL config should be left, it seems to transmit on right
      .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,     // Interrupt level 1
      .dma_buf_count = 4,                           // number of buffers
      .dma_buf_len = 8                              // 8 samples per buffer (minimum)
  };

  // The pin config as per the setup
  const i2s_pin_config_t pin_config = {
      .bck_io_num = 14,   // BCKL
      .ws_io_num = 15,    // LRCL
      .data_out_num = -1, // not used (only for speakers)
      .data_in_num = 32   // DOUT
  };

  // Configuring the I2S driver and pins.
  // This function must be called before any I2S driver read/write operations.
  err = i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
 // SET_PERI_REG_BITS(I2S_TIMING_REG(1), 0x1, 2, I2S_TX_DSYNC_SW_S);
 

  if (err != ESP_OK) {
    Serial.printf("Failed installing driver: %d\n", err);
    while (true);
  }
  err = i2s_set_pin(I2S_PORT, &pin_config);
  if (err != ESP_OK) {
    Serial.printf("Failed setting pin: %d\n", err);
    while (true);
  }
  Serial.println("I2S driver installed.");

  //  REG_SET_BIT(  I2S_TIMING_REG(1),BIT(9));   /*  #include "soc/i2s_reg.h"   I2S_NUM -> 0 or 1*/
  // REG_SET_BIT( I2S_CONF_REG(1), I2S_RX_MSB_SHIFT);
 
}

void loop() {
   
  int32_t sample = 0;
  int16_t Data=0;
   int bytes_read = i2s_pop_sample(I2S_PORT, (char *)&sample, portMAX_DELAY); // no timeout
    sample = ~(sample >> 14) + 1;
    Data= (int16_t) sample;
    Serial.println(Data);
  if (bytes_read > 0) {
    Serial.println(Data);
  
  }
  }

```
I have also attached the noise I get on serial plot. This is what I get without making any sound. 
Any help?
```

[image]

X-Ryl669, 2020-05-15:
You need to swap the samples you are receiving two by two. So if you have a `buffer[]` full of samples, you'll access by `buffer[pos ^ 1]` instead of `buffer[pos]`, as soon as you do that, you'll get your nice sine showing!

Your code should read:
```
void loop() {

int32_t sample[2] = 0;
int16_t Data=0;
int bytes_read = i2s_pop_sample(I2S_PORT, (char *)&sample[0], portMAX_DELAY); // no timeout
bytes_read = i2s_pop_sample(I2S_PORT, (char *)&sample[1], portMAX_DELAY); // no timeout
sample[0] = ~(sample[0] >> 14) + 1;
sample[1] = ~(sample[1] >> 14) + 1;
Serial.println(sample[1]); // Swap here, 1 not 0
Serial.println(sample[0]);
```

X-Ryl669, 2020-05-15:
Any reason to use 32 bits instead of 16 bits ?
You should display all the sample value (Serial.println(sample)) to check if there is another channel embedded in the higher bits. I'm not 100% sure how the I2S is behaving in 32 bits mode.

Younes-SadatNejad, 2020-05-19:
@X-Ryl669 Thank you so much for your reply. 
It turns out that the microphone I am using  (MH-ET LIVE Omnidirectional Microphone Module I2S Interface INMP441 MEMS) must be soldered on both sides. The noise was due to not soldering the face side (the side that has pin numbers on it). 
Also I believe I2S only works on 32 bits, other bits doesn't give any value out

closed 2020-05-19 by Younes-SadatNejad

rujha, 2020-10-19:
Hi Younes, 

I am also facing the same issue while using INMP441 after 1 week.  It was working fine with I2S communication but now I am getting big noise on top of my sound. Checked the soldering and it seems fine.  Changed ESP32, wires and INMP441 also but it was not successful.  Anything else to look for.

thanks,
Rupak

Younes-SadatNejad, 2020-10-22:
Hello Rupak
Sometimes soldering both sides help, but unfortunately I still have this
issue sample to sample.
Let me know what if you figure it out the solution.

Best
Younes

AlexFerreira1984, 2021-01-08:
Younes, solder a 10k ( around that) between SD and GND. The factory pulldown (100k) is not enough.

Orel-A, 2021-04-05:
> 
> 
> Younes, solder a 10k ( around that) between SD and GND. The factory pulldown (100k) is not enough.

What does it do ?

AlexFerreira1984, 2021-04-05:
Just a theory: as pull up is very week, SD does not go completelly down and hence you have aditional noise. Either way, it solved my noise problem.

EDsteve, 2021-06-17:
I also facing some noise issues. 
Not really sure what you mean with "between SD and GND". Between which pins?

VojtechBartoska, 2022-05-04:
on what version are you running @Huyyuh2000?

cc @PilnyTomas

Huyyuh2000, 2022-05-05:
I have tried ESP32 DEVKIT V1, WeMos wifi & Bluetooth R32 Base ESP32 v1.0.0 and ESP32 Lite v1.0.0
I used Arduino IDE all the time with Flash Frequency: 80Mhz and Upload Speed: "921600"
My recording a code is copy of this source: [link](https://github.com/0015/ThatProject/blob/master/ESP32_MICROPHONE/ESP32_INMP441_RECORDING/ESP32_INMP441_RECORDING.ino)

PilnyTomas, 2022-05-05:
Hi, @Huyyuh2000 
I believe Vojtech was asking about the board core version in Arduino IDE.
You can check the version in Arduino IDE > Tools > Board > Boards Manager > search ESP32,  on the right corner, is the version.
Currently, the newest version is 2.0.3 so if you have an older one, please update a try with the new version.

Huyyuh2000, 2022-05-05:
Hi, I have update to newest version (v2.0.3) but the noise still high. I have tried add extra 10k between SD and GND like [AlexFerreira1984](https://github.com/AlexFerreira1984) but still not working.

PilnyTomas, 2022-05-05:
@Huyyuh2000 ok, can you please create a new issue? This one was closed by the OP.
You can tag me in the new issue to make sure I won't forget :)

jtwaleson, 2024-10-12:
In my case I added a 10k ohm resistor between SD and VDD and it worked, so a pull-up instead of a pull-down.

AhmadaliyevB, 2026-09-11:
HELLO, Is there anyone who has found a solution to the noise?

AlexFerreira1984, 2026-10-02:
Have you tried the 10K resistor? Otherwise try to remove the capacitor or use a smaller one.

renolabs114-glitch, 2026-11-02:
hy did anyone find solution of it

Huyyuh2000, 2026-11-02:
Hi, you can read this: https://github.com/espressif/arduino-esp32/issues/6699. It might help