espressif/arduino-esp32 #3941: ESP32 HTTPClient does not like to talk to ESP8266WebServer

awyork, 2020-04-25:
### Hardware:
Board:							Custom Board
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino and Visual Micro
Flash Frequency:					80
PSRAM enabled:                                    no
Upload Speed:					9Mbit (full speed serial) or OTA
Computer OS:                                        N/A
### Description:
I'm doing a HTTP Get to an ESP8266 IOT Device.  I've seen a similar issue perhaps here:  

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/)

Example sketch for BasicHTTPClient.ino

### Debug Messages:
I think this is the problem.  HTTP Client does not like it when the web server does a RST instead of a ACK to the FIN message.  (Is ESP8266Web Server also broken?  Perhaps.  However, the RST should not cause the ESP32 to not reset.)

[image]

[HTTP] begin...
[V][HTTPClient.cpp:235] beginInternal(): url: http://pumpsaver/data
[D][HTTPClient.cpp:276] beginInternal(): host: pumpsaver port: 80 url: /data
[D][HTTPClient.cpp:1025] connect():  connected to pumpsaver:80
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Type: text/json'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Content-Length: 171'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1123] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 171
[D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 171).
[D][HTTPClient.cpp:370] disconnect(): tcp stop

{"running":false,"pressure":0,"guageconnected":true,"raw":103,"avgTemp":24.21,"index":3,"junction":32.8125,"fault":false,"noprobe":false,"shortgnd":false,"shortvcc":false}
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4015e1fc  PS      : 0x00060430  A0      : 0x800d408b  A1      : 0x3ffb1e90  
A2      : 0x3ffb1f10  A3      : 0x00000000  A4      : 0x000000ab  A5      : 0x3f402750  
A6      : 0x400d20e8  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1e70  
A10     : 0x3ffba7b8  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4015e1fc:0x3ffb1e90 0x400d4088:0x3ffb1eb0 0x400d4159:0x3ffb1ed0 0x400d1b73:0x3ffb1ef0 0x400d7029:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

closed 2020-07-17 by stale[bot]

kaangoksal, 2020-09-04:
I encountered an error very similar to this, my solution was to add a wificlient to http.begin() functionality, 

```

WiFiClient client;
HTTPClient http;

http.begin(client, "http://" + serverBaseUrl + "/api/data/relay1");

```

michellekoeth, 2021-02-09:
> I encountered an error very similar to this, my solution was to add a wificlient to http.begin() functionality,
> 
> ```
> 
> WiFiClient client;
> HTTPClient http;
> 
> http.begin(client, "http://" + serverBaseUrl + "/api/data/relay1");
> ```

This also worked for me. Thank you!