espressif/arduino-esp32 #3946: sizeof not returning size_t

n9wxu, 2020-04-28:
### Hardware:
Board:							WROOM32
Core Installation/update date:		1.12.0
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200?
Computer OS:                                          Windows 10

### Description:

I am trying to use min with 2 values I am expected to be of time size_t.

Unfortunately this provides the following error:
"no instance of overloaded function "min" matches the argument list -- argument types are: (size_t, 
unsigned long long)"

This leads me to believe that sizeof is not returning an size_t.

The bug can be duplicated with the following code:

```
void main(void)
{
    char bigBuffer[256];
    size_t foo = Serial.available();
    length = min(foo,sizeof(bigBuffer));
}

```

n9wxu, 2020-04-28:
This may be an issue with Visual Studio Code C/C++ extension.  I am not seeing the error in a build, only in the IDE.

noorhaq, 2020-04-29:
Try the fix describe here, it will most probably fix your issue:
https://stackoverflow.com/questions/14508022/error-no-matching-function-for-call-to-minlong-unsigned-int-unsigned-int

n9wxu, 2020-04-29:
Maybe...
In my example above, size_t is clearly the type of foo
and sizeof is specified to return a size_t.
My error suggests that the size_t returned by sizeof and the size_t of foo are not the same.
In fact the following works:
`void main(void)
{
   char bigBuffer[256];
   size_t foo = Serial.available();
   size_t length = min(foo,(size_t) sizeof(bigBuffer));

}`
Now, compiling these never produced an error.  I think the fault actually lies in the Visual Studio C IntelliSense making a bad assumption about the type.  size_t is typedef'd differently for Arduino ESP32 than the IntelliSense expects so it flags an error.

closed 2020-04-29 by n9wxu