espressif/arduino-esp32 #3947: I2C produces jumps in the time sample

jgomez109, 2020-04-28:
I am trying to make a PID Controller for a Gimbal 2DOF (Here is the Gimbal I am using: https://i5.walmartimages.com/asr/0c3e2526-d29f-4503-9ffa-41d7e81448e9_1.1e442c2b6975f45507cce1ab828ae056.jpeg?odnWidth=612&odnHeight=612&odnBg=ffffff) . I am meassuere the angle with a MPU6050 via I2C (Wire). To control the brushless motors I am Using the ESP32 who sents pwm signals to a L6234 Driver. The code uses a Hardware Timer Group to achive the time sample for the controller, however, when the motors start to work, the read I2C algorithm sometimes take a lot of time. 

The time sample it is shown below in the first column, and in the second, the I2C read Time. When the motors are working, sometimes the I2C read is too long producing jumps in the time sample. 

2700	1825
2710	1818
2720	1804
2730	1818
2740	1826
2750	1816
2760	1827
2770	1830
2780	1824
2790	1822
2800	1820
2810	1828
2820	1831
2830	1824
3830	1001713
3840	1822
3850	1817
3860	1832
3870	1829
3880	1824
3890	1833
3900	1819
3910	1829
4920	1001568
4930	1826
4940	1756
4950	1822
4960	1821
4970	1824
4980	1815
4990	1827
5000	1826

This do not happend when the motors are turned off. 

Showing up next is the full code I am using. I have used Wire.setTimeout(), yield() but it keep being the same thing.  (The switch case makes a sequence for the motors):

#include <driver/timer.h>
#include <SimpleKalmanFilter.h>
#include <Wire.h> 

#define MPU 0x68
#define A_C 16384.0    
#define frecuenciaPWM 20000 // Frecuencia de trabajo PWM
#define resolucionPWM 8 // Resolución del PWM 
#define canalPWM0 0 // Canal 0 PWM
#define canalPWM1 1 // Canal 1 PWM
#define canalPWM2 2 // Canal 2 PWM
#define canalPWM3 3 // Canal 3 PWM
#define canalPWM4 4 // Canal 4 PWM
#define canalPWM5 5 // Canal 5 PWM
#define EN1_motor1 27  // Pin
#define EN2_motor1 13  // Pin
#define EN3_motor1 25  // Pin
#define IN1_motor1 14  // Pin   
#define IN2_motor1 12  // Pin  
#define IN3_motor1 26  // Pin
#define EN1_motor2 19  // Pin
#define EN2_motor2 23  // Pin
#define EN3_motor2 18  // Pin
#define IN1_motor2 15  // Pin   
#define IN2_motor2 4   // Pin 
#define IN3_motor2 5   // Pin
#define N_SIN 256 // cantidad de muestras de la señal senoidal
// VARIABLES 
byte xDuty_a_motor1, xDuty_b_motor1, xDuty_c_motor1, xDuty_a_motor2, xDuty_b_motor2, xDuty_c_motor2;
[... 391 lines cut ...]
           case 12:{ // ESPERA
              
            const int tEspera = 5000; // Tiempo de espera en ms
              xCountSec++;     
              if ( (tm*xCountSec) >= tEspera){ // ESPERA "tEspera" ms
                xCountSec = 0;
                suich1 = 1;
              }
             
            }break;
            
           }

          Serial.print(xTiempo);   
          Serial.print("\t");
          Serial.println(t2-t1);    

          actualizarPWM(xIndex1 ,xIndex2, maxPWM1,maxPWM2); // actualizarPWM(u1,u2)
}

Here is the electric circuit:

[image]

I think the problem is electric. Someone can help me?

lbernstone, 2020-04-28:
1) Does you get similar time gaps if the motors are not powered?  This will rule out electrical issues.
2) TL;DR your code, but if the motor and I2C functions are independent, then spin out different tasks ([xTaskCreate](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv429xTaskCreateStaticPinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC11StackType_tPC12StaticTask_tK10BaseType_t)) to manage each of them.

jgomez109, 2020-04-29:
> 1. Do you get similar time gaps if the motors are not powered?  This will rule out electrical issues.
> 2. TL;DR your code, but if the motor and I2C functions are independent, then spin out different tasks ([xTaskCreate](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv429xTaskCreateStaticPinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC11StackType_tPC12StaticTask_tK10BaseType_t)) to manage each of them.

Hi, first, thanks for your comment. 
1. The time gaps are normal when the motors are NOT powered. I think is an electric circuit problem (I posted the circuit it above). I tried with RC Filter Circuit but the results keep being the same. The circuit of the RC Filter i used was this:

[image]

It based on this pdf: [](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwww.latticesemi.com%2F-%2Fmedia%2FLatticeSemi%2FDocuments%2FWhitePapers%2FHM%2FImprovingNoiseImmunityforSerialInterface.ashx%3Fdocument_id%3D50728&psig=AOvVaw0hBn0VFvTJEHEfsCWRF00_&ust=1588213783337000&source=images&cd=vfe&ved=0CAMQjB1qFwoTCOjFxMfLjOkCFQAAAAAdAAAAABAN
)

2. I tried but I2C function ("leerIMU") does not read the MPU6050 data when is outside from the Void Loop () {} I do not why. It is the same algorithm above (It works well as I told, when motors are not powered).

closed 2020-07-12 by stale[bot]

hegoku, 2021-07-11:
I have the same issue, can any one help?