espressif/arduino-esp32 #3948: Deallocation outside allocating method causes crash

lokeb, 2020-04-28:
I am experiencing crashes trying to free allocated memory. Here's the message before crash:
``
19:33:01.242 -> CORRUPT HEAP: Bad tail at 0x3ffbaa89. Expected 0xbaad5678 got 0xbaad5600
19:33:01.242 -> assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c"
``
I am trying to deallocate memory allocated inside another function like this:

```
//deallocating function
void getKey() {
  byte* key = crypt->generateKey();
  /*other code */
  free(key);
}

//allocating function in another scope and file
byte* Crypt::generateKey(size_t len) {
  byte* iv = (byte*) malloc(len);
  return iv;
}
```

That's all. What am I doing wrong?

Window 10, Arduino 1.83

### Hardware:
Board:							node32
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

lokeb, 2020-04-29:
I can confirm that it's happening in the same scope when I call a method in a Class like this:
```
if(hex) {
   //Calling free causes the Multi-heap crash as reported.
   free(hex);
    hex = nullptr;
  }

  hex = (char*) malloc(2*len + 1);
  ```
My reported chip is:
`Chip is ESP32D0WDQ6 (revision 1)`

lokeb, 2020-04-29:
It occurs even when I try to `realloc`. To be more clear. It's a class that is initialized once inside the WebServer instance listening for requests. After that, when another request comes in the class is active and resident.

lokeb, 2020-04-29:
More updates (peculiarities):
When the allocated type is `byte` no crashes, but with `char` it crashes.

```
const byte* Crypto::generateKey(size_t len) {
  iv = (byte*) realloc(iv, len);
  return iv;
}

const char* Crypto::toHex(const byte* str, size_t len) {
  hex = (char*) realloc(hex, 2*len + 1);
  return hex;
}
```

if I initialize the two variables with malloc(0), it throws no error in case of `byte` but the following in case of `char`:

```
  iv = (byte*) malloc(0);
  hex = (char*) malloc(0);
```
``
assertion "old_size > 0" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 308, function: heap_caps_realloc
21:53:52.111 -> abort() was called at PC 0x4010faeb on core 1
``

On `realloc` a similar behavior, crash on `char`, okay on `byte`:
``
CORRUPT HEAP: Bad tail at 0x3ffd17f1. Expected 0xbaad5678 got 0xbaad5600
22:10:41.101 -> assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
22:10:41.101 -> abort() was called at PC 0x4010faeb on core 1
``

closed 2020-07-12 by stale[bot]

avidNewb, 2020-12-06:
I don't believe this should be closed.  I'm having the same issue when using free() to free memory allocated in psram as a char. I've had to resort to making the variable a global and reusing it to avoid a memory leak. This brings other issues though and isn't really a solution.

magillus, 2021-02-10:
I run into this too, I had `char *` value that I wanted to pass between Tasks and free up on last one that would use it, this issue was casing crash same as above when trying to release the `char *`.
I worked this around using `std::string`

judepereira, 2021-02-28:
+1 for this. Exact same scenario.

chegewara, 2021-03-01:
It is not a bug. Even if variable is make will malloc it is still local variable and its scope is local. What you have to do is to make it `static`.

https://stackoverflow.com/questions/9423909/return-a-local-pointer-which-points-to-a-static-variable

avidNewb, 2021-03-01:
> It is not a bug. Even if variable is make will malloc it is still local variable and its scope is local. What you have to do is to make it `static`.
> 
> https://stackoverflow.com/questions/9423909/return-a-local-pointer-which-points-to-a-static-variable

Then why does it only crash on char and not byte? The variable is not local to the function.  It was declared elsewhere.  It was only allocated within the function.

chegewara, 2021-03-01:
It is local here, but should be static:

> //allocating function in another scope and file
byte* Crypt::generateKey(size_t len) {
  byte* iv = (byte*) malloc(len);
  return iv;
}

lokeb, 2021-03-01:
I think this problem occurred because of it being in a interrupt call, which is probably justified. 

Regards
Loknath Bharti