espressif/arduino-esp32 #3950: when called a function run for 90 minutes then stop until called again

ilteo85, 2020-04-29:
Hi, when i recive a imput thrugh ESP-NOW i need to run a function for 90 minutes than when 90 minutes are reached function must stop until she's called again. Should i use timer() how does it works?
I need to use a thermistor and a rele that will warm a resistence at 42 C for 90 minutes.

`
void treatment(){
if (timer < 90 minutes) {
 if (temperature<42 Degrees) {
     rele = high;
    }
}
}
`

in need to run it like a loop becouse it must check temperature every 1 second.

Thank you very much :)

lbernstone, 2020-04-29:
Please follow the issue template, otherwise it is impossible to help you.  The most likely problem is you are making a conversion somewhere that you expect to be a long int, but is being cast to int.

ilteo85, 2020-04-29:
Ok i'm sorry..i think i've solved without using timer i just will run a while cicle with a delay of 1000
and run it 5400 times ;) thanks for your kind response.. i will close the issue :)

closed 2020-04-29 by ilteo85