espressif/arduino-esp32 #3954: Stack Trace after HTTP connect (HTTP_HEAD redeclaration) [Resolved]

MrDornback, 2020-04-30:
### Hardware:
Board:  
    Board1:  NodeMCU ESP8266 1.0 
    Board2:  Adafruit Feather Huzzah ESP8266

Core Installation version:			
    system_get_chip_id(): 0x20279
    system_get_sdk_version(): 2.2.2-dev(38a443e)
    system_get_boot_version(): 31

ESP8266 Libraries:  2.6.3

IDE name:						
    IDE1:  Arduino Studio 1.8.13 hourly build
    IDE2:  Arduino Studio 1.9.0 Preview Build

Upload Speed:   115200

Computer OS:    Microsoft Windows 10

### Description:
When using both ESP8266WebServer library and a WiFiManager library file embedded within the SwitchDocLabs OurWeather software, I was receiving a stack trace, which I narrowed down to a re-declaration of HTTP_HEAD between the ESP8266WebServer.h and the WiFiManager.h.
This would prevent any sustained connection to the AP, as well as any connections to the web pages.

The specific software I was using is the OurWeather ESP8266 software, and specifically the WiFiManager.h cpp which are included locally in the files.

I solved this in two ways...  

1)  I updated the WiFiManager.h .cpp files in my app to use HTTP_HEADER in place of HTTP_HEAD, and then I corrected all of the handlers/references/calls.

2)  I removed the ESP8266WebServer library and replaced with the updated ESPWebServer.  *  I have submitted updates to that library relating to the file names and handlers/calls to be uniform.  (The latest version omits the HTTP_HEAD declaration.

Each of these independently with resolve the HTTP_HEAD re-declaration issue.
Together, they provide even greater protection.

### Sketch:  
True with any code using the ESP8266 core, ESP8266Webserver.h .cpp and some other code using ESP8266WebServer library.

//Change the code below by your sketch
Updated code to name HTTP_HEADER.
Remove ESP8266WebServer and replace with updated ESPWebServer

### Software Affected
    ESP8266WebServer library 
    ESPWebServer library 
    OurWeatherPlus v 36 and previous. (or any software utilizing a declaration of HTTP_HEAD

MrDornback, 2020-04-30:
[RESOLVED]  

After making the changes mentioned in the original issue I posted here, it appears and tests to all work.

1)  Download the modifed ESPWebServer library from my 'Updated' link below and install that library
2)  In OurWeather code, change references to ESP8266WebServer to ESPWebServer.

Original:  [https://github.com/esp8266/ESPWebServer](https://github.com/esp8266/ESPWebServer)
Updated:  [https://github.com/MrDornback/ESPWebServer](https://github.com/MrDornback/ESPWebServer)

lbernstone, 2020-04-30:
Why don't you submit this as a PR on the source repo?