espressif/arduino-esp32 #3955: Problem getting right temperature with NTC thermistor

ilteo85, 2020-04-30:
### Hardware:
Board:							ESP32D0WDQ6 (revision 1)
Core Installation version:			unknown
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    unknown
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
Hi, i'm trying to get themperature from a NTC thermistor  (NTCLE100E3103GB0 ) i need it to be quite accurate (less the1 C error)..I've read about ADC non linearity trouble so I've tried to calibrate using this library to get LUT [https://github.com/e-tinkers/esp32-adc-calibrate](url) 
And then the sketch above

But i didn't get good results..it's really far from right temperature misured with same ntc and arduino. How can i get better recult? Do i need to buy an external ADC to solve?
Thank you very much!

`int ThermistorPin;
double adcMax, Vs;

double R1 = 10000.0;   // voltage divider resistor value
double Beta = 3977.0;  // Beta value
double To = 298.15;    // Temperature in Kelvin for 25 degree Celsius
double Ro = 10000.0;   // Resistance of Thermistor at 25 degree Celsius

// The ESP32 lookup table varies from device to device, use the program from 
// https://github.com/e-tinkers/esp32-adc-calibrate
// to generate lookup table for your own ESP32
const float ADC_LUT[4096] PROGMEM = { 0,
7.4000,17.0000,18.8000,20.8000,22.8000,24.8000,26.6000,28.6000,30.4000,32.2000,33.6000,34.8000,36.0000,37.2000,38.6000,
39.8000,41.0000,42.2000,43.8000,44.8000,46.0000,47.2000,48.6000,49.6000,50.6000,51.6000,52.4000,53.4000,54.4000,55.4000,
56.4000,57.2000,58.2000,59.2000,60.2000,61.2000,62.2000,63.0000,64.0000,65.2000,66.6000,67.8000,68.8000,69.8000,71.0000,
72.0000,73.0000,74.2000,75.4000,76.6000,77.8000,78.8000,80.0000,81.0000,82.4000,83.8000,85.0000,86.2000,87.6000,88.8000,
90.0000,91.4000,92.8000,94.0000,95.0000,96.4000,97.2000,98.0000,99.0000,99.8000,100.8000,101.8000,102.6000,103.4000,104.0000,
105.0000,106.0000,106.8000,107.8000,108.8000,109.6000,110.4000,111.0000,112.0000,113.4000,114.8000,116.0000,117.2000,118.8000,120.0000,
121.2000,122.8000,124.0000,125.2000,126.8000,128.0000,129.0000,130.0000,131.0000,132.0000,133.2000,134.4000,135.4000,136.6000,137.8000,
138.8000,139.8000,140.8000,141.8000,143.0000,144.0000,145.0000,146.6000,147.8000,149.0000,150.6000,151.8000,153.0000,154.4000,155.8000,
157.0000,158.2000,159.8000,160.8000,161.4000,162.0000,163.0000,163.8000,164.6000,165.0000,166.0000,166.8000,167.6000,168.2000,169.0000,
169.8000,170.8000,171.2000,172.0000,172.8000,173.8000,174.4000,175.0000,175.8000,177.0000,178.0000,179.2000,180.6000,181.8000,182.8000,
184.0000,185.0000,186.4000,187.8000,188.8000,190.0000,191.0000,192.2000,193.2000,194.2000,195.4000,196.4000,197.4000,198.4000,199.4000,
200.6000,201.6000,202.6000,203.6000,204.8000,205.8000,206.8000,207.8000,208.8000,209.8000,210.8000,211.8000,212.8000,214.0000,215.0000,
216.0000,217.0000,218.0000,219.0000,220.0000,221.2000,222.2000,223.4000,224.4000,225.8000,226.8000,228.0000,229.0000,230.2000,231.4000,
232.8000,233.8000,235.0000,236.0000,237.2000,238.6000,239.8000,241.0000,242.6000,244.0000,245.8000,247.0000,248.8000,250.2000,251.8000,
253.2000,254.8000,256.2000,257.2000,258.2000,259.0000,260.0000,261.0000,262.0000,263.0000,264.0000,265.0000,266.0000,267.0000,267.8000,
268.8000,269.8000,270.8000,271.8000,272.6000,273.0000,273.8000,274.6000,275.0000,275.8000,276.6000,277.0000,277.8000,278.4000,279.0000,
279.8000,280.4000,281.0000,281.8000,282.2000,283.0000,283.8000,284.2000,285.0000,285.8000,286.2000,287.0000,287.8000,288.0000,291.0000,
293.8000,296.4000,299.0000,301.8000,304.2000,305.2000,306.2000,307.2000,308.2000,309.2000,310.0000,311.0000,312.0000,313.0000,314.0000,
315.0000,316.0000,317.0000,318.0000,319.0000,320.0000,321.0000,322.2000,323.6000,324.8000,325.8000,326.8000,328.0000,329.0000,330.0000,
331.4000,332.6000,333.8000,334.8000,335.8000,336.8000,337.8000,338.8000,339.8000,340.6000,341.4000,342.2000,343.2000,344.0000,345.0000,
346.0000,347.0000,347.8000,348.8000,349.8000,350.8000,351.6000,352.4000,353.2000,354.0000,355.0000,356.0000,357.0000,357.8000,358.8000,
359.8000,360.8000,361.6000,362.4000,363.2000,364.0000,365.0000,366.0000,367.0000,367.8000,369.0000,370.4000,371.8000,373.0000,374.8000,
376.0000,377.4000,378.8000,380.0000,381.8000,383.0000,384.4000,386.0000,387.8000,389.0000,390.8000,392.2000,393.8000,395.4000,397.0000,
398.8000,400.0000,401.6000,402.8000,404.0000,405.4000,406.8000,408.0000,409.2000,410.8000,411.8000,413.0000,414.6000,415.8000,417.0000,
418.0000,419.0000,420.2000,421.4000,422.6000,423.8000,424.8000,425.8000,427.0000,428.0000,429.0000,430.2000,431.4000,432.4000,433.0000,
434.0000,434.8000,435.6000,436.2000,437.0000,437.8000,438.8000,439.4000,440.0000,441.0000,441.8000,442.6000,443.2000,444.0000,444.8000,
445.8000,446.4000,447.0000,448.0000,448.8000,449.8000,450.8000,451.6000,452.4000,453.2000,454.0000,455.0000,456.0000,456.8000,457.8000,
458.8000,459.8000,460.6000,461.4000,462.2000,463.0000,464.0000,465.6000,467.0000,468.8000,470.2000,472.0000,473.6000,475.0000,476.8000,
478.2000,479.8000,480.8000,481.8000,482.6000,483.2000,484.0000,485.0000,485.8000,486.8000,487.4000,488.0000,489.0000,489.8000,490.8000,
491.6000,492.4000,493.0000,494.0000,494.8000,495.8000,496.8000,498.2000,499.8000,501.4000,503.0000,504.8000,506.0000,507.8000,509.4000,
511.0000,512.4000,513.8000,514.8000,516.0000,517.0000,518.0000,519.4000,520.6000,521.8000,522.8000,524.0000,525.0000,526.2000,527.6000,
528.6000,529.4000,530.2000,531.0000,532.0000,533.0000,534.0000,534.8000,535.8000,536.8000,537.8000,538.6000,539.4000,540.2000,541.0000,
542.0000,543.0000,544.0000,545.4000,547.0000,548.8000,550.6000,552.2000,554.0000,555.8000,557.6000,559.0000,560.8000,561.8000,562.8000,
563.6000,564.6000,565.6000,566.6000,567.4000,568.4000,569.4000,570.4000,571.2000,572.2000,573.2000,574.2000,575.0000,576.0000,576.8000,
577.4000,578.0000,578.6000,579.0000,579.8000,580.2000,580.8000,581.4000,582.0000,582.8000,583.0000,583.8000,584.2000,585.0000,585.6000,
586.0000,586.8000,587.2000,587.8000,588.4000,589.0000,589.8000,590.0000,590.8000,591.2000,592.0000,593.0000,594.8000,596.6000,598.0000,
599.8000,601.8000,603.2000,605.0000,606.8000,608.2000,609.2000,610.0000,611.0000,612.0000,613.0000,614.0000,615.0000,615.8000,616.8000,
617.8000,618.8000,619.8000,620.6000,621.6000,622.4000,623.4000,624.2000,625.2000,626.2000,627.4000,628.4000,629.4000,630.4000,631.6000,
632.6000,633.6000,634.6000,635.8000,636.8000,637.8000,638.8000,639.8000,640.8000,641.8000,642.8000,643.8000,644.8000,645.8000,646.8000,
647.8000,648.8000,649.8000,650.8000,651.8000,652.8000,653.8000,654.8000,655.8000,657.0000,658.8000,660.2000,662.0000,663.8000,665.2000,
667.0000,668.6000,670.0000,671.8000,673.0000,674.0000,675.4000,676.6000,677.8000,678.8000,680.0000,681.0000,682.0000,683.2000,684.6000,
685.8000,686.8000,688.0000,689.0000,690.2000,691.6000,692.8000,693.8000,695.0000,696.2000,697.6000,698.8000,700.0000,701.0000,702.2000,
[... 240 lines cut ...]

void loop() {
  double Vout, Rt = 0;
  double T, Tc, Tf = 0;

  double adc = 0;
  double somma=0;
  int i = 0;
  while(i<20){
    i=i+1;
    adc = analogRead(ThermistorPin);
    somma = adc + somma;
    delay(10);
  }
    adc=somma/20;
    adc = ADC_LUT[(int)adc];
  Vout = adc * Vs/adcMax;
  Rt = R1 * Vout / (Vs - Vout);

  T = 1/(1/To + log(Rt/Ro)/Beta);    // Temperature in Kelvin
  Tc = T - 273.15;                   // Celsius
  if (Tc > 0) Serial.println(Tc);

  delay(1000);
}`

lbernstone, 2020-04-30:
3rd party libraries are not supported here, but you can use the espressif calibration as discussed in https://github.com/espressif/arduino-esp32/issues/1804.  The code in the idf-release/v4.0 branch has calibration built into analogRead.

ilteo85, 2020-04-30:
Thanks for your quick response.. i'm sorry i didn't understand is this the code i should run?
(I use arduino IDE) 
[https://github.com/espressif/arduino-esp32/issues/1804#issuecomment-475281577](url)
i need to install a library?
do i need to connect some pins?
how it works it gives me a LUT 

last question..you think after calibration i should have quite accurate results?
thank you really much! :)!

closed 2020-07-14 by stale[bot]