espressif/arduino-esp32 #3957: library  seems broken / does not compile

JosvL-prog, 2020-05-01:
Hi,
I have just downloaded (as a .zip) from the master branch,
and have updated my  Arduino IDE  with it,  need it for a TTGO v5  project.

I already had an older version of the SW (dated 10-2-19) but that one had issues with getString() so I decided to try a newer version.
However the version I just downloaded does not compile.
This is the error message :

/home/jos/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiGeneric.h:31:39: fatal error: wifi_provisioning/manager.h: No such file or directory

I checked and indeed wifi_provisioning/manager.h is missing  (and the name / path of it is strange as well).

What is wrong / how can I fix this ????
NEW version of WifiGeneric.h contains this :

#include <esp_err.h>
#include <esp_event_loop.h>
#include <functional>
#include "WiFiType.h"
#include "IPAddress.h"
#include <wifi_provisioning/manager.h>

typedef struct
{
     wifi_prov_cb_event_t event;
     void *event_data;
}wifi_prov_event_t;

OLD version of WifiGeneric.h contains this :

#include <esp_err.h>
#include <esp_event_loop.h>
#include <functional>
#include "WiFiType.h"

typedef void (*WiFiEventCb)(system_event_id_t event);
typedef std::function<void(system_event_id_t event, system_event_info_t info)> WiFiEventFuncCb;
typedef void (*WiFiEventSysCb)(system_event_t *event);

### Hardware:
Board:							TTGO V5 V2.3 2.13 E-INK
IDE name:						Arduino IDE  on OpenSuse Linux 
Flash Frequency:					default
PSRAM enabled:                                     default
Upload Speed:					default
Computer OS:                                        OpenSuse Linux 15.0

lbernstone, 2020-05-01:
Please install using recommended methods.
https://github.com/espressif/arduino-esp32#installation-instructions

JosvL-prog, 2020-05-01:
Ok. Did all that just now. Result is that when opening up the library manager of the Arduino IDE it issues the following message about a 100 times :
Invalid library found in /home/jos/Arduino/libraries/arduino-esp32-master: no headers files (.h) found in /home/jos/Arduino/libraries/arduino-esp32-master

And I checked to see if there is a file wifi_provisioning/manager.h  somewhere now, but still can't find it anywhere , not in the new created folders either.

lbernstone, 2020-05-01:
arduino-esp32 is not a library.  Delete that incorrect folder, then start over and follow the directions.

JosvL-prog, 2020-05-01:
Strike 2...    This time I removed the incorrect folder(s),  removed the existing ESP32 board(s) using the Board Manager of the IDE and reinstalled according to the instructions in "Installation instructions using Arduino IDE Boards Manager".   This resulted in installation of the same 1.0.4 version (dated 10-2-19) as I already had before, but now with date/time stamps of today on the folder 1.0.4 .  So I 'm sure they were renewed.  It did away with the error message around wifi_provisioning/manager.h  as well.  So that is good.    However...  I am also back to the issue I had in the first place,  getString() not working reliable....    On which topic I have found a number of references and tried them all. However none of them really helped. Will continue the search but   maybe I will open a new issue with extensive description of what/where it's going wrong.

closed 2020-05-01 by JosvL-prog