espressif/arduino-esp32 #3959: Why a Watchdog error when doing OTA within an AsyncWebServer callback?

communque, 2020-05-03:
I'm trying to get an ESP32 to run an AsyncWebServer that listens for a curl or ajax request (e.g. curl -d 'req=execOTA') to execute an OTA.

The problem: The OTA triggers a task watchdog, but only when wrapped in the AsyncWebServer callback.

### Hardware:
Board:					ESP32 Wroom
Core Installation versio:	v3.2.3-14-gd3e562907
IDE name:				Arduino IDE 1.8.12
Flash Frequency:			80MHz
PSRAM enabled:			Disabled
Upload Speed:			921600
Computer OS:			Mac OS 10.15.4.

### Description:
Executing an OTA from within a callback triggers a task watchdog.
(Attempting the same OTA immediately on bootup works just fine.)

### Sketch:
(Adapted from [arduino-esp32/libraries/Update/examples/AWS_S3_OTA_Update/](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino))
```
#include <WiFi.h>
#include <Update.h>
#include "AsyncJson.h"
#include "ArduinoJson.h"

AsyncWebServer EspServer(80);

void setup(){
	Serial.begin(115200);
	const char* ClientSsid = "***";
	const char* ClientPwd = "***";
	Serial.print("Connecting to LAN ");
	WiFi.begin(ClientSsid, ClientPwd);
	int CreepConnect;
	CreepConnect=0;
	while (WiFi.status()!=WL_CONNECTED && CreepConnect<30){
		delay(250);
		Serial.print(".");
		CreepConnect++;
	}
	Serial.println(" on SSID " + WiFi.SSID() + " at " + String(WiFi.localIP()));
	
	EspServer.on("*", HTTP_POST, [](AsyncWebServerRequest * Req){
		AsyncWebParameter* keyVal=Req->getParam(0);
		String key=keyVal->name();
		String val=keyVal->value();
		if(key=="req" && val=="execOTA"){
			Serial.println("Updating...");
			Req->send(200, "text/plain", "Updating...");
			//EspServer.end(); //Tested disabling server
			execOTA();
		}else{
			Serial.println("Request ignored");
			Req->send(200, "text/plain", "Request ignored");
		}
	});
	EspServer.begin();
	//execOTA();  //This works just fine -- running immediately after booting up.
}

String getHeaderValue(String header, String headerName){
	return header.substring(strlen(headerName.c_str()));
}

void execOTA(){
	WiFiClient client;
	String host = "192.168.1.10";
	int port = 80;
	String bin = "/ServerOTA.php";
	long contentLength = 0;
	bool isValidContentType = false;
	Serial.println("Connecting to: " + String(host));
	if(client.connect(host.c_str(), port)){
		Serial.println("Fetching Bin: " + String(bin));
		client.print(String("GET ") + bin + " HTTP/1.1\r\n" +
			"Host: " + host + "\r\n" +
			"Cache-Control: no-cache\r\n" +
			"Connection: close\r\n\r\n");
		unsigned long timeout = millis();
		while(client.available()==0){
			if(millis() - timeout > 5000){
				Serial.println("Client Timeout !");
				client.stop();
				return;
			}
		}
		while(client.available()){
			String line = client.readStringUntil('\n');
			line.trim();
			if(!line.length()){
				break;
			}
			if(line.startsWith("HTTP/1.1")) {
				if(line.indexOf("200") < 0){
					Serial.println("Got a non 200 status code from server. Exiting OTA Update.");
					break;
				}
			}
			if(line.startsWith("Content-Length: ")) { // extract headers here Start with content length
				contentLength = atol((getHeaderValue(line, "Content-Length: ")).c_str());
				Serial.println("Got " + String(contentLength) + " bytes from server");
			}
			if(line.startsWith("Content-Type: ")) {// Next, the content type
				String contentType = getHeaderValue(line, "Content-Type: ");
				Serial.println("Got " + contentType + " payload.");
				if(contentType=="application/octet-stream"){
					isValidContentType = true;
				}
			}
			vTaskDelay(10);
		}
	}else{
		Serial.println("Connection to " + String(host) + " failed. Please check your setup");
	}
	Serial.println("contentLength : " + String(contentLength) + ", isValidContentType : " + String(isValidContentType));
	if(contentLength && isValidContentType){
		bool canBegin = Update.begin(contentLength);
		if(canBegin){
			Serial.println("Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!");
			size_t written = Update.writeStream(client);
			if(written==contentLength){
				Serial.println("Written : " + String(written) + " successfully");
			}else{
				Serial.println("Written only : " + String(written) + "/" + String(contentLength) + ". Retry?" );
			}
			if(Update.end()){
				Serial.println("OTA done!");
				if(Update.isFinished()){
					Serial.println("Update successfully completed. Rebooting.");
					ESP.restart();
				}else{
					Serial.println("Update not finished? Something went wrong!");
				}
			}else{
				Serial.println("Error Occurred. Error #: " + String(Update.getError()));
			}
		}else{
			Serial.println("Not enough space to begin OTA");
			client.flush();
		}
	}else{
		Serial.println("There was no content in the response");
		client.flush();
	}
	client.stop();
}

void loop(){
	//Nothing doing
}
```

### Debug Messages:
```
Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
E (18537) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18537) task_wdt:  - async_tcp (CPU 0/1)
E (18537) task_wdt: Tasks currently running:
E (18537) task_wdt: CPU 0: tiT
E (18537) task_wdt: CPU 1: loopTask
E (18537) task_wdt: Aborting.
abort() was called at PC 0x400dc657 on core 0

Backtrace: 0x4008c434:0x3ffbe170 0x4008c665:0x3ffbe190 0x400dc657:0x3ffbe1b0 0x40084771:0x3ffbe1d0 0x4000bfed:0x3ffb3b00 0x40089ce9:0x3ffb3b10 0x4008b9f3:0x3ffb3b30 0x4008c078:0x3ffb3b50 0x40084a14:0x3ffb3b70 0x4008161d:0x3ffb3b90 0x40139cff:0x3ffb3bb0 0x4013a5d5:0x3ffb3bf0 0x400f6f47:0x3ffb3c10 0x4012e805:0x3ffb3c40 0x4012e3f9:0x3ffb3c60 0x401280f9:0x3ffb3c80 0x40128419:0x3ffb3ca0 0x401293be:0x3ffb3cd0 0x40129409:0x3ffb3d10 0x40129439:0x3ffb3d40 0x40125749:0x3ffb3d70 0x401257fb:0x3ffb3da0 0x4011d7ef:0x3ffb3dd0 0x40119d48:0x3ffb3df0 0x40118781:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...
```

lbernstone, 2020-05-03:
Your ExecOTA is way to heavy to run as an "interrupt" in the WebServer.  You should set a flag and handle the OTA in the main loop, or else see how to break it into bite sized pieces in https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino

communque, 2020-05-03:
> lbernstone: see how to break it into bite sized pieces...

I'm all for that, but... I'm not yet 100% sure I'm seeing how the bite-sized pieces are being handled in [AsyncUpdate.ino](https://github.com/lbernstone/asyncUpdate/blob/master/AsyncUpdate.ino)

It looks like Update.write() can be looped as many times as necessary, each 'write' appending the next chunk delivered for the server.  Then Update.end() when done, correct?

closed 2020-07-16 by stale[bot]