espressif/arduino-esp32 #3964: How to change RTC Clock Source ?

fededim, 2020-05-04:
Is it possible to change the RTC clock source from default 150Khz RC oscillator to internal 8Mhz or external 32.768Khz crystal in order to achieve a better accuracy ? In [esp-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html) it is possible by configuring CONFIG_ESP32_RTC_CLK_SRC, but there seems not to be an equivalent in Arduino. Any plan to port it ?

lbernstone, 2020-05-04:
No.  This is compiled into the libraries.  You can [build your own libraries](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) if you wish to change the clock source, but it is likely there are other dependencies on that value.

fededim, 2020-05-04:
Ok but what do you mean by "it is likely there are other dependencies on that value" ? I have seen that for building the Arduino library you have to download the ESP-IDF sources...if I configure the CONFIG_ESP32_RTC_CLK_SRC value in make menuconfig and rebuild everybuild I expect it will work. The only problem I see is that you have to replace manually all your Arduino libraries, just for a single project (and revert back when building other projects).

lbernstone, 2020-05-04:
You are changing one of the primary clock signals. I don't know what all that affects.

fededim, 2020-05-05:
Well it's not the main CPU clock, only for RTC and moreover it is supported and well documented. I'll let you know if it works.

fededim, 2020-05-13:
Managed to compile the libraries. I am testing it right now and I would say that I have very nice results with RTC clock. Thank you for everything, we can close the issue.

closed 2020-05-13 by fededim

fededim, 2020-08-24:
I do not remember correctly which libraries I copied (probably I copied all of them, I do not have now with me the virtual machine to check). As it regards the message from bootloader this should happen when the wiring is not correct or it could be also the capacitance of the GND capacitors. You can find the schematics [here see fededim posts](https://www.esp32.com/viewtopic.php?f=12&t=1175&p=59356&hilit=crystal+schematics#p59356), there is a recommended circuit from espressif even though in a [board (zoom and search for C12)](https://dl.espressif.com/dl/schematics/ESP-WROVER-KIT_SCH-3.pdf) built by them they just put 2 capacitors of 12pf. In my case I had a general 32.768 crystal which came without a datasheet; I tried with 12pf (ceramic capacitors) and it worked. To calculate the capacitance of the 2 capacitors you should read [this](https://blog.adafruit.com/2012/01/24/choosing-the-right-crystal-and-caps-for-your-design/). The [datasheet of your crystal](https://www.mouser.it/datasheet/2/3/ABS07-120-32.768kHz-T-218607.pdf) states it has 6pf of load capacitance so from the equation you should have (let's use C1=C2)

6=C1^2/(2*C1)+3 => C1=6 pf (Cstray = 3pf)
6=C1^2/(2*C1)+5 => C1=2 pf (Cstray = 5pf)

Consider that I am not an hardware engineer (I do not know how to determine the right Cstray value), so I used the 2 values provided. It seems that 10pf is too high, try to lower to 6pf and see if it works.

fabiuz7, 2020-10-01:
Hi, I'm trying to change the RTC config to make use of an external oscillator on ESP32-S2... I succeed in getting the RTC working with the "hello_world" example (plus a "gettime()") with the latest version of ESP-IDF. I tested the firmware on 2 boards: one with the oscillator and the other without the oscillator, and on the latter I get the message "clk: 32 kHz XTAL not found, switching to internal 90 kHz oscillator", so I think everything is fine.

However I would recompile the Arduino core to support this RTC setting. I don't understand what should I do. The instructions provided at this [link](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) doesn't work on Windows 10 (I get error about the files location on `make menuconfig` command). Should I run in powershell, prompt or MSYS2? Which are the requirements? Do I need a specific version of python? I had tried other paths such as the one at [this link](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md), but the command `make menuconfig` doesn't work.

fededim, 2020-10-02:
> However I would recompile the Arduino core to support this RTC setting. I don't understand what should I do. The instructions provided at this [...]

I preferred to switch to a Linux Virtual Machine, I did not recompile the Arduino libraries in Windows. If you want to use Windows you should use [Windows Subsystem for Linux](https://docs.microsoft.com/en-GB/windows/wsl/install-win10), but I did not try. Switching to Linux has an advantage: Arduino's compilation of sketches is faster, notwithstanding the virtual machine.

fabiuz7, 2020-10-02:
Thanks for the hint! Did you compiled for esp32 or esp32-s2?

fededim, 2020-10-02:
esp32

fabiuz7, 2020-10-02:
I had tried to applied [this procedure](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) to compile the core for esp32s2, but the usual command `idf.py set-target esp32s2` doesn't work: it says: 

    CMake Error at /home/fab/esp/esp-idf/tools/cmake/build.cmake:186 (message): Failed to resolve component 'main'.

I had checkout esp32s2 branch of arduino, and ESP-IDF is at the latest commit on master. Does anyone know how to do it?

ebelouet, 2021-04-05:
HI,

Is it possible to compile this libray with special function to use external crysatl ?

me-no-dev, 2021-04-05:
https://github.com/espressif/esp32-arduino-lib-builder is your option.