espressif/arduino-esp32 #3965: IP_EVENT_AP_STAIPASSIGNED passes NULL instead of ip_event_ap_staipassigned_t

aquiot-inc, 2020-05-04:
Using ESP-IDF 4.0:
tcpip_adapter.h has the following structure:

~~~
/** Event structure for IP_EVENT_AP_STAIPASSIGNED event */
typedef struct {
    ip4_addr_t ip; /*!< IP address which was assigned to the station */
} ip_event_ap_staipassigned_t;
~~~

Therefor I expect the event `IP_EVENT_AP_STAIPASSIGNED` to pass `ip_event_ap_staipassigned_t`. However, it passes `NULL` instead.

Will this be fixed? Has been fixed?

JJDS, 2020-06-12:
I have the same problem. I tried to get the IP of the station that connects to the ESP32 Soft AP without sucess.
	if ((event->event_id == SYSTEM_EVENT_AP_STAIPASSIGNED) )
		{
			strcIPInfo = event->event_info.got_ip.ip_info;
			snprintf(sIP, 22, "IP:" IPSTR, IP2STR(&strcIPInfo.ip));
			printf("SYSTEM_EVENT_AP_STAIPASSIGNED STA connected to ESP IP Assigned %s\n", sIP);
		}

I tried several ways with 0.0.0.0  as result.
What I am making the mistake?
Any ideias?

lbernstone, 2020-06-12:
IDF4.0 is not supported here, but had some significant changes to the tcpip adapter.  Make sure the enum for those events has not changed (in which case you are getting the incorrect event).  There were also some changes to ip4_addr_t which might be causing problems (these should make the compiler throw warnings or errors).

JJDS, 2020-06-12:
Thanks for taking your time to advice.
I forgot to say I am using the ESP32 as Soft AP.
I get this:

(20917) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
SYSTEM_EVENT_AP_STAIPASSIGNED STA connected to ESP IP Assigned IP:0.0.0.0

The IP is there, the system says it, but I am not able to get it.
This is a problem since before IDF4.0.
Any how I will check your advice.

JJDS, 2020-06-12:
I got it:

		if ((event->event_id == SYSTEM_EVENT_AP_STAIPASSIGNED) )
		{
			ip4_addr_t ipInfo = event->event_info.ap_staipassigned.ip;
			snprintf(sIP, 22, "IP:" IPSTR, IP2STR(&ipInfo));
			printf("SYSTEM_EVENT_AP_STAIPASSIGNED STA connected to ESP IP Assigned %s\n", sIP);
		}

and now

(15416) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
SYSTEM_EVENT_AP_STAIPASSIGNED STA connected to ESP IP Assigned IP:192.168.4.2

Any how, thanks for taking your time!

closed 2020-08-26 by stale[bot]