espressif/arduino-esp32 #3966: BLEClient::~BLEClient() crashes with double free in BLERemoteService::removeCharacteristics()

jackjansen, 2020-05-04:
### Hardware:
Board:							lolin32, but others as well
Core Installation version:			1.0.4
IDE name:						Platform.io
PSRAM enabled:                                    no

### Description:

Trying to disconnect BLE device after using a connection to set some values.

I have found that creating a `BLEClient()` connection and then later calling delete on it is a sure way of getting a crash: panic() while freeing memory.

Digging down through the BLE stack showed that the double free is in `BLERemoteService::removeCharacteristics()`. It loops over two maps, `m_characteristicMap` and `m_characteristicMapByHandle`, calling `delete` on all the values.

But: the two maps share the same set of values (the characteristics objects), indexed by different keys. To fix, the delete loop should only be applied to m_characteristicMap and the other map should simply be cleared without deleting the values.

### Stack trace

```
0x400929c8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:155
0x40092bf9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:170
0x40113aa3: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x400925cd: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:214 (discriminator 1)
0x40085492: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:268
0x40085a4d: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
0x4000bec7: ?? ??:0
0x4008eea3: vQueueDelete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1809
0x400db907: FreeRTOS::Semaphore::~Semaphore() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/FreeRTOS.cpp:133
0x400d8b93: BLERemoteCharacteristic::~BLERemoteCharacteristic() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteCharacteristic.cpp:52
0x400d92a5: BLERemoteService::removeCharacteristics() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteService.cpp:311 (discriminator 1)
0x400d9692: BLERemoteService::~BLERemoteService() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteService.cpp:39
0x400d96b1: BLERemoteService::~BLERemoteService() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteService.cpp:40
0x400d7962: BLEClient::~BLEClient() at /Users/jack/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLEClient.cpp:60 (discriminator 1)
0x400d46ce: IotsaBLEClientConnection::disconnect() at /Users/jack/src/dis-git/iotsaBLEDimmer/iotsaBLEClientConnection.cpp:46 (discriminator 1)
0x400d2c8e: BLEDimmer::loop() at /Users/jack/src/dis-git/iotsaBLEDimmer/BLEDimmer.cpp:189
```

h2zero, 2020-05-05:
Yes you are correct, that's a bug that was fixed in the original repo as seen [here](https://github.com/h2zero/esp32-snippets/blob/448bc857264177a3b29270c64bfba7d54afd9ab9/cpp_utils/BLERemoteService.cpp#L267). 

You should make a PR for this, or maybe @chegewara will fix it.

yongkimleng, 2020-05-28:
Tried this fix, verified it works.

closed 2020-08-10 by stale[bot]