espressif/arduino-esp32 #3968: Sorted entries in boards.txt

zendes, 2020-05-04:
Hi,

This is more of a request for information and if you would accept a PR for improving this:

Currently all entries in boards.txt are sorted in the order that they were added, which makes it hard to find the specific device you have. It would be more user friendly if these entries were sorted alphabetically. Are the entries sorted this way on purpose, or would it be ok if I send a PR to sort them alphabetically (keeping the Espressif ones on top)?

Thanks,
Dennis

lbernstone, 2020-05-04:
Not worth it, and esp32 dev module needs to be at the top.  If you want to write a python script to generate it from json, then it will be easy to alphabetize :smile:

zendes, 2020-05-05:
any pointers to where that json file you talk about would be?

also: worth it is subjective. if you're willing to accept a PR I'm happy to put in the effort to sort it, and then either people follow the instructions in the file to keep it sorter or when things get out of hand we sort it again. just because we don't have automation to keep it perfect doesn't mean we can't make incremental improvements. :-)

lbernstone, 2020-05-05:
The json was just a suggestion, since that would make it very easy to modify.  The script used in esp8266 is [here](https://github.com/esp8266/Arduino/blob/master/tools/boards.txt.py)

zendes, 2020-05-05:
got it. sounds like a good improvement, but not a hard requirement. boards.txt doesn't seem to be modified all that often, so doing this manually once will still benefit a lot of users.

I've created a pull request https://github.com/espressif/arduino-esp32/pull/3969 with the manually sorted file. It's not as nice as the script you mentioned, but it's there now and it'll help people find their board easier.

zendes, 2020-05-10:
Ping. Any update/thoughts on the pill request I posted?

closed 2020-07-24 by stale[bot]