espressif/arduino-esp32 #3970: Can't advertise data with setCompleteServices and setServiceData at the same time

martimalek, 2020-05-05:
### Hardware:
Board:							OLIMEX ESP32-EVB
Core Installation version:			1.0.4
IDE name:						VSCode (Also tried Arduino IDE)
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:

Hello, I'm having an issue when trying to advertise service data. My goal is to advertise a **complete list of 128-bit service uuid** and some **service data** at the same time. But currently i can't get them to coexist, meaning if one is advertised the other isn't.

I don't understand why this behaviour is happening. I can for example set the **service data** and also some flag with `advertisementData.setFlags` but i can't do it with the **service data** and the list of uuids.

I think that this _bug/error/strange behaviour_ is not related to my hardware at all.

### Sketch:
This is the relevant part of the sketch
```cpp

BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
BLEAdvertisementData advertisementData = BLEAdvertisementData();
BLEUUID bleUUID = BLEUUID::fromString(SERVICE_UUID);

advertisementData.setCompleteServices(bleUUID); // This sets the list of uuids
advertisementData.setServiceData(bleUUID, "Some data"); // This sets the service data
// WARNING!!! If the above two lines are used at the same time then only one of the
// two takes effect

pAdvertising->setAdvertisementData(advertisementData);
pAdvertising->setScanResponse(true);
pAdvertising->start();
```

### Debug Messages:
There is no Debug Message because nothing "breaks", it just doesn't advertise what it should.

### Any help is much appreciated

chegewara, 2020-05-06:
You have to remember that both, advertising data and scan response data cant exceed 31 bytes each.
If you can test with nRF connect with empty advertising, you can see how many bytes is used by some included data like flags and tx power. Then you can start adding your data like service UUID, then service data. Remember that service UUID is using 2 + 16  bytes and service data 2 + data length.

martimalek, 2020-05-06:
You are absolutely right, I was using 128bit UUIDs and was even advertising some unnecessary data. I made some adjustments and now I can advertise what I need. 
Thank you for your reminder, I'm closing this now.

closed 2020-05-06 by martimalek