espressif/arduino-esp32 #3971: Build as component fails

PaulMartinsen, 2020-05-05:
How does the build system know which paths to add as includes? In particular `wifi_provisioning/manager.h` is missing from the include list (although it is present in `components/wifi_provisioning/include`)

I have been following the instructions at [esp-idf_component.md](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) to build the library but the build fails at files that include `wifi_provisioning/manager.h`. This seems to be a [recent addition](https://github.com/espressif/arduino-esp32/commit/2c9b648502f1cfdbe2e9a1ef9c26582718633830).

It looks like the include for `components\wifi_provisioning\include` is missing from the build command. Here's the offending build step, reformated for clarity. 
```FAILED: esp-idf/components/CMakeFiles/idf_component_components.dir/libraries/WiFiClientSecure/src/ssl_client.cpp.obj
ccache C:\Programming\Languages\ESP-IDF\Tools\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" 
-I../components/variants/esp32 
-I../components/cores/esp32 
-I../components/libraries/ArduinoOTA/src 
-I../components/libraries/AsyncUDP/src 
-I../components/libraries/AzureIoT/src 
-I../components/libraries/BLE/src 
-I../components/libraries/BluetoothSerial/src 
-I../components/libraries/DNSServer/src 
-I../components/libraries/EEPROM/src 
-I../components/libraries/ESP32/src 
-I../components/libraries/ESPmDNS/src 
-I../components/libraries/FFat/src 
-I../components/libraries/FS/src 
-I../components/libraries/HTTPClient/src 
-I../components/libraries/HTTPUpdate/src 
-I../components/libraries/NetBIOS/src 
-I../components/libraries/Preferences/src 
-I../components/libraries/SD_MMC/src 
-I../components/libraries/SD/src 
-I../components/libraries/SimpleBLE/src 
-I../components/libraries/SPIFFS/src 
-I../components/libraries/SPI/src 
-I../components/libraries/Ticker/src 
-I../components/libraries/Update/src 
-I../components/libraries/WebServer/src 
-I../components/libraries/WiFiClientSecure/src 
-I../components/libraries/WiFi/src 
-I../components/libraries/Wire/src 
-I../components/cores/esp32/libb64 
-Iconfig 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/spi_flash/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/mbedtls/port/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/mbedtls/mbedtls/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/esp32/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/driver/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/esp_ringbuf/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/tcpip_adapter/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/include/apps 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/include/apps/sntp 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/lwip/src/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/port/esp32/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/port/esp32/include/arch 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/lwip/include_compat 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/vfs/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/esp_event/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/log/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/efuse/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/efuse/esp32/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/newlib/platform_include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/newlib/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/freertos/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/app_trace/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/heap/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/soc/esp32/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/soc/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/mdns/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/console 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/ethernet/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/esp_adc_cal/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/fatfs/src 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/wear_levelling/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/sdmmc/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/nvs_flash/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/app_update/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/bootloader_support/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/spiffs/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/openssl/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/bt/include 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/bt/bluedroid/api/include/api 
-IC:/Programming/Languages/ESP-IDF/Release-3-3-2/components/bt/common/osi/include 
-mlongcalls   -mfix-esp32-psram-cache-issue -mlongcalls -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -fstack-protector -ggdb -std=gnu++11 -fno-rtti -fexceptions -MD -MT esp-idf/components/CMakeFiles/idf_component_components.dir/libraries/WiFiClientSecure/src/ssl_client.cpp.obj -MF esp-idf\components\CMakeFiles\idf_component_components.dir\libraries\WiFiClientSecure\src\ssl_client.cpp.obj.d -o esp-idf/components/CMakeFiles/idf_component_components.dir/libraries/WiFiClientSecure/src/ssl_client.cpp.obj -c ../components/libraries/WiFiClientSecure/src/ssl_client.cpp

In file included from ../components/libraries/WiFi/src/WiFiSTA.h:28:0,
                 from ../components/libraries/WiFi/src/WiFi.h:32,
                 from ../components/libraries/WiFiClientSecure/src/ssl_client.cpp:20:
../components/libraries/WiFi/src/WiFiGeneric.h:31:39: fatal error: wifi_provisioning/manager.h: No such file or directory
```

Ben79543, 2020-05-08:
Hi Paul, 
Did you find a way? 
Fyi: [https://github.com/espressif/arduino-esp32/issues/1531](url)

Ben79543, 2020-05-08:
I tried a dirty hack and copied /wifi_provisioning in the wifi src directory. 
That error is gone. But I then get another one. 

`C:/esp/blink/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
C:/esp/blink/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
`
I found what I thought was a solution(?) for this one:  https://github.com/espressif/arduino-esp32/issues/3760

But then I get a consequence of that fix too because it is lwip that fails next.
 
`
C:/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:769:17: error: incompatible types when assigning to type 'ip_addr_t {aka struct ip_addr}' from type 'const ip_addr_t * {aka const struct ip_addr *}'
         dns->ip = dns_getserver(type);
                 ^
`

`C:/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:769:17: error: incompatible types when assigning to type 'ip_addr_t {aka struct ip_addr}' from type 'const ip_addr_t * {aka const struct ip_addr *}'
         dns->ip = dns_getserver(type);
`

Ben79543, 2020-05-08:
I tried upgrading esp-idf to 3.2.
Now it compiles without errors after all the hacks above....
Now need to try in arduino... With the .a in the right place...

Ben79543, 2020-05-08:
... and that does not compile in arduino. :-(

`Linking everything together...
"C:\\Users\\etawy\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\etawy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/lib" "-LC:\\Users\\etawy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\sketch\\Test_ble_proximity.ino.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLE2902.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLE2904.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEAddress.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEAdvertisedDevice.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEAdvertising.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEBeacon.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLECharacteristic.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLECharacteristicMap.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEClient.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEDescriptor.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEDescriptorMap.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEDevice.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEEddystoneTLM.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEEddystoneURL.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEExceptions.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEHIDDevice.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLERemoteCharacteristic.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLERemoteDescriptor.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLERemoteService.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEScan.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLESecurity.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEServer.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEService.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEServiceMap.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEUUID.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEUtils.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\BLEValue.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\FreeRTOS.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\BLE\\GeneralUtils.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\ETH.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFi.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiAP.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiClient.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiGeneric.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiMulti.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiSTA.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiScan.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiServer.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\libraries\\WiFi\\WiFiUdp.cpp.o" "C:\\Users\\etawy\\AppData\\Local\\Temp\\arduino_build_785938\\core\\core.a" -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lw [... cut ...]
C:\Users\etawy\AppData\Local\Temp\arduino_build_785938\core\core.a(esp32-hal-misc.c.o):(.literal.initArduino+0x20): undefined reference to `esp_partition_erase_range'

C:\Users\etawy\AppData\Local\Temp\arduino_build_785938\core\core.a(esp32-hal-misc.c.o): In function `initArduino':

C:\Users\etawy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-misc.c:202: undefined reference to `esp_log_level_set'

C:\Users\etawy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal-misc.c:205: undefined reference to `esp_partition_find_first'`

and many more....

Ben79543, 2020-05-08:
humm looks like one issue is persisting:
`CC build/tcpip_adapter/tcpip_adapter_lwip.o
C:/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c: In function 'tcpip_adapter_get_dns_info':
C:/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:770:28: error: incompatible types when assigning to type 'ip4_addr_t {aka struct ip4_addr}' from type 'const ip_addr_t * {aka const struct ip_addr *}'
         dns->ip.u_addr.ip4 = dns_getserver(type);
`

PaulMartinsen, 2020-05-14:
Just coming back to this after being distracted by another project. Thanks so much for all your work Ben. I really appreciate it. 

I followed along but got stuck when I tried to copy `wifi_provisioning/manager.h` from tagged branch release/v3.2: it wasn't there! I realize I was on the wrong branch as I write this. Should have been release/v3.2.3. 

Somewhere in there I went down the rabbit hole trying to figure out how the make files were working. I found if I add `wifi_provisioning` to line 208 of [CMakeLists.txt](https://github.com/espressif/arduino-esp32/blob/master/CMakeLists.txt) then the wifi provisioning path gets added to the include list so I don't have to copy across the files anymore. So that line becomes:
`set(COMPONENT_REQUIRES spi_flash mbedtls mdns ethernet esp_adc_cal wifi_provisioning)`

That looks like the key to telling the build system which components in the ESP32's `components` folder are required for the build. The ESP32 documentation talks about it in [Preset Component Variables](https://docs.espressif.com/projects/esp-idf/en/v3.3/api-guides/build-system-cmake.html#preset-component-variables).

Then with your fixes for the return type of `dns_getserver` in `libraries/WiFi/src/ETH.cpp` and `libraries/WiFi/src/WiFiSTA.cpp` I was able to build against ESP-IDF release 3.3.2. 

I turned the `hello_world_main.c` into an Arduino sketch with `setup()` and `loop()` functions and am able to build that with `idf.py build`. I haven't gotten as far as moving the library files into the Arduino directories and building with the Arduino IDE. I'll try to test that this weekend but I expect to get the same result you did. Once I verify that works, I'll try to figure out exactly which `esp-idf` commit the `arduino-esp32` library is using, update this issue & consider doing a PR. 

Thanks again for your help Ben. I was really stuck on this.

PaulMartinsen, 2020-05-16:
Turns out the change to the return type of `dns_getserver` is not in the esp-idf branch referenced in this commit:
[1977370e6fc069e93ffd8818798fbfda27ae7d99](https://github.com/espressif/arduino-esp32/commit/1977370e6fc069e93ffd8818798fbfda27ae7d99)
IDF release/v3.3 (#3339)
* IDF release/v3.3 46b12a560
* fix build
* IDF release/v3.3 367c3c09c

Which refers to [this esp-idf commit](https://github.com/espressif/esp-idf/commit/367c3c09ccd13130f1b02547fe2ba98958400e7d). Which is not the latest commit on the `release/v3.3` branch. 

So provided you check out the [specific esp-idf commit](https://github.com/espressif/esp-idf/commit/367c3c09ccd13130f1b02547fe2ba98958400e7d), [arduino-esp32](https://github.com/espressif/arduino-esp32/commit/49b76649f13acdb8abaf795015d2740bc5f6b340) (the latest as of 2020-05-16) will build provided `wifi_provisioning` is added to `CMake.txt` as described above.

PaulMartinsen, 2020-05-16:
I created #4003 to implement this, with thanks to @Ben79543 for all his help.

ryanzmi, 2020-05-22:
Urr  when is this fix going to merge into master? 
I'm using arduino-esp32 origin/master and IDF release/v3.3 367c3c09c
idf.py build did fail at the file Paul indicated.
After this wifi_provisioning is added to CMake, this works again.
Which I think it's a good fix.

me-no-dev, 2020-05-22:
you guys want to try the latest version? the `esp32s2` branch compiles against IDF master. There is also a 4.0 branch that compiles with IDF 4.0.

Ben79543, 2020-05-22:
hi @me-no-dev 
I actually tried to compile arduino as a component (branch idf-release/v4.0) on idf v4.
I described it here: https://esp32.com/viewtopic.php?f=13&t=15564

I got it to compile indeed with idf.py. Then I copied the .a librairies back to arduino.  Which was my goal from start (to get low power ble working). 
A simple blink sketch would compile, but another simple sketch with BLE commands would not. 

Not sure if it has to do with the compile librairies or how I copied them back to arduino...

me-no-dev, 2020-05-22:
your best bet to compile the libs is to use the lib-builder :) it's what compiles the libs for Arduino currently (and even creates the pull request for the new libs) https://github.com/espressif/esp32-arduino-lib-builder

me-no-dev, 2020-05-22:
uhmm lib builder 4.0 branch that is

ryanzmi, 2020-05-22:
Hi @me-no-dev 
Just curious, is there any significance open defect/ known bugs to be resolve on the latest IDF v4.0 branch?
If not I would like to give it a try 👍

me-no-dev, 2020-05-22:
I can not comment on bugs in IDF, the ones in Arduino do not change with those versions. Would change going 4.2

lbernstone, 2020-05-22:
ESP-IDF is [thataway](https://github.com/espressif/esp-idf/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc).  768 open issues.  I would assume some of them affect IDF 4.0

ryanzmi, 2020-05-26:
@me-no-dev  So finally, I successfully ported my project to the arduino-esp32 v4.0 branch that compiles with IDF 4.0.
I was having un-solvable issue with bt_ble related header file not found, so I give 4.0 a try.
My originally concern is I don't know if the arduino-esp32 v4.0 branch's porting progress / or have been tested (or not). But seem fine with my application.

Actual used:
`
arduino-esp32:
commit d2761a9eab94a906fe393e21e54fc53ee55ff4f5 (idf-release/v4.0)
`
`
esp-idf:
commit 4c81978a3e2220674a432a588292a4c860eef27b (release/v4.0)
`

I do like the new idf.py build, it's much faster than the old GNU make method.
But I gotta learn the new CMake Syntax and try to make the 3rd party arduino library compile & included via CMakeList.txt..

Couple things might be useful to know for other users to compile the 4.0 branch successfully: 
Some default setting need to change if copy from the hello_world idf example. 
the [https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](url) might need to update too with new CMake.
Under menuconfig:
1. Seems like the Arduino Auto-start setup() and loop() didn't work. 
Compiler have issue finding the setup() and loop().
But the another way works.
`
extern "C" void app_main()
`
2. Need to Enable some TLS Key exchange method.
mbedTLS -> TLS Key Exchange Methods -> Enable Pre-shared-key cipher & Enable PSK based ciphersuite 
3. I also changed the following in menuconfig to make my project work. 
Increase Flash size to 4MB (i'm using M5StickC)
Custom Partition Table for OTA
increase ESP_MAIN_TASK_STACK_SIZE to 8192
increase FREERTOS_HZ=1000  (RTOS tick rate)

In my short testing, the basic functionality works with libraries
```
M5StickC
ArduinoJson
AutoConnect (one compiler error need to fix one line)
efont
PageBuilder
pubsubclient (MQTT Client)
```
Thanks for the hard work!

me-no-dev, 2020-05-26:
you have the sdkconfig used to build Arduino in the tools/sdk folder. Copy that to your hello_world and then change only what you might need (disable BT or else). Instructions will soon update when we move to 4.2 (skipping 4.0 and 4.1)

PaulMartinsen, 2020-05-27:
Hi @me-no-dev, 

I had another go at this today and found I was able to build the esp32s2 branch with the esp-idf/master branch without any problems as you suggested. 

Next I tried version 4. Unfortunately I wasn't able to build arduino-esp32/idf-release/v4.0 with the esp32-idf/release/v.0 branch though. I think I hit the same linker problem as @ryanzmi, specifically:
`[1185/1186] Linking CXX executable HelloWorld.elf
FAILED: HelloWorld.elf
cmd.exe /C "cd . && C:\Programming\Languages\ESP-IDF\Tools\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\HelloWorld.elf.rsp  -o HelloWorld.elf  && cd ."
c:/programming/languages/esp-idf/tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/components/libcomponents.a(main.cpp.obj):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
c:/programming/languages/esp-idf/tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/components/libcomponents.a(main.cpp.obj):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
c:/programming/languages/esp-idf/tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/components/libcomponents.a(main.cpp.obj): in function `loopTask(void*)':
c:\users\paul\projects\n8i\esp32\arduinolibs\build/../components/cores/esp32/main.cpp:14: undefined reference to `setup()'
c:/programming/languages/esp-idf/tools/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:\users\paul\projects\n8i\esp32\arduinolibs\build/../components/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

I have these files in [main.zip](https://github.com/espressif/arduino-esp32/files/4686769/main.zip). The rest is just the `arduino-esp32` repository checked out into the `components` folder. 

I also copied `sdkconfig` from `tools/sdk`. I am using `idf.py build` to start the compilation. Unfortunately I haven't had a lot of luck getting the Arduino lib builder working on Windows. But the same program did build with the esp32s2 and esp32-idf branch, so I'm not sure that's the problem. 

I can see that `HelloWorld.cpp` is complied into `build/esp-idf/libmain.a` and that `loop` and `setup` are present in that file (rather `__Z5setupv` `__Z4loopv` with name mangling. I just can't figure out why the linker isn't picking them up. 

Any thoughts what might have gone wrong here?

me-no-dev, 2020-05-27:
first let's keep this closed, you are asking for things that are in progress. 4.0 we support through Make and not through IDF.py.

closed 2020-05-27 by me-no-dev