espressif/arduino-esp32 #3975: WiFi.begin(...,...) does not execute if called within Ticker callback

olonsoft, 2020-05-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.12.1
IDE name:						Platform.io
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I don't know if this is the expected behavior.
I wanted to use a ticker to run some functions but my program stops at WiFi.begin().
So I made a small program to test that functionality. It's the simple `esp32wifi_events` just added a scanNetworks and ticker. The similar code runs ok in ESP8266.

### Sketch: 
```cpp
#include <Arduino.h>
#include <Ticker.h>
#include <WiFi.h>

Ticker ticker;

const char* ssid = "mySSID";
const char* password = "myPASSWORD";

void doConnect() {
  Serial.println("Do Connect");
  WiFi.begin(ssid, password);
  Serial.println();
  Serial.println("Wait for WiFi... ");
}

void WiFiEvent(WiFiEvent_t event) {
  Serial.printf("[WiFi-event] event: %d\n", event);

  switch (event) {
    case SYSTEM_EVENT_WIFI_READY:
      Serial.println("WiFi interface ready");
      break;
    case SYSTEM_EVENT_SCAN_DONE:
      Serial.println("Completed scan for access points");
      ticker.once(1, doConnect);  // <--- added ticker here
      break;
    case SYSTEM_EVENT_STA_START:
      Serial.println("WiFi client started");
      break;
    case SYSTEM_EVENT_STA_STOP:
      Serial.println("WiFi clients stopped");
      break;
    case SYSTEM_EVENT_STA_CONNECTED:
      Serial.println("Connected to access point");
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("Disconnected from WiFi access point");
      break;
    case SYSTEM_EVENT_STA_GOT_IP:
      Serial.print("Obtained IP address: ");
      Serial.println(WiFi.localIP());
      break;
    case SYSTEM_EVENT_STA_LOST_IP:
      Serial.println("Lost IP address and IP address is reset to 0");
      break;
    default:
      break;
  }
}

void setup() {
  Serial.begin(115200);
  // delete old config
  WiFi.disconnect(true);
  delay(1000);
  WiFi.scanNetworks(true, true);
  WiFi.onEvent(WiFiEvent);
}

void loop() {
  delay(1000);
}
```

### Debug Messages:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[WiFi-event] event: 1
Completed scan for access points
Do Connect
```

lbernstone, 2020-05-07:
Ticker callbacks should be treated as interrupts, so no interaction with hardware, or any processing that takes significant time.  Set a flag or semaphore in the ticker interrupt and handle the processing in the main loop.

olonsoft, 2020-05-08:
> Ticker callbacks should be treated as interrupts, so no interaction with hardware, or any processing that takes significant time. Set a flag or semaphore in the ticker interrupt and handle the processing in the main loop.

The Ticker examples have interaction with hardware using digitalRead and digitalWrite.

closed 2020-07-21 by stale[bot]