espressif/arduino-esp32 #3976: OTA update error : esp_image: image at 0x150000 has invalid magic byte after flash enctiption

sboicu, 2020-05-07:
Hi.

I have enabled flash encryption on my esp32 board in release mode.
Everything is working fine. I have also a script to have OTA updates. I compile everything with Arduino IDE.

Unfortunately at the end of the upload of the new .BIN file I get those errors :
esp_image: image at 0x150000 has invalid magic byte

E (91901) esp_image: image at 0x150000 has invalid magic byte
E (91902) boot_comm: image has invalid chip ID, expected at least 0, found 717
E (91904) boot_comm: image has invalid chip revision, expected at least 1, found 224
Could Not Activate The Firmware

Before flash encryption, everything was working just fine. Updates over OTA were working very well. After encryption not.

What I also did is to program another board with ESP IDF native ota example, and set the upload new bin file to the one I am trying to update on my ota http application. I flash board with native ota, and then native ota uploads the new code, but when I use my own ota page is not working.

What would be the cause?

Maybe it helps, I am using async wifi libraries.

Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: fc:f5:c4:4a:5a:cc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

robertpoll, 2020-05-07:
The current Arduino OTA doesn’t support encrypted flash. There are two issues 1) the bootloader is pre-compiled and doesn’t have encryption support. You’ve worked around this one though by programming with ESP-IDF. For the second issue see [my PR](https://github.com/espressif/arduino-esp32/pull/3898) for details of the problem and a proposed fix. Let me know how you get on.

sboicu, 2020-05-07:
> The current Arduino OTA doesn’t support encrypted flash. There are two issues 1) the bootloader is pre-compiled and doesn’t have encryption support. You’ve worked around this one though by [...]

Thanks for your reply. I am not quite sure I understand what I need to do.
Do I have to update the libraries for Arduino OTA?

sboicu, 2020-05-07:
1.I have updated: Esp.cpp, Esp.h, Update.h and Update.cpp as per instructions.

2.I exported my .ino project in the compiled bin with Arduino IDE

3.went to my app website and upload the new bin file. Unfortunately, I have the same error.

I still have the same result
E (81486) esp_image: image at 0x150000 has invalid magic byte
E (81486) boot_comm: image has invalid chip ID, expected at least 0, found 717
E (81487) boot_comm: image has invalid chip revision, expected at least 1, found 224
Could Not Activate The Firmware

sboicu, 2020-05-07:
Ok. this fixed my issue.

I have uploaded the new code to a new board, with flash encryption as well.

Everything is working fine now. My OTA is also working.

Thank you very much for your great support.

closed 2020-07-20 by stale[bot]

naturalmechanics, 2020-09-30:
Hello, I have done this, and edited the files Esp.cpp, Esp.h, Updater.cpp, and Update.h

However, now I have this error: 

```
Inteding to start update
21:35:34.306 -> [D][Updater.cpp:130] begin(): OTA Partition: app1
21:35:34.505 -> Escritos : 17995 successfully
21:35:34.505 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:35:34.505 -> Core 1 register dump:
21:35:34.505 -> PC      : 0x4000c2e0  PS      : 0x00060930  A0      : 0x80087b10  A1      : 0x3ffb1bf0  
21:35:34.505 -> A2      : 0x3ffb1c10  A3      : 0x00000000  A4      : 0x00000010  A5      : 0x3ffb1c10  
21:35:34.538 -> A6      : 0x3ffb0019  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00150000  
21:35:34.538 -> A10     : 0x3ffbd49c  A11     : 0x0000464b  A12     : 0x3ffc1a00  A13     : 0x3ffc1d38  
21:35:34.538 -> A14     : 0x3ffc1ee4  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
21:35:34.571 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
21:35:34.571 -> 
21:35:34.571 -> Backtrace: 0x4000c2e0:0x3ffb1bf0 0x40087b0d:0x3ffb1c00 0x4013b339:0x3ffb1c50 0x400db99e:0x3ffb1c70 0x400d94b2:0x3ffb1c90 0x400d965e:0x3ffb1cb0 0x400d9809:0x3ffb1cd0 0x400d1761:0x3ffb1d10 0x400d3925:0x3ffb1d60 0x400d6639:0x3ffb1da0 0x400d6843:0x3ffb1e60 0x400d6d86:0x3ffb1f20 0x400d6db6:0x3ffb1f40 0x400d6dff:0x3ffb1f60 0x400d6e87:0x3ffb1f80 0x400dccab:0x3ffb1fb0 0x400896f9:0x3ffb1fd0
21:35:34.604 -> 
21:35:34.604 -> Rebooting...
21:35:34.604 -> ets Jun  8 2016 00:22:57

```

Using the backtrace decoder, i find : 
```
0x40087b0d: spi_flash_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 393
0x4013b339: esp_partition_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 274
0x400db99e: EspClass::partitionWrite(esp_partition_t const*, unsigned int, unsigned int*, unsigned int) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Esp.cpp line 203
0x400d94b2: UpdateClass::_enablePartition(esp_partition_t const*) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d965e: UpdateClass::_verifyEnd() at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d9809: UpdateClass::end(bool) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d1761: performUpdate(Stream&, unsigned int) at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 2004 (discriminator 14)
0x400d3925: updateFromFS() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1977
0x400d6639: check_OTA() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1799
0x400d6843: reset_GSM() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 516
0x400d6d86: reactivate_cycle() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1953
0x400d6db6: activate_wakeupProtocol(int) at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1929
0x400d6dff: differentiate_Wakeup() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1912
0x400d6e87: setup() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1653
0x400dccab: loopTask(void*) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x400896f9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

Going to ` EspClass::partitionWrite(esp_partition_t const*, unsigned int, unsigned int*, unsigned int) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Esp.cpp line 203`, I find that the code is : 

```
  md5.calculate();
    result = md5.toString();
    return result;
}

```
It does not call  partitionWrite. PartitionWrite is in line 317. 

I am trying to download the OTA file via GSM modem. The relevant code is here: 

 ```
uint32_t contentLength = knownFileSize;
		 
		 File file = SPIFFS.open("/update.bin", FILE_APPEND);
		 
		 while (client.available())
		 {
			 String line = client.readStringUntil('\n');
			 line.trim();
			 line.toLowerCase();
			 if (line.startsWith("content-length:"))
			 {
				 contentLength = line.substring(line.lastIndexOf(':') + 1).toInt();
[... 77 lines cut ...]
		 }
		 if (Update.end()) {																							Serial.println("OTA finish!");
			 if (Update.isFinished())			 {																		Serial.println("Ota exited ... rebotting!");
				 ESP.restart();
			 }
			 else
			 {
				 Serial.println("Ota no terminó? Algo salió mal!");
			 }
		 }
		 else
		 {
			 Serial.println("Ocurrió Error #: " + String(Update.getError()));
		 }
	 }
	 else
	 {
		 Serial.println("Sin espacio suficiente para hacer OTA");
	 }
 }
```

sorry for the tab formatting, It works perfectly on kate.

The target file is a *.ino file, exported as compressed binary via arduino 1.8.11 IDE. I wonder what I wrong. Thank you.

LusterWong, 2021-02-23:
> Ok. this fixed my issue.
> 
> I have uploaded the new code to a new board, with flash encryption as well.
> 
> Everything is working fine now. My OTA is also working.
> 
> Thank you very much for your great support.
Hello, I also use Arduino development and Flash encryption, also encountered OTA errors
###
E (130773) esp_image: image at 0x1f0000 has invalid magic byte
E (130774) boot_comm: image has invalid chip ID, expected at least 0, found 51630
E (130775) boot_comm: image has invalid chip revision, expected at least 1, found 36
###
How to solve this problem?

Dvl-s, 2021-03-25:
> Hello, I also use Arduino development and Flash encryption, also encountered OTA errors
> `E (130773) esp_image: image at 0x1f0000 has invalid magic byte`
> `E (130774) boot_comm: image has invalid chip [...]

Hii Have you solved encryption + OTA + Arduino IDE ?