espressif/arduino-esp32 #3978: Anthony

AnthonyJParker, 2020-05-08:
I have the Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. issue with my ESP32 Dev:
Can anyone help with determining the source of the problem, please

My sketch reads a setup.js file for the wifi credentials, then uploads to my server BME280 data....

Serial output................
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
SD Card Type: SDHC
SD Card Size: 14804MB
Listing directory: /
  DIR : /System Volume Information
  FILE: /config.txt  SIZE: 38
  FILE: /Temp.txt  SIZE: 34
  FILE: /foo.txt  SIZE: 7
  FILE: /setup.js  SIZE: 482
  FILE: /test.txt  SIZE: 1048576
Total space: 14796MB
Used space: 1MB
setup.js file size: 482
Deserialization succeeded
number of Wi-Fi networks: 3
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060130  A0      : 0x8012fe4c  A1      : 0x3ffb1840  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1c00  
A10     : 0x3ffb8e34  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x3ffb8fd4  
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffb1840 0x4012fe49:0x3ffb1850 0x401373c2:0x3ffb1b60 0x401373fe:0x3ffb1bf0 0x400ded1a:0x3ffb1c30 0x400d3331:0x3ffb1cd0 0x400d34cd:0x3ffb1f70 0x400dffe3:0x3ffb1fb0 0x4008956d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

Decoding stack results
0x4012fe49: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x401373c2: _vsnprintf_r at ../../../.././newlib/libc/stdio/vsnprintf.c line 72
0x401373fe: vsnprintf at ../../../.././newlib/libc/stdio/vsnprintf.c line 41
0x400ded1a: Print::printf(char const*, ...) at C:\Users\Anthony\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 55
0x400d3331: ReadSetupFile(char const*) at D:\Arduino_1.8.10\Mysketchs\ESP32_website_bme280V3/ESP32_website_bme280V3.ino line 258
0x400d34cd: setup() at D:\Arduino_1.8.10\Mysketchs\ESP32_website_bme280V3/ESP32_website_bme280V3.ino line 136
0x400dffe3: loopTask(void*) at C:\Users\Anthony\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x4008956d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

### Hardware:
Board:							ESP32 DevkitV1
Core Installation version:			?
IDE name:						Arduino IDE 1.8.12
Flash Frequency:					80Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					921600?
Computer OS:                                        Windows 10

### Description:
Reboots over and over

Thank you for any help you can give...

     const char* filename1 = "/setup.js"; 

    // *****************************************************************************
    // void ReadSetupFile(const char* filename[20]) (myFile)
    // *****************************************************************************
      void ReadSetupFile(const char* filename){
      //read network credentials from SD card
      myFile = SD.open(filename);//"/setup.js");
      if (!myFile) {
        Serial.println("setup.js file doesn't exist. Save it on SD card and restart.");
        return;
      }
      Serial.printf("setup.js file size: %d\r\n", myFile.size());
    
      //Arduino JSON uses a preallocated memory pool to store the JsonObject tree, this is done by the StaticJsonBuffer. You can use ArduinoJson Assistant to compute the exact buffer size, but for this example 200 is enough.
      const size_t capacity = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(1) + 3 * JSON_OBJECT_SIZE(2) + 130;
      //DynamicJsonDocument doc(capacity);
      StaticJsonDocument<512> doc;
    
      char* json = new char [myFile.size() + 1];
      myFile.read((uint8_t*)json, myFile.size());
      json[myFile.size()] = 0;  //make a CString
      myFile.close();
    
      //const char* json = "{\"WifiNetwork\":[{\"ssid\":\"iiNet7F****\",\"password\":\"7E3CK5FQJTA****\"},{\"ssid\":\"officenet\",\"password\":\"admin1\"},{\"ssid\":\"iPhone(Johny)\",\"password\":\"12345\"}]}";
      DeserializationError  err = deserializeJson(doc, json);
      switch (err.code())
      {
        case DeserializationError::Ok:
          Serial.print(F("Deserialization succeeded"));
          break;
        case DeserializationError::InvalidInput:
          Serial.print(F("Invalid input!"));
          break;
        case DeserializationError::NoMemory:
          Serial.print(F("Not enough memory"));
          break;
        default:
          Serial.print(F("Deserialization failed"));
          break;
      }
    
      JsonArray WifiNetwork = doc["WifiNetwork"];

      int wifi_net_no = WifiNetwork.size();
      Serial.println();
      Serial.printf("number of Wi-Fi networks: %d\r\n", wifi_net_no);
    
      if (wifi_net_no > 10) {
        Serial.println("too many WiFi networks on SD card, 10 max");
        wifi_net_no = 10;
      }
      for (int i = 0; i < wifi_net_no; i++) {
        String ssid = WifiNetwork[i]["ssid"];
        String password = WifiNetwork[i]["password"];
        const char* sensorLocation = WifiNetwork[i]["sensorLocation"];          // added Station Location***************
        strcpy(sensLocatArray[i],sensorLocation);                               // added Station Location***************
        
        wifiMulti.addAP(ssid.c_str(), password.c_str());
    
        //Serial.printf("WiFi %d: SSID: \"%s\" ; PASSWORD: \"%s\" ; LOCATION: \"%s\"\r\n", i, ssid.c_str(), password.c_str(), sensorLocation);
        Serial.printf("LOCATION: \"%s\"\r\n", i, sensorLocation);
      }
      Serial.print("Device location is: ");
      Serial.println(sensLocatArray[0]);
     }

ONLYstcm, 2020-05-09:
In `Serial.printf("LOCATION: \"%s\"\r\n", i, sensorLocation);` you pass two arguements `i` and `sensorLocation` but you have a single `delimiter - %s`. Remoe `i`

AnthonyJParker, 2020-05-11:
Hello ONLYstcm,
Thank you for your reply.
I have solved the issue by removing the const char* sensorLocation......and a few other bits...
It now reads the "/setup.js" and loads multiple wifi credentials and corresponding sensor locations...

    // Replace with your network credentials
    typedef struct {                        // struct for the network details
      char ssid[30];
      char password[30];         
      char sensorLocation[30];              // struct for the Station details, Name/Location
    } WifiNetwork;
    
    struct WifiConfig {                     // struct for the networks available
      WifiNetwork net[10];
    };
    
    WifiConfig wifi_conf;
    int wifi_net_no = 0;
    const char *filename1 = "/setup.js";           // NOTE: SD card "SETUP.JS" filename
    void ReadSetupFile(const char *filename,WifiConfig &wifi_conf );

    String sensorName = "BME280";
    String sensorLocation = "Fafa Island";
    String sensorHome = "deviceSpot";                   // for the http POST data to SQL
    char sensLocatArray[5][40];                         // added Station SSID, PSWRD and Location array
    char sensSSIDArray[5][40];
    char sensPSWRDArray[5][40];

     // etc..........
    
    // *****************************************************************************
    // void ReadSetupFile(const char* filename, WifiConfig &wifi_conf)
    // *****************************************************************************
         void ReadSetupFile(const char* filename, WifiConfig &wifi_conf){
          //read network credentials from SD card
           myFile = SD.open(filename);                    //"/setup.js"
           if (!myFile) {
            Serial.println("setup.js file doesn't exist. Save it on SD card and restart.");
            return;
          }
          Serial.printf("setup.js file size: %d\r\n", myFile.size());
    
      //Arduino JSON uses a preallocated memory pool to store the JsonObject tree, this is done by the StaticJsonBuffer. You can use ArduinoJson Assistant to compute the exact buffer size, but for this example 200 is enough.
      const size_t capacity = JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(1) + 3 * JSON_OBJECT_SIZE(2) + 130;
      //DynamicJsonDocument doc(capacity);
      StaticJsonDocument<512> doc;

      DeserializationError  err = deserializeJson(doc, myFile);
      switch (err.code())
      {
        case DeserializationError::Ok:
          Serial.print(F("Deserialization succeeded"));
          break;
        case DeserializationError::InvalidInput:
          Serial.print(F("Invalid input!"));
          break;
        case DeserializationError::NoMemory:
          Serial.print(F("Not enough memory"));
          break;
        default:
          Serial.print(F("Deserialization failed"));
          break;
      }
    
      JsonArray WifiNetwork = doc["WifiNetwork"];

      int wifi_net_no = WifiNetwork.size();
      Serial.println();
      Serial.printf("number of Wi-Fi networks: %d\r\n", wifi_net_no);
    
      if (wifi_net_no > 5) {
        Serial.println("too many WiFi networks on SD card, 5 max");
        wifi_net_no = 5;
      }
      for (int i = 0; i < wifi_net_no; i++) {
        String ssid = WifiNetwork[i]["ssid"];
        String password = WifiNetwork[i]["password"];
        strcpy(sensSSIDArray[i],WifiNetwork[i]["ssid"]);
        strcpy(sensPSWRDArray[i],WifiNetwork[i]["password"]);
        strcpy(sensLocatArray[i],WifiNetwork[i]["sensorLocation"]);                               // added Station Location***************
        
        wifiMulti.addAP(ssid.c_str(), password.c_str());
    
        Serial.printf("WiFi %d: SSID: \"%s\" ; PASSWORD: \"%s\" ; LOCATION: \"%s\"\r\n", i, ssid.c_str(), password.c_str(), sensorLocation.c_str());
      }
      myFile.close();
   }

closed 2020-05-11 by AnthonyJParker