espressif/arduino-esp32 #3983: camera_pins.h How can I download this file? Anyone suggest me please I need this for my project??? I am trying to do download but I can't download it why ?

killer4097, 2020-05-09:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2020-05-09:
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer

killer4097, 2020-05-09:
Hi,
How can I Download this file??

killer4097, 2020-05-09:
Hi,

How can I download it? There is no option for download. Can u provide this
code file in an email “camera_pins.h”

ONLYstcm, 2020-05-09:
@lbernstone has already linked the file. I saw you created 2 new issues for the same "problem". Go to the link click on "camera_pins.h" then click on the raw button
[image]. This will open the file then right click and save the file with "save as"

killer4097, 2020-05-10:
Hi, I follow your instruction but when I did It shows zip doesn't contain a library. 

can you please provide me a full zip file for camera_pins.h so I can download and run the program directly.

closed 2020-07-23 by stale[bot]