espressif/arduino-esp32 #3986: Device hangs after 3 hours

skaterdude, 2020-05-10:
### Hardware:
Board:							ESP32S Node MCU Clone
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					?
PSRAM enabled:                                    ?
Upload Speed:					115200
Computer OS:                                        Windows 7

### Description:
I've written a controller for a process that can take several hours to complete.  As part of this controller, I provided a status web page that can be requested.  The main loop checks for a request, prints a notation that a request was received, provides HTML via client.print, then continues.  It ran fine for about 3 hours, and then everything just stopped.  No web response, no controller action (solid-state relay indicators did not change, etc.).
In researching the issue, I see several references to a hang after 3 hours and 20 minutes across several platforms,  Examples here:
https://github.com/esphome/issues/issues/1196
https://github.com/knolleary/pubsubclient/issues/624
One makes reference to this being fixed in 1.11.1.  I note that this platform uses 1.0.4, though I don't know enough to say if these are the same versioning systems.  Another mentions the problem goind away by nacking up to 1.0.0 or 1.0.2.  I also found this reference to a fix in the esp_mqtt_destroy_config in December:
https://github.com/espressif/esp-idf/issues/4433

Is there any plan to move ahead to a later version of the core for Arduino IDE?
Thanks

lbernstone, 2020-05-10:
CSB.  What is your question?  Is an upgrade going to fix conflicts and errors in your code? Time is pretty completely 64bit in arduino and esp-idf, so check that your code isn't rolling over somewhere.

atanisoft, 2020-05-10:
@skaterdude the version reference of 1.11.1 is from platformio which would map to v1.0.4 which is the same as used in Arduino IDE *IF* you have that version selected.

This issue is not very descriptive or following the issue template so we can't really help you fix a bug in your code...

skaterdude, 2020-05-10:
The version mapping answers the question about whether or not the Arduino IDE is the latest.  I am using the latest version shown in the "update" list in the board management page.  Thanks

I completely agree that an upgrade isn't going to fix a problem inherent in my code, but the problem sounded similar enough to the issues I saw in other projects that I thought it worth asking (I didn't see another way of asking this other than to log an issue).  I'm concerned that there is a problem in the underlying libraries, as noted in the referenced issues.  I'll take another look at my code.  I make it a practice to use long variables for anything involving time or counters, which should be good for something like 49 days.  But perhaps I missed something.

If you're satisfied that there are no issues, please feel free to close this.  If others run into this issue, they'll see the closed issue and perhaps re-open it.

closed 2020-07-24 by stale[bot]

mailparikshit, 2021-04-01:
i am also facing same hang issue,
i am using enc_28j60 ether shield
and connection hangs in few minutes..
have to power off esp32 and than ON to resume task..