espressif/arduino-esp32 #3988: httpupdate always Fails on Arduino IDE (modified default example)

MG-Tawfeek, 2020-05-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE?
Flash Frequency:					80Mhz
Computer OS:                                        Ubuntu

### Description:
Describe your problem here

Can't get the smallest Sketch in Arduino IDE for OTA update (from internet) to work 
Always Errors 
In Arduino IDE => Connection refused 
HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused

in PlatformIO => The famous Stream Read Timeout ! 

Using IDF with the same link I'm putting below , OTA works , It's just Arduino Core 

```cpp

#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

WiFiClient client;

static void InitWifi()
{
  Serial.println("Connecting...");
  WiFi.begin("wifi ssid", "wifi pass");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  //hasWifi = true;
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup() {
Serial.begin(115200);
InitWifi();
}

void loop() {

 t_httpUpdate_return ret = httpUpdate.update(client, "http://mymodbus.weebly.com/uploads/1/3/2/1/132175186/firmware.bin");
    // Or:
    //t_httpUpdate_return ret = httpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
}

usane1, 2020-06-04:
Same problem here.
Probably related to the HTTPClient issue mentioned in #4039 and refers to #3347 , there is a workaround to HTTPClient, but i don't know if that can solve the update too.

usane1, 2020-06-04:
The workaround is working with the update too.

MG-Tawfeek, 2020-06-04:
Thank you for your reply 
for some reason , as reported by many other users .. if you use a VPS , all goes fine 
but with a shared hosting (cheap or free one) ., it usually fails . 
I'll call this a FIX for the issue 
Thanks again for your reply

closed 2020-06-04 by MG-Tawfeek