espressif/arduino-esp32 #3989: OTA update from arduino-esp32 to ESP-IDF fails and crashes

maikeriva, 2020-05-11:
### Hardware:
Board:						Custom ESP32WROOM32D board
Core Installation/update date:		V1.0.3
IDE name:					VSCode + Arduino extension

### Description:
I am attempting to update a device with a firmware based on arduino-esp32 (1.0.3) to a new firmware based on ESP-IDF v4.1. Both firmwares work properly when flashed individually, as well as when performing OTA updates within the same firmware families.

However, when performing an OTA update from the arduino-esp32 firmware to the new ESP-IDF one, the device crashes while initializing the app from its partition. You can find the traceback below.

While studying the issue, I observed the crash to happen during the SPI flash initialization. I attempted to enable the legacy SPI flash driver implementation through menuconfig, yet I still couldn't get the device to complete the initialization (now it hangs rather than crashing, and the WDT intervenes).

I assume this is most likely due to incompatible bootloaders, yet I could not find any information over the bootloader used in arduino-esp32, not even by looking at the code.

Could maybe the devs point me to the right direction? If I know the bootloader version I could consider whether to rebase the new firmware on a lower IDF version, or opt for an intermediate update to rewrite the bootloader.

### Debug Messages:
Freshly flashed new IDF firmware
```
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6972
load:0x40078000,len:13072
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (28) boot: ESP-IDF v4.1-beta1-112-g4fe18a925-dirty 2nd stage bootloader
I (28) boot: compile time 21:17:00
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  1 app0             OTA app          00 10 00010000 00250000
I (85) boot:  2 app1             OTA app          00 11 00260000 00250000
I (93) boot:  3 nvs              WiFi data        01 02 004b0000 00080000
I (100) boot:  4 spiffs           Unknown data     01 82 00530000 002d0000
I (108) boot: End of partition table
I (112) boot_comm: chip revision: 1, min. application chip revision: 0
I (119) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2f020 (192544) map
I (202) esp_image: segment 1: paddr=0x0003f048 vaddr=0x3ffbdb60 size=0x00fd0 (  4048) load
I (203) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0xacd9c (707996) map
0x400d0020: _stext at ??:?

I (478) esp_image: segment 3: paddr=0x000ecdc4 vaddr=0x3ffbeb30 size=0x031ec ( 12780) load
I (483) esp_image: segment 4: paddr=0x000effb8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/micheleriva/esp-idf/components/freertos/xtensa_vectors.S:1778

I (485) esp_image: segment 5: paddr=0x000f03c4 vaddr=0x40080404 size=0x18550 ( 99664) load
I (537) esp_image: segment 6: paddr=0x0010891c vaddr=0x400c0000 size=0x00064 (   100) load
I (550) boot: Loaded app from partition at offset 0x10000
I (550) boot: Disabling RNG early entropy source...
I (551) cpu_start: Pro cpu up.
I (554) cpu_start: Application information:
I (559) cpu_start: Project name:     hdd
I (564) cpu_start: App version:      3.0.0-alpha
I (569) cpu_start: Compile time:     May  9 2020 15:17:14
I (575) cpu_start: ELF file SHA256:  e7dd6031af873866...
[... 194 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e2d00: 6f000000 ffffffff ffffffff
0x400e2d00: esp_flash_init_default_chip at /home/micheleriva/esp-idf/components/spi_flash/esp_flash_spi_init.c:196 (discriminator 1)

Core 0 register dump:
PC      : 0x400e2d05  PS      : 0x00060330  A0      : 0x80081384  A1      : 0x3ffe3b70  
0x400e2d05: esp_flash_init_default_chip at /home/micheleriva/esp-idf/components/spi_flash/esp_flash_spi_init.c:198 (discriminator 1)

A2      : 0x00000000  A3      : 0x3f400018  A4      : 0x00000018  A5      : 0x3ffe3bd8  
A6      : 0x80123bfe  A7      : 0x01000000  A8      : 0x00800000  A9      : 0x00800000  
A10     : 0x00000000  A11     : 0x0000000a  A12     : 0x3ffbf4b4  A13     : 0x00000010  
A14     : 0xffffff0f  A15     : 0x00000030  SAR     : 0x00000037  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: ffffffffffffffffffffffffffffffff0000006fffffffffffffffffffffffff

Backtrace: 0x400e2d02:0x3ffe3b70 0x40081381:0x3ffe3bb0 0x40081645:0x3ffe3c00 0x40079033:0x3ffe3c40 |<-CORRUPTED
0x400e2d02: esp_flash_init_default_chip at /home/micheleriva/esp-idf/components/spi_flash/esp_flash_spi_init.c:197

0x40081381: start_cpu0_default at /home/micheleriva/esp-idf/components/esp32/cpu_start.c:429

0x40081645: call_start_cpu0 at /home/micheleriva/esp-idf/components/esp32/cpu_start.c:263

Rebooting...
```

atanisoft, 2020-05-11:
arduino-esp32 is currently using IDF v3.2 with plans to move up to IDF v3.3 and soon there after IDF v4.x. For the arduino-esp32 code, are you using the idf-v4.0 branch?

maikeriva, 2020-05-11:
Thanks for your quick answer. I am using IDF v4.1 beta (while waiting for it to become stable in the next few months).
IDF v3.2 unfortunately lacks of several features I need. Do you think it is feasible to achieve compatibility (if that's the problem) by tweaking IDF v4.x configuration?

atanisoft, 2020-05-11:
I'm not certain as I haven't moved beyond IDF v4.0 yet myself. 

@me-no-dev any ideas?

lbernstone, 2020-05-11:
This could be a bootloader incompatibility.  It is unclear to me if the issue occurs using a new bootloader (IDF4.1) with an older firmware (Arduino 1.0.4), or the other way around.  The new bootloader should work with older code, the inverse is not guaranteed.

maikeriva, 2020-05-11:
> This could be a bootloader incompatibility. It is unclear to me if the issue occurs using a new bootloader (IDF4.1) with an older firmware (Arduino 1.0.4), or the other way around. The new bootloader should work with older code, the inverse is not guaranteed.

That's the case indeed. I flash the old arduino-based firmware, then perform an OTA update to the new IDF v4.1 firmware, and the problem occurs.

lbernstone, 2020-05-11:
Really, that ought to work.  Is it reproducible with a simple example sketch or an empty sketch?

atanisoft, 2020-05-11:
IDF v3.3.1 fixed a bootloader issue related to this it looks like. You may need to reflash the bootloader piece to v3.3.1 first and then make the jump to IDF v4. I don't know if there is any forward compatibility from IDF v3.2.x to IDF v4.x app.

maikeriva, 2020-05-11:
> Really, that ought to work. Is it reproducible with a simple example project?
> Check that the bootloaders match. The error is pointing to flash init, so maybe one bootloader is DIO, and then you are trying to OTA a QIO compiled project?

That was the right guess! The old firmware was compiled for QIO mode at 80MHz while the new one for DIO at 40MHz. I aligned them both on QIO at 80MHz and now the boot completes successfully.
Thanks a lot for the suggestions!

closed 2020-07-25 by stale[bot]