espressif/arduino-esp32 #3994: joystick and wifi.h problem

amirjak, 2020-05-13:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?   joystick add to esp32 and use wifi at same time 
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? 
Core Installation version:			?1.0.4?
IDE name:						?Arduino IDE? 
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?no? 
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>

void setup() {

  Serial.begin(115200);
  Serial.print("Connecting to Network");
  WiFi.mode(WIFI_STA);                      // Mode Station
  WiFi.begin(ssid, password);               // Matching the SSID and Password
  // Waiting to connect to wifi
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);

}

void loop() {
  int x = analogRead(34);
  int y = analogRead(27);
  Serial.print("x : ");
  Serial.print(x);
  Serial.print("____ y : ");
  Serial.println(y);
}

amirjak, 2020-05-13:
x : 2672____ y : 4095

amirjak, 2020-05-13:
serial monitor

x : 2666____ y : 4095
x : 2654____ y : 4095
x : 2669____ y : 4095
x : 2671____ y : 4095
x : 2655____ y : 4095
x : 2659____ y : 4095
x : 2684____ y : 4095

amirjak, 2020-05-13:
I use pin 27 ADC2_ch7 and ADC1_ch6 
Whine wifi connected not working correct
pin  32-33-34-35-36-39 not working correct 

serial monitor

x : 2666____ y : 4095
x : 2654____ y : 4095
x : 2669____ y : 4095
x : 2671____ y : 4095
x : 2655____ y : 4095
x : 2659____ y : 4095
x : 2684____ y : 4095

amirjak, 2020-05-13:
[image]

amirjak, 2020-05-13:
I use pin 27 ADC2_ch7 and ADC1_ch6
When I am not touching the joystick and wifi connected not working correct 
pin 32-33-34-35-36-39 not working correct

serial monitor Read:

x : 2666____ y : 4095
x : 2654____ y : 4095
x : 2669____ y : 4095
x : 2671____ y : 4095
x : 2655____ y : 4095
x : 2659____ y : 4095
x : 2684____ y : 4095

amirjak, 2020-05-13:
I use ADC2 pin and find out. ADC2 is used by the Wi-Fi driver. Therefore the application can only use ADC2 when the Wi-Fi driver has not started.

But not working whit ADC1 pin.

atanisoft, 2020-05-13:
27 is on ADC2, while WiFi is active only ADC1 should be used which maps to pins above 30.

atanisoft, 2020-05-13:
Can you share your connections on the joystick module?

amirjak, 2020-05-13:
Hi, thank you for your quick answering

amirjak, 2020-05-13:
#define SW_PIN 5
#define VRY_PIN 34
#define VRX_PIN 27
#define LOW_RANGE 0
#define HIGH_RANGE 3500
#define RANGE_DIVITION 100

amirjak, 2020-05-13:
#define VRY_PIN 34 or 35.36.39
#define VRX_PIN 27  or 34.35.36.39

amirjak, 2020-05-13:
When  wifi connected not working correct

atanisoft, 2020-05-13:
And how are you powering the joystick module? If you use say 34 and 35 for x and y, does it always read 4095 for y? Can you swap the pins for x and y?

amirjak, 2020-05-13:
### **I change my code**

#define SW_PIN 5
#define VRY_PIN 35
#define VRX_PIN 34
#define LOW_RANGE 0
#define HIGH_RANGE 4095
#define RANGE_DIVITION 100

### **now**   Up and the right work together at same time

amirjak, 2020-05-13:
### **but left and dawn  working correct**

amirjak, 2020-05-13:
Powering the joystick module with 5v

atanisoft, 2020-05-13:
Power the joystick with 3.3v as anything higher will read as 4095 when you are beyond a certain point in the axis.

atanisoft, 2020-05-13:
And an example found via Google: https://esp32.com/viewtopic.php?f=19&t=14803 (where this should have likely been posted since it looks like user error or faulty joystick module)

amirjak, 2020-05-13:
Power the joystick with 3.3v 
### **working correct**
 thank you for your quick answering
### **When it's up**
 Up: 1 | Down: 0 | Left: 0 | Right: 0 | VRx: 2002 | VRy: 4095 | SW: 1 |
### **When it's Right**
Up: 0 | Down: 0 | Left: 0 | Right: 1 | VRx: 4095 | VRy: 1840 | SW: 1 |
### **When it's Down**
Up: 0 | Down: 1 | Left: 0 | Right: 0 | VRx: 2066 | VRy: 0 | SW: 1 |
### **When it's Left**
Up: 0 | Down: 0 | Left: 1 | Right: 0 | VRx: 0 | VRy: 1781 | SW: 1 |

amirjak, 2020-05-13:
thanks a lot### ****

amirjak, 2020-05-13:
### **thanks a lot**

atanisoft, 2020-05-13:
Keep in mind the esp32 is a 3v3 device even if it has a 5v pin on most boards, which is usually connected to the USB connection and prior to the voltage regulator for the esp32.

If your issue has been fixed please close.

amirjak, 2020-05-13:
### **Thanks for information.**

closed 2020-05-13 by amirjak