espressif/arduino-esp32 #4000: "Guru Meditation Error" when calibrating IMU sensor on ESP32

froggynardo, 2020-05-15:
I am trying to calibrate an Adafruit precision NXP 9dof IMU (full name) on an esp32 board, and whenever I run Adafruit's library to calibrate the magnetometer, I get the following error message:
``` 
Adafruit AHRS - IMU Calibration!
Calibration filesys test
Loaded existing calibration
------------------------------------
Sensor:       FXOS8700_A
Type:         Acceleration (m/s2)
Driver Ver:   1
Unique ID:    8701
Min Value:    -78.45
Max Value:    78.45
Resolution:   0.06
------------------------------------

------------------------------------
Sensor:       FXAS21002C
Type:         Gyroscopic (rad/s)
Driver Ver:   1
Unique ID:    2162732
Min Value:    -4.36
Max Value:    4.36
Resolution:   0.00
------------------------------------

------------------------------------
Sensor:       FXOS8700_M
Type:         Magnetic (uT)
Driver Ver:   1
Unique ID:    8702
Min Value:    -1200.00
Max Value:    1200.00
Resolution:   0.00
------------------------------------

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2551  PS      : 0x00060530  A0      : 0x800d25e8  A1      : 0x3ffb1f10  
A2      : 0x00000000  A3      : 0x00000002  A4      : 0x00000000  A5      : 0x0000006a  
A6      : 0x3ffc023c  A7      : 0x000000c3  A8      : 0x00000000  A9      : 0x3ffc01b0  
A10     : 0x3ffc01b0  A11     : 0x00000000  A12     : 0x00000024  A13     : 0x3ffc01d4  
A14     : 0x00000002  A15     : 0x00000002  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400d2551:0x3ffb1f10 0x400d25e5:0x3ffb1f60 0x400d1129:0x3ffb1f80 0x400d3ab9:0x3ffb1fb0 0x400889d9:0x3ffb1fd0

Rebooting...
```

I have not changed any of Adafruit's code (link to the guide I'm using here: https://learn.adafruit.com/nxp-precision-9dof-breakout/arduino-code), so I am completely in the dark as to why this is happening. Any possible solutions?

chegewara, 2020-05-16:
https://github.com/espressif/arduino-esp32#decoding-exceptions

froggynardo, 2020-05-16:
I've decoded my error using the esp32exceptiondecoder tool, and what results is this:
```
PC: 0x400d3c09: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 146
EXCVADDR: 0x00000010

Decoding stack results
0x400d3c09: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 146
0x400d3c9d: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 164
0x400d1462: HardwareSerial::begin(unsigned long, unsigned int, signed char, signed char, bool, unsigned long) at C:\Users\MyUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\HardwareSerial.cpp line 44
0x400d93f1: uart_tcsetattr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs_uart.c line 453
0x400889d9: taskYIELD_OTHER_CORE at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 660
```
The only line I was able to trace anything to was this line (I couldn't find where /home/runner/work... was located on my computer):
``` 
0x400d1462: HardwareSerial::begin(unsigned long, unsigned int, signed char, signed char, bool, unsigned long) at C:\Users\MyUser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\HardwareSerial.cpp line 44
```
This is the snippet of code that line was pointing to, line 44 is the one with all the astericks:
```
HardwareSerial::HardwareSerial(int uart_nr) : _uart_nr(uart_nr), _uart(NULL) {}

void HardwareSerial::begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, bool invert, unsigned long timeout_ms)
{
    if(0 > _uart_nr || _uart_nr > 2) {
        log_e("Serial number is invalid, please use 0, 1 or 2");
        return;
    }
    if(_uart) {
        end();
    }
    if(_uart_nr == 0 && rxPin < 0 && txPin < 0) {
        rxPin = 3;
        txPin = 1;
    }
    if(_uart_nr == 1 && rxPin < 0 && txPin < 0) {
        rxPin = RX1;
********txPin = TX1;**********
    }
    if(_uart_nr == 2 && rxPin < 0 && txPin < 0) {
        rxPin = RX2;
        txPin = TX2;
    }
```
It seems to have something to do with the rx and tx pins, but I am not using those pins at all, because I have my IMU sensor connected to my esp32 via I2C. Is there something I'm missing?

chegewara, 2020-05-16:
Do you call `Serial.begin()` more than 1 time in your code? Dont do that.

froggynardo, 2020-05-16:
Nope, serial is only called once. Here is the code I'm using for calibration:
```
/***************************************************************************
  This is an example for the Adafruit SensorLab library
  It will look for a supported magnetometer and output
  PJRC Motion Sensor Calibration Tool-compatible serial data

  PJRC & Adafruit invest time and resources providing this open source code,
  please support PJRC and open-source hardware by purchasing products
  from PJRC!

  Written by PJRC, adapted by Limor Fried for Adafruit Industries.
 ***************************************************************************/

#include <Adafruit_SensorLab.h>
#include <Adafruit_Sensor_Calibration.h>

Adafruit_SensorLab lab;

#if defined(ADAFRUIT_SENSOR_CALIBRATION_USE_EEPROM)
  Adafruit_Sensor_Calibration_EEPROM cal;
#else
  Adafruit_Sensor_Calibration_SDFat cal;
#endif

Adafruit_Sensor *mag = NULL, *gyro = NULL, *accel = NULL;
sensors_event_t mag_event, gyro_event, accel_event;

int loopcount = 0;

void setup(void) {
  Serial.begin(115200);
  while (!Serial) delay(10);     // will pause Zero, Leonardo, etc until serial console opens
  
  Serial.println(F("Sensor Lab - IMU Calibration!"));
  lab.begin();

  Serial.println("Calibration filesys test");
  if (!cal.begin()) {
    Serial.println("Failed to initialize calibration helper");
    while (1) yield();
  }
  if (! cal.loadCalibration()) {
    Serial.println("No calibration loaded/found... will start with defaults");
  } else {
    Serial.println("Loaded existing calibration");
  }

  Serial.println("Looking for a magnetometer");
  mag = lab.getMagnetometer();
  if (! mag) {
    Serial.println(F("Could not find a magnetometer, skipping!"));
  } else {
    mag->printSensorDetails();
  }
  
  Serial.println("Looking for a gyroscope");
  gyro = lab.getGyroscope();
  if (! gyro) {
    Serial.println(F("Could not find a gyroscope, skipping!"));
  } else {
    gyro->printSensorDetails();
  }
  
  Serial.println("Looking for a accelerometer");
  accel = lab.getAccelerometer();
  if (! accel) {
    Serial.println(F("Could not find a accelerometer, skipping!"));
  } else {
    accel->printSensorDetails();
[... 146 lines cut ...]
    }
    // look for 117 in last byte
    if (caldata[67] == 117) {
      caldata[0] = 117;
      calcount = 1;
    } else {
      calcount = 0;
    }
  }
}

uint16_t crc16_update(uint16_t crc, uint8_t a)
{
  int i;
  crc ^= a;
  for (i = 0; i < 8; i++) {
    if (crc & 1) {
      crc = (crc >> 1) ^ 0xA001;
    } else {
      crc = (crc >> 1);
    }
  }
  return crc;
}
```

lbernstone, 2020-05-16:
You assign mag in setup, then use it in loop.  Scoping issues.

chegewara, 2020-05-16:
Also, this code is not consistent with logs from OP.

froggynardo, 2020-05-16:
Turns out it was a mixture of things. The sensor kept turning on and off, randomly not being detected until I shook it, and it worked, for a few seconds. I've redone all my soldering and it seems perfectly fine. Probably a defective sensor there; I'll be ordering another one.

As for my issue in code, one of the libraries was trying to write to a non-existent memory address. I changed it and it worked perfectly (for the few seconds that the sensor decided to function at all). 

Thanks for your help guys (or gals)! I'll be using esp32exceptiondecoder a lot more often now.

closed 2020-05-16 by froggynardo