espressif/arduino-esp32 #4006: ESP32-POE-ISO upload error: serial port write timeout

chris-kuhr, 2020-05-16:
### Hardware:
Board:							ESP32-PoE-ISO
Core Installation version:			1.0.3, 1.0.4, 1.0.4-rc1
IDE name:						Arduino IDE 1.8.12
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Linux Mint 18.3 (Ubuntu 16.04 base), kernel 4.10.0-38

### Description:
I am using  on a Lenovo Thinkpad T440p. Frequency scaling is on performance and PSU is plugged in. I try to upload a sketch and after "Connecting....." a Python "write timeout" exception is raised.
Until yesterday I used my newer Lenovo Thinkpad T460p with Linux Mint 18.2 (Ubuntu 16.04 base), but I updated the OS to Ubuntu 20.04 and there is no Python2 anymore, so I switched laptops. With the T460p I had never any problems and all my sketches and installations worked fine. 
With the T440p I tried different ESP-PoE-ISO boards, different USB cables, and stepped through all the ttyACM ports, but all the same error. I suspect that using ttyACM is the root of the problem, since this is, after all my digging, the only thing that changed between the two setups. But I may be wrong. Any suggestions?

Thanks,
Ck

### Sketch: 
```
void setup() {}
void loop() {}
```

### Debug Messages:
```
Sketch uses 707938 bytes (54%) of program storage space. Maximum is 1310720 bytes.
Global variables use 39856 bytes (12%) of dynamic memory, leaving 287824 bytes for local variables. Maximum is 327680 bytes.
python /home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/christoph/.arduino15/packages/esp32/hardware/esp32/1.0.4-rc1/tools/partitions/boot_app0.bin 0x1000 /home/christoph/.arduino15/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /tmp/arduino_build_478982/esp32MQTTBase_v1.1.ino.bin 0x8000 /tmp/arduino_build_478982/esp32MQTTBase_v1.1.ino.partitions.bin 
esptool.py v2.6
Serial port /dev/ttyACM0
Connecting.....
Traceback (most recent call last):
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2959, in <module>

    _main()
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2952, in _main
    main()
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 2653, in main
    esp.connect(args.before)
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 460, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 440, in _connect_attempt
    self.sync()
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 379, in sync
    timeout=SYNC_TIMEOUT)
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 322, in command
    self.write(pkt)
  File "/home/christoph/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 285, in write
    self._port.write(buf)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 535, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
the selected serial port serial.serialutil.SerialTimeoutException: Write timeout
 does not exist or your board is not connected
```

lbernstone, 2020-05-16:
Does it work if you just leave the port option out?  The default is typically /dev/ttyUSB0.  You should see an active port in `ls /dev/tty*`

chris-kuhr, 2020-05-16:
There are no ttyUSB ports, only 3  ttyACM0..2 ports. I found that suspcious, but don't know.

chris-kuhr, 2020-05-17:
Ok, this is super weired. I fired up the T440p to get the requested info. Suddenly there is `/dev/ttyUSB0` ! And the sketch upload worked as usual. Did it have to take the night off??? I did reboot yesterday, several times...
So, this issue can be closed as ghost in the machine...

closed 2020-05-17 by chris-kuhr