espressif/arduino-esp32 #4009: Use PSRAM for malloc()

Avenitos, 2020-05-18:
Hello!

I use Platformio IDE +arduino-esp32 (1.12.1)

Use flags:
```
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```

PSRam are available for ps_malloc(), but not for malloc()

Avenitos, 2020-05-18:
I try in latest Arduino IDE.

`void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.print("Heap: ");
  Serial.print(ESP.getFreeHeap());
  Serial.print(" PsRam: ");  
  Serial.println(ESP.getFreePsram());  
  
  uint8_t *testMallocBuffer = (uint8_t*)malloc(11000);
  if(testMallocBuffer == NULL) Serial.println("malloc failed");
  
  delay(5000);
}`

`Heap: 353420 PsRam: 4194252
Heap: 342404 PsRam: 4194252
Heap: 331388 PsRam: 4194252
Heap: 320372 PsRam: 4194252
Heap: 309356 PsRam: 4194252
Heap: 298340 PsRam: 4194252
Heap: 287324 PsRam: 4194252
Heap: 276308 PsRam: 4194252
Heap: 265292 PsRam: 4194252
`

lbernstone, 2020-05-18:
This is not available in arduino for the simple reason that PSRAM != SRAM.  If psram were added to the primary memory map, there would be hundreds of issues due to inapropriate use.

Avenitos, 2020-05-18:
I need to use the CONFIG_SPIRAM_USE_MALLOC parameter as described in:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-use

lbernstone, 2020-05-18:
How is that any better than using ps_alloc?  In any case, you would need to compile your own libraries.

Avenitos, 2020-05-18:
This option is the default in ESP-IDF, it is strange why this is not the case in Arduino. Since it is very convenient and does not require fixing third-party code to work with PSRAM. Especially since my code can work on boards without PSRAM, then it must use the built-in memory.

chegewara, 2020-05-18:
Even if its default case in idf that does not mean its best option. I am always changing it in IDF.

closed 2020-07-31 by stale[bot]