espressif/arduino-esp32 #4013: ESP32 device keeps rebooting error

Kevistar, 2020-05-20:
Hello, I have been using the Arduino IDE to program my ESP32. However, recently I discover there is a software error. Whenever I uploaded a code, the device will keep on rebooting. This is the error message displayed on the serial monitor:

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
`

Is there any method to solve this problem? Any help is appreciated. I really hope is just some firmware error that could be solved.

lbernstone, 2020-05-20:
Please add more information as per the [ISSUE TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).
This message you have posted is not really an error, it is the boot log, indicating that the RTC watchdog is being triggered. Does the upload complete successfully?  Do you have pins being held down or up during boot?  Does it boot loop with a simple sketch like [StartCounter](https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences/examples/StartCounter)?

Kevistar, 2020-05-20:
Hardware:
Board: node32  
Core Installation/update date: Unknown (Is there any guide for me to check?)
IDE name: Arduino IDE
Flash Frequency: 40Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

Description:
My program upload is successful. All the while it is working well with most codes until only this morning. I do not have any pins being held down or up during boot. It does not boot loop with a simple sketch.

closed 2020-08-02 by stale[bot]