espressif/arduino-esp32 #4016: Any downsides to increasing flash speed?

Schaggo, 2020-05-20:
Hi,

I am wondering what the downsides of increasing the flash speed from 40MHz to 80MHz are.

I am working on a project with the esp cam (Ai thinker) and found out that if you increase the flash speed to 80MHz it is possible to take stills with virtually no compression (jpgQuality = 1 -> file size ~ 1 MB).

Anything I should keep an eye on?

WereCatf, 2020-05-23:
@Schaggo Increasing the speed just increases power-consumption a little. If your project isn't battery-powered, then the increased power-consumption is irrelevant.

Schaggo, 2020-05-23:
> @Schaggo Increasing the speed just increases power-consumption a little. If your project isn't battery-powered, then the increased power-consumption is irrelevant.

Thanks for the reply!
I already measured the devices running with 80MHz and cannot see a difference in power consumption. My meter isn’t really accurate though.
Anyway, thanks for calming my nerves!

closed 2020-08-06 by stale[bot]