espressif/arduino-esp32 #4017: Serious Ubuntu 20.04 issues

phaleas, 2020-05-20:
New PC, fresh install of Ubuntu 20.04 LTS Desktop (Release), Arduino IDE 1.8.12

Error: "exec: "python": executable file not found in $PATH

Reason: python2 is obsolete, so there is no /usr/bin/python by default any more

Workaround:
sudo apt install python-is-python3 (just sets missing symlink)
sudo apt install python3-pip

Prevent python(2) to be installed via dependencies:
sudo apt-mark hold python2 python2-minimal python2.7 python2.7-minimal libpython2-stdlib

After this compiling works – but upload fails (A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header)

Upload works after pressing the „boot“ button on an NODEMCU 32S (on my 2nd system. Ubuntu 16.04 LTS, the upload works without pressing any button)

Ubuntu 20.04 is a major OS and should be supported without any (bleeding) workarounds

lbernstone, 2020-05-21:
This is ubuntu's issue, not arduino (or [esptool](https://github.com/espressif/esptool/issues/528)).  python3 got hammered in too quickly.  Do you think it is acceptable for a *ix system to not have a python in $PATH in 2020?

phaleas, 2020-05-21:
I agree with you that the radical change to python3 came as a surprise to many (including me). But python3 is already available since December 2008. Since it is still possible to install python2 additionally, a standard symlink to python3 under /usr/bin/python would be dangerous. Instead, there is one under /usr/bin/python3.

If you search the web for "arduino esp32 ubuntu 20.04" you will find countless posts about exactly this problem. In addition, there are many fixes that differ from mine - and some of which will lead to serious problems elsewhere.

We are talking about the NIX version with 50% market share, which is still quite fresh at the moment. In the future more and more users will migrate there - and start directly with a bad user experience at Espressif.

Since "arduino-esp32" works (to a large extent) with python3, a query to the python version should be a simple solution.

closed 2020-08-03 by stale[bot]