espressif/arduino-esp32 #4019: ESP32+CAM Wifi cannot start

555mortadha, 2020-05-21:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2020-05-21:
Set board type to "esp32 dev module", and then set core debug level to verbose.  Test with simple sketches like [WiFiClient](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino) and [WiFiScan](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiScan).
If you cannot find any SSIDs in a scan, you have signal issues with your antenna.

555mortadha, 2020-05-21:
I try with WifiClient and the code Goes well with Access Point Wifi,
I dont understant why this Wifi camera code did'nt work!

lbernstone, 2020-05-21:
Test with espressif example: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer

555mortadha, 2020-05-21:
[image]

I have launching the project but Wifi could not connect !
the dotted lines remain dotted

me-no-dev, 2020-05-21:
enable debug in the board menu (select ESP32 Dev Module if you haven't already). When you do so, you will see some messages on why wifi was not able to disconnect. In any way, not something that is a bug here :)

closed 2020-05-21 by me-no-dev

555mortadha, 2020-05-22:
⸮⸮[E][camera.c:205] skip_frame(): Timeout waiting for VSYNC
[E][camera.c:1270] esp_camera_init(): Camera init failed with error 0x20003

Here is the Debug message,
And if i restart for many time my ESP32+CAM :
Debug show : ..............................................
which mean that it loop in : 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

555mortadha, 2020-05-22:
Thanks for your support , i find my stupid way to connect, i was thinking that my ESP32+CAM will serve as Access Point, know everything seems good, thanks a lot 👍

Mahmoud7Osman, 2022-01-20:
> Thanks for your support , i find my stupid way to connect, i was thinking that my ESP32+CAM will serve as Access Point, know everything seems good, thanks a lot +1

what is the solution !! I have the same problem please help me