espressif/arduino-esp32 #4024: Is there a replacement for BLEAdvertisementData::setServiceSolicitation?

ryancole38, 2020-05-23:
### Description:
I'm trying to connect an ESP-32 to an iPhone to receive notifications, per [this thread](https://github.com/nkolban/esp32-snippets/issues/430).

The repo discussed in the thread uses older versions of the ESP-32 BLE libraries, and I'm trying to update it to use the current libraries.  In particular, it uses BLEAdvertisementData::setServiceSolicitation() to request the Apple Notification Center Service 
[here](https://github.com/S-March/esp32_ANCS/blob/master/ANCS/ANCS.ino) on line 206.  This references line 403 [here](https://github.com/S-March/esp32_ANCS/blob/master/Libraries/BLEAdvertising.cpp).

In the latest version of [BLEAdvertising](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertising.cpp) this function has been removed.

I've searched for replacement functions and googled to see what others have done, but I've found nothing so far.

What's the proper way of soliciting a service? Why was this removed in the first place?

Jeroen88, 2020-05-24:
There is an example in the BLE library that explains how to ask for a notification, and get it in a callback [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino)

ryancole38, 2020-05-27:
Thanks for pointing me in the right direction! I also realized that setServiceSolicitation was added specifically for that repository that I linked to, so it wasn't actually removed.

In case anyone in the future tries to go down this same path, there's already a [library](https://github.com/Smartphone-Companions/ESP32-ANCS-Notifications) that takes care of getting iOS notifications.

closed 2020-05-27 by ryancole38