espressif/arduino-esp32 #4026: Feature request: allow to have UART with only one pin inverted

positron96, 2020-05-25:
### Hardware:
Board:							Lolin32
Core Installation version:			1.0.4 installed via PlatformIO
IDE name:						VS Code + PlatformIO 

### Description:

Hello. 
I'm writing a ESP32 program that runs on a board with UART that has normal RX pin (idle high, received via a comparator), and TX pin is driven via open-collector transistor and hence MCU pin needs to be inverted (idle low).

I use `uartBegin` function from `esp32-hal-uart` header that has an argument to invert both pins (and it does not allow me to work with the board since pins are either both inverted or not), but from its implementation it's clear that pin inversion be enabled for each pin separately via `uartAttachRx` and `uartAttachTx` functions (https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L218). Unfortunately, they are not exported from .c file via .h file, as is `uart_struct_t`. If they were exported and could be used by end-users, it seems that my problem would be solved trivially.

So, this issue is either a feature request to allow to invert UART pins separately via:
 * an overloaded `uartBegin` (with two bool arguments for inverting Rx pin and Tx pin) or 
 * exposing `uartAttachRx`/`Tx` (and `uartDetachRx`/`Tx`) functions in header.

Or, if I can achieve my goal without requested modifications, please kindly show me the way.

lbernstone, 2020-05-25:
You can invert a single line using the IDF uart driver.
`uart_set_line_inverse(uart_num, UART_INVERSE_TXD);`

atanisoft, 2020-05-25:
@positron96 be warned about using IDF uart, it will conflict with Arduino uart.. it is one reason I did not solve it in LocoNet2 via code. It is a relatively simple to fix in hardware as well. Hans Tanner has also provided a better library than my LocoNet2 fork and a working hardware design that I'll be adopting soon

positron96, 2020-05-25:
@atanisoft What conflicts should I expect? @lbernstone's proposal seems to be what I need.

BTW LocoNet is exactly where I need that functionality! I've noticed that most DIY schematics use this mixed setup and since I'm much more a programmer than I am an electronics engineer, I'd like to keep the schematics and tweak the code.

atanisoft, 2020-05-25:
@positron96 The problem is the ISR usage is not consistent or shared between the two implementations. It is not recommended to mix/match the two implementations in a single application.

positron96, 2020-05-28:
Ok, got it. Then the question/feature request remains.

Meanwhile I have a solution to manually declare those functions as `extern`:
```
extern "C" void uartDetachRx(uart_t* uart);
extern "C" void uartDetachTx(uart_t* uart);
extern "C" void uartAttachRx(uart_t* uart, uint8_t rxPin, bool inverted);
extern "C" void uartAttachTx(uart_t* uart, uint8_t txPin, bool inverted);

...

_uart = uartBegin(uartNum, 115200, SERIAL_8N1, _rxPin, _txPin, 256, false);
if(_invertedRx) {		
		uartDetachRx(_uart);
		uartAttachRx(_uart, _rxPin, true);
}
if(_invertedTx) {
		uartDetachTx(_uart);
		uartAttachTx(_uart, _txPin, true);
}
```
Effectively, this method uses a non-public API, so can break in future versions of core. Still, it can be used until there is a better option.

closed 2020-08-11 by stale[bot]