espressif/arduino-esp32 #4028: Unable to modify file contents using Fs.h with any open mode

ignaciosolergarcia, 2020-05-26:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      yes
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Cannot modify contents of a file. FILE_WRITE  empties the file, FILE_APPEND makes seek to not have effect and "a+" works like FILE_APPEND.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "SD.h"
#include "Fs.h"

void setup() {
  Serial.begin(115200);
  while (!Serial) { ; }
}

void loop() {
  do {
    delay(100);
  } while (Serial.available() && Serial.read() >= 0);

  Serial.println(F("\nType any character to start\n"));
  while (!Serial.available()) {
    delay(100);
  }
  
  Serial.print("Initializing SD card...");

  if (!SD.begin(5)) {
    Serial.println("Card failed, or not present");
    while (1);
  }
  Serial.println("card initialized.");

  File file = SD.open("/test.txt", FILE_WRITE);
  file.write((const uint8_t*)"     is a test", 14);
  file.close();

  File file2 = SD.open("/test.txt", "a+"); // FILE_WRITE, FILE_APPEND do not work too
  file2.seek(0);
  file2.write((const uint8_t*)"this", 4);
  file2.close();

  File file3 = SD.open("/test.txt", FILE_READ);
  while (file3.available()) {
    Serial.write(file3.read());
  }
  file3.close();

  SD.remove("/test.txt");
}
```
### Result
This outputs:

Initializing SD card...card initialized.
     is a testthis
Type any character to start

While what it should is:

Initializing SD card...card initialized.
this is a test
Type any character to start

lbernstone, 2020-05-26:
append != insert.  If you want to shift the data in the file, you will need to capture the data and reconstruct the file.

ignaciosolergarcia, 2020-05-26:
I'm not shifting, I'm trying to overwrite it, I left the empty space at the beginning. a+ allows seek to work based on the documentation I got.

lbernstone, 2020-05-26:
This is posix behavior.  append locks the write position at the end of file.  Use "w"

ignaciosolergarcia, 2020-05-26:
w truncates the file before opening it so you get always an empty file, not valid for modification. Regarding POSIX [here](https://pubs.opengroup.org/onlinepubs/009695399/functions/fopen.html) says that a+ is open or create for update.

atanisoft, 2020-05-26:
> Regarding POSIX [here](https://pubs.opengroup.org/onlinepubs/009695399/functions/fopen.html) says that a+ is open or create for update.

@ignaciosolergarcia in that same page regarding "a+" it clearly states "writing at end-of-file." What you want is "r+" not "w+" (truncate) or "a+" (append!)

ignaciosolergarcia, 2020-05-27:
Got it, thanks. I got confused with the input & output terms and [this page](http://www.cplusplus.com/reference/cstdio/fopen/), I thought input meant entering data into the file.

closed 2020-05-27 by ignaciosolergarcia

aeljandro2, 2021-04-09:
Awesome!! I got the same issue! thanks