espressif/arduino-esp32 #4029: Stack canary watchpoint triggered (esp_timer) when sending an email after the end of a timer

dreamkiller52, 2020-05-26:
### Hardware:
Board:							?NodeMCU-32S?
Core Installation version:			?1.0.4?
IDE name:						?Arduino IDE? 
Flash Frequency:					?80Mhz?
PSRAM enabled:                                   
Upload Speed:					?921600?
Computer OS:                                        ?Windows 10?

### Description:
Hello, it's the first time i post an issue so i hope it's complete.

I have a fatal error when trying to send an email at the end of a ticker function it throw me a Stack canary watchpoint triggered (esp_timer) . I think the mail function is not directly in cause because it work outside my ticker callback function.

Somebody could help me with this one?

### Sketch: 
```cpp

#include <WiFi.h>
#include <Ticker.h> 
#include <ESP32_MailClient.h>

////////////Email notification
#define emailSenderAccount    "myemail@gmail.com"    
#define emailSenderPassword   "myapppasswd"
#define emailRecipient        "recipientemail@gmail.com"
#define smtpServer            "smtp.gmail.com"
#define smtpServerPort        465
#define emailSubject          "pump delivery"
SMTPData smtpData;
///////////////////////////////

////////////////////////wifi
const char* ssid = "myssid";
const char* password = "mywifipasswd";
///////////////////////////////////////////////////

Ticker PumpTimer;

void setup() {
  Serial.begin(115200);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  PumpTimer.once(10, end_timer);

}

void end_timer()
{
  sendmail("Test ok");
}

void sendCallback(SendStatus info);

void sendCallback(SendStatus msg) {
  // Print the current status
  Serial.println(msg.info());

  // Do something when complete
  if (msg.success()) {
    Serial.println("----------------");
  }
}

void sendmail(String message)
{
  smtpData.setLogin(smtpServer, smtpServerPort, emailSenderAccount, emailSenderPassword);
  
  // Set the sender name and Email
  smtpData.setSender("Pump nano 60L", emailSenderAccount);

  // Set Email priority or importance High, Normal, Low or 1 to 5 (1 is highest)
  smtpData.setPriority("High");

  // Set the subject
  smtpData.setSubject(emailSubject);

  // Set the message with HTML format
  smtpData.setMessage("<div style=\"color:#2f4468;\"><h1>Nano 60L delivery system</h1><p>-"+message+"</p></div>", true);

  // Add recipients, you can add more than one recipient
  smtpData.addRecipient(emailRecipient);
  //smtpData.addRecipient("YOUR_OTHER_RECIPIENT_EMAIL_ADDRESS@EXAMPLE.com");

  smtpData.setSendCallback(sendCallback);

  //Start sending Email, can be set callback function to track the status
  if (!MailClient.sendMail(smtpData))
    Serial.println("Error sending Email, " + MailClient.smtpErrorReason());

  //Clear all data from Email object to free memory
  smtpData.empty();
}

void loop() {
  yield();
}
```

### Debug Messages:
```
17:42:16.328 -> WiFi connected
17:42:16.328 -> IP address: 192.168.0.39
17:42:26.319 -> Connecting to SMTP server...
17:42:26.352 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
17:42:26.352 -> Debug exception reason: Stack canary watchpoint triggered (esp_timer) 
17:42:26.352 -> Core 0 register dump:
17:42:26.352 -> PC      : 0x400ed527  PS      : 0x00060c36  A0      : 0x800ee44b  A1      : 0x3ffb6630  
17:42:26.352 -> A2      : 0xf3a23d8b  A3      : 0x3ffb67c0  A4      : 0x3ffb6740  A5      : 0x70e70336  
17:42:26.386 -> A6      : 0x0000001c  A7      : 0x1a816bd8  A8      : 0x477f4002  A9      : 0x3d409e34  
17:42:26.386 -> A10     : 0xce000000  A11     : 0x3ffb6660  A12     : 0x3ffb6660  A13     : 0x3ffb5530  
17:42:26.386 -> A14     : 0x00000000  A15     : 0x3ffb67c0  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
17:42:26.386 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
17:42:26.419 -> 
17:42:26.419 -> Backtrace: 0x400ed527:0x3ffb6630 0x400ee448:0x3ffb6980 0x400ee9e2:0x3ffb69a0 0x400e64e1:0x3ffb6aa0 0x400e65ce:0x3ffb6b10 0x400e66b8:0x3ffb6bc0 0x400eed7e:0x3ffb6c30 0x400eee67:0x3ffb6de0 0x400eeea2:0x3ffb6e40 0x400d73a3:0x3ffb6e60 0x400d6bc2:0x3ffb70f0 0x400d6cbd:0x3ffb7120 0x400d355b:0x3ffb7140 0x400d4eea:0x3ffb7170 0x400d1fdd:0x3ffb7870 0x400d20c9:0x3ffb78c0 0x400df560:0x3ffb78f0 0x400df5c7:0x3ffb7910 0x40088ddd:0x3ffb7930
17:42:26.453 -> 
17:42:26.453 -> Rebooting...
```
Backtrace decode
```
PC: 0x400ed527: bt_i2c_write_set at phy_chip_v7_ana.c line 1176
EXCVADDR: 0x00000000

Decoding stack results
0x400ed527: bt_i2c_write_set at phy_chip_v7_ana.c line 1176
0x400ee448: ram_check_noise_floor at phy_chip_v7.c line 2056
0x400ee9e2: ram_spur_coef_cfg at phy_chip_v7.c line 1730
0x400e64e1: Print::printFloat(double, unsigned char) at C:\Users\flo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 278
0x400e65ce: Print::printFloat(double, unsigned char) at C:\Users\flo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Print.cpp line 307
0x400e66b8: Stream::readBytes(char*, unsigned int) at C:\Users\flo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\Stream.cpp line 239
0x400eed7e: set_rx_gain_cal_dc at phy_chip_v7.c line 547
0x400eee67: set_rx_gain_cal_dc at phy_chip_v7.c line 503
0x400eeea2: set_rx_gain_cal_dc at phy_chip_v7.c line 524
0x400d73a3: AsyncWebServerRequest::beginResponse(int, String const&, String const&) at C:\Users\flo\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 730
0x400d6bc2: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\flo\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 459
0x400d6cbd: AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at C:\Users\flo\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 488
0x400d355b: end_delivery(int) at C:\Users\flo\Documents\Arduino\test_spiff/test_spiff.ino line 782
0x400d4eea: WiFiClient::write(unsigned char const*, unsigned int) at C:\Users\flo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiClient.cpp line 353
0x400df560: SMTPData::SMTPData() at C:\Users\flo\Documents\Arduino\libraries\ESP32_Mail_Client\src/ESP32_MailClient.h line 1874
0x400df5c7: SMTPData::SMTPData() at c:\users\flo\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h line 544
0x40088ddd: vQueueWaitForMessageRestricted at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 2427

```

lbernstone, 2020-05-26:
Without knowing the internals of that library, I would say you are running out of stack memory.  The easy fix is to increase the stack size in [\<arduino\>/cores/esp32/main.cpp](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L28) to ~10000.  The real fix is to allocate large variables to heap instead of stack.  Passing around strings instead of pointers will also contribute to the problem.

dreamkiller52, 2020-05-26:
i modify this line 
```cpp
    //xTaskCreateUniversal(loopTask, "loopTask", 8192, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
	xTaskCreateUniversal(loopTask, "loopTask", 10240, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
```
but i have always the same issue

closed 2020-08-08 by stale[bot]