espressif/arduino-esp32 #4032: How to enable BT and BTLE at the same Time (Dual Mode)

BrushlessPower, 2020-05-28:
Hello i try to get the ESP32 Wroom Mudule working with BLE and BT (SerialBT) at the same Time (Dual Mode.

Only BLE is working withour Problems.
Only SerialBT is working without Problems.

But SerialBT and BLE doesn't work.
I can connect via BLE and BT but Notify with BLE isn't working any more.

Error Message:
[E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error

Maybe someone could give me a hint, how to get the ESP working in BT Dual Mode

caleedubya, 2020-06-29:
try this...

init BLE
init BT Classic

configure BLE

BrushlessPower, 2020-06-29:
yeah, i have done this already. It was mentioned somewhere else. But it was a earlier Version of ESP32 Arduino Code
For now it doesn't work.

// Create the BLE Device
  BLEDevice::init("SBBUA_ESC"); 
  BLEDevice::setPower(ESP_PWR_LVL_P7);
// Init BT Classic
  SerialBT.begin("SBBUA_ESC"); //Bluetooth device name

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  //BlHeli_Write_CHR.setCallbacks(new MyCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(UUID128_SVC_COMMUNICATION);

  pService->addCharacteristic(&BlHeli_Read_CHR);

  pService->addCharacteristic(&BlHeli_Write_CHR);
  BlHeli_Write_CHR.setCallbacks(new MyCallbacks());
 
  //BlHeli_Read_CHR.setValue("Read");
  BlHeli_Read_CHR.addDescriptor(&ReadDescriptor);
  BlHeli_Read_CHR.addDescriptor(new BLE2902());
  //BlHeli_Write_CHR.setValue("Write");
  BlHeli_Write_CHR.addDescriptor(&WriteDescriptor);
  BlHeli_Write_CHR.addDescriptor(new BLE2902());

    // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->setMinPreferred(0x0);
  pServer->getAdvertising()->start();

caleedubya, 2020-06-29:
I know this may sound silly, but try removing BLEDevice::setPower(ESP_PWR_LVL_P7);

BrushlessPower, 2020-06-30:
I can't reproduce it right now.

Everytime i start and connect the BLE-App it gives an Error because the custom BLE Service is not available.

Is there a way to advertise different Names for BT and BLE?

BrushlessPower, 2020-06-30:
with this line i could reproduce the error

SerialBT.begin("Test"); //Bluetooth device name
BLEDevice::init("SBBUA_ESC"); 
BLEDevice::setPower(ESP_PWR_LVL_P7);

Error Log:

V][BLECharacteristic.cpp:646] setValue(): >> setValue: length=10, data=244d3e04024254464c1a, characteristic UUID=00001002-0000-1000-8000-00805f9b34fb
08:58:45.017 -> [V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
08:58:45.017 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffba6f4), owner: <N/A> for <Unknown>
08:58:45.017 -> [V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
08:58:45.017 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffba6f4), owner: <Unknown>
08:58:45.017 -> [V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent
08:58:45.051 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SetValue (0x3ffba6f4), owner: <Unknown>
08:58:45.051 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
08:58:45.051 -> [V][BLECharacteristic.cpp:655] setValue(): << setValue
08:58:45.051 -> [V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: Unknown
08:58:45.051 -> [V][BLECharacteristic.cpp:477] notify(): >> notify: length: 10
08:58:45.084 -> [V][BLEUtils.cpp:1720] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
08:58:45.084 -> [D][BLECharacteristic.cpp:782] onNotify(): BLECharacteristicCallbacks
08:58:45.084 -> [V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
08:58:45.084 -> [D][BLECharacteristic.cpp:783] onNotify(): BLECharacteristicCallbacks
08:58:45.084 -> [V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
08:58:45.118 -> [V][GeneralUtils.cpp:296] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
08:58:45.118 -> [V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
08:58:45.118 -> [V][GeneralUtils.cpp:297] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
08:58:45.118 -> [V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
08:58:45.152 -> [V][GeneralUtils.cpp:323] hexDump(): 0000 24 4d 3e 04 02 42 54 46 4c 1a                    $M>..BTFL.
08:58:45.152 -> [V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
08:58:45.152 -> [E][BLECharacteristic.cpp:524] notify(): << esp_ble_gatts_send_ notify: rc=-1 Unknown ESP_ERR error
08:58:45.152 -> [V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent
08:58:45.152 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ConfEvt (0x3ffba694), owner: <N/A>
08:58:45.187 -> [D][BLECharacteristic.cpp:794] onStatus(): BLECharacteristicCallbacks
08:58:45.187 -> [D][BLECharacteristic.cpp:795] onStatus(): BLECharacteristicCallbacks

caleedubya, 2020-07-03:
Yeah BT needs to be right after BLE init.

closed 2020-09-16 by stale[bot]