espressif/arduino-esp32 #4033: ESPRUINO QIO/DIO

Zibri, 2020-05-28:
Today I was trying to flash ESPRUINO on ESP32-WROOM-32.

I noticed that if I flash it using QIO the boot fails.
If I flash it with DIO it works.

Any Idea why?

My module is this:

[image]

Zibri, 2020-05-28:
Note: also QOUT seems to work.
But if I flag SpiAutoSet, it selects QIO and QIO does not work.

Zibri, 2020-05-28:
By the way on this board the right LED is blue and is connected to D2... in espruino ESP32 LED1 is not defined.

lbernstone, 2020-05-29:
Putting jet fuel in the tank will not make your car fly.  If the board is manufactured with DIO flash ram, it is not capable of QIO operation.

Zibri, 2020-05-29:
I just don't know.. because it works both ways apparently.

closed 2020-08-11 by stale[bot]