espressif/arduino-esp32 #4036: Disable Auto Negotiation LAN8720 ESP32 Arduino

josephpal, 2020-05-29:
My setup: ESP32-WROVER-B connected to LAN8720A, selfmade PCB, the PHY will be connected transformerless, therefore, Auto-Negotiation has to be disabled.
IDE: Arduino IDE with ESP32 core installed

I want to disable auto-negotiation and don't know whether I have do to this by setting register values (http://ww1.microchip.com/downloads/en/Appnotes/00002190A.pdf, page 3) with Pull-UP/DOWN resistors or by using a provided function of ETH.h. It turns out that regarding this link (https://www.esp32.com/viewtopic.php?t=11873), mode configuration pins will be overwritten.

Thx in advance.

_Originally posted by @josephpal in https://github.com/espressif/arduino-esp32/issues/3407#issuecomment-635861218_

GitNik1, 2020-08-11:
We are also using the esp32 for our new products in the company and want disable the auto-negotiation but i think it is not required for the most private/hobby projects. I hope they have a solution in the future, also for professional users. The PHY should startup with the hardware configuration, without software resets and overrides some registers.

josephpal, 2020-08-14:
> We are also using the esp32 for our new products in the company and want disable the auto-negotiation but i think it is not required for the most private/hobby projects. I hope they have a solution [...]

I currently found a solution but i am  right now to busy because i am writing my final thesis. So in the next few weeks i hope i will provide a satisfying answer ;)

GitNik1, 2020-09-16:
hey josephpal could you tell me your solution?

josephpal, 2020-09-21:
> hey josephpal could you tell me your solution?

sry for my late response. Sure I can tell you how I was able to get it done, and for all the others trying to disable auto-negotiaion....right now, i can only provide a short answer, but i think I will create a new repo for this issue with a full LAN8720A PHY ESP32 Schematic, the C++ soure code and so on....

The most important point is the default `phy_lan8720_default_ethernet_config` [configuration](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/network/esp_eth.html#_CPPv412eth_config_t) provided by the espressif idf. This struct consists of the following properties and function pointers:

```cpp
typedef struct {
    eth_phy_base_t phy_addr;                                            /*!< PHY address (0~31) */
    eth_mode_t mac_mode;                                                /*!< MAC mode: only support RMII now */
    eth_clock_mode_t clock_mode;                                        /*!< external/internal clock mode selection */
    eth_tcpip_input_func tcpip_input;                                   /*!< tcpip input func */
    eth_phy_func phy_init;                                              /*!< phy init func */
    eth_phy_check_link_func phy_check_link;                             /*!< phy check link func */
    eth_phy_check_init_func phy_check_init;                             /*!< phy check init func */
    eth_phy_get_speed_mode_func phy_get_speed_mode;                     /*!< phy check init func */
    eth_phy_get_duplex_mode_func phy_get_duplex_mode;                   /*!< phy check init func */
    eth_gpio_config_func gpio_config;                                   /*!< gpio config func  */
    bool flow_ctrl_enable;                                              /*!< flag of flow ctrl enable */
    eth_phy_get_partner_pause_enable_func phy_get_partner_pause_enable; /*!< get partner pause enable */
    eth_phy_power_enable_func phy_power_enable;                         /*!< enable or disable phy power */
    uint32_t reset_timeout_ms;                                          /*!< timeout value for reset emac */
    bool promiscuous_enable;                                            /*!< set true to enable promiscuous mode */
} eth_config_t;
```

In order to configure the PHY, I followed the example for the arduino-esp32 plugin [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.h). Inside [Eth.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.cpp), line 86, you can clearly see the struct elements being overwritten and assigned.

To disable auto-negotiation, we have to make a few changes, because they are missing in the example above. First, like announced in the data sheet of the [LAN8720 PHY](http://ww1.microchip.com/downloads/en/devicedoc/00002165b.pdf#page=42) (p. 42), some bits have to be set in `BASIC CONTROL REGISTER`. This can be done by (e.g. `FULL_DUPLEX, 100Mbps`):

```cpp
esp_eth_smi_write(BASIC_MODE_CONTROL_REG, 0x2100);
```

To question now is where exactly we have to add this line of code. To follow the order provided by espressif itself, we can use a function call by the `eth_config_t` struct element (`eth_phy_func phy_init`):

```cpp
 /* PHY load default configuration */
eth_config = phy_lan8720_default_ethernet_config;

 /* ------------ override default configuration ------------ */
eth_config.phy_addr = (eth_phy_base_t) phy_addr;
eth_config.clock_mode = clk_mode;
eth_config.tcpip_input = tcpip_adapter_eth_input;
eth_config.flow_ctrl_enable = true;
eth_config.phy_init = _eth_phy_lan8720_init;
...

static esp_err_t  eth_phy_lan8720_init() {
     esp_eth_smi_write(BASIC_MODE_CONTROL_REG, 0x2100);
}
   
```

Right now, it won't work anyway....but why? I figured out that if you don't override the functions mention in `eth_config_t` struct element, the default functions will be used, since we have loaded the `phy_lan8720_default_ethernet_config` configuration. That means, in my case (arduino), precompile functions by the espressif idf version. 

The `phy_lan8720_default_ethernet_config` is defined [here](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/ethernet/eth_phy/phy_lan8720.h), whereas the corresponding functions which will be called using this configuration are implemented using this [source file](https://github.com/espressif/esp-idf/blob/release/v3.2/components/ethernet/eth_phy/phy_lan8720.c). And this is where it gets ugly....

```cpp
void phy_lan8720_check_phy_init(void)
{
    phy_lan8720_dump_registers();

    esp_eth_smi_wait_set(MII_BASIC_MODE_STATUS_REG, MII_AUTO_NEGOTIATION_COMPLETE, 0);
    esp_eth_smi_wait_set(PHY_SPECIAL_CONTROL_STATUS_REG, AUTO_NEGOTIATION_DONE, 0);
}
```

In file `phy_lan8720.c` we can clearly see, that using the `phy_lan8720_default_ethernet_config` configuration, the `phy_lan8720_check_phy_init` function above will be used (or a simillar one depending one the idf version you are using). But this means, that no matter what, we will wait until the auto-negotiation bits are set by the PHY. Thats the reason why setting the corresponding bits inside the  `BASIC CONTROL REGISTER` using the `eth_phy_lan8720_init` won't be enough.

[... 11 lines cut ...]
eth_config.phy_check_init = _eth_phy_lan8720_check_phy_init;

if(_eth_phy_power_pin >= 0){
       _eth_phy_power_enable_orig = eth_config.phy_power_enable;
       eth_config.phy_power_enable = _eth_phy_power_enable;
}
[... cut ...]
    //esp_eth_smi_wait_set(MII_BASIC_MODE_STATUS_REG, MII_AUTO_NEGOTIATION_COMPLETE, 1);
    //esp_eth_smi_wait_set(PHY_SPECIAL_CONTROL_STATUS_REG, AUTO_NEGOTIATION_DONE, 1);

    /* TODO: check preset register values */

}
```

All the other steps are similar to the implementation inside [Eth.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.cpp). This is how i was able to disable auto-negotiationI while configuring the PHY LAN8720A. hope this helps ;) If you have any question left, ask me.

josephpal, 2020-11-20:
> 
> 
> hey josephpal could you tell me your solution?

Did my solution work for you?

closed 2021-02-04 by stale[bot]