espressif/arduino-esp32 #4040: Why espressif esp32 packet is not working?

berguinsson, 2020-05-30:
Hi guys.
I downloaded the packet for esp32 months ago, and it was working perfectly, but today i receive this message:

Error downloading https://dl.espressif.com/dl/package_esp32_index.json

Therefore it doesn't appear any esp32 board. 

What can i do? I'm using Arduino IDE. 

Regards

chegewara, 2020-05-30:
Try again, maybe there was glitch on server. I can open it in browser.

berguinsson, 2020-05-30:
Hello chegewara.
Just delete everything of AppData/Local/Arduino15 and download again esp32 packet from the boards manager. Thas was the solution. 
Thank you for your response.
I hope this can help to others with the same issue.
Have a good day.

closed 2020-08-12 by stale[bot]