espressif/arduino-esp32 #4043: Connection GY-86 to ESP32

BestNgineering, 2020-05-30:
Hi I'm having issues connecting GY-86 to ESP32. I'm trying to use the gyroscope and accelerometer which use MPU6050 chip on the module. I'm using the MPU6050 library on Arduino IDE and I get this warning saying that MPU6050 is claiming that it is for avr board but your board is ESP32 architecture. I got it to uploaded but the value I received was 0 in every dimensions.

lbernstone, 2020-05-31:
We don't support 3rd party libraries here.  Did you google "esp32 mpu6050"?

BestNgineering, 2020-06-04:
> We don't support 3rd party libraries here. Did you google "esp32 mpu6050"?

Sorry, I've changed to Jrowberg's MPU6050 and I2cdev library and then use the MPU6050 DMP example. I've managed to upload to the board but don't know what pin I should connect to.

The I2C scanner also can't find the device

closed 2020-08-17 by stale[bot]